coolsnowwolf/lede #10226: MT7621 小米R3G编译最新版出错，使用来自Mtk SDK的Hwnat驱动

Destroy-God, 2022-10-04:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述
MT7621 小米R3G编译最新版出错，使用来自Mtk SDK的Hwnat驱动
### (1) 具体问题

A：编译最新版出错

### (2) 路由器型号和固件版本

A： 小米R3G

### (3) 详细日志

u171785@s099-n018:~/lede$ make V=s -j1
WARNING: Makefile 'package/feeds/kenzo/luci-app-bypass/Makefile' has a dependency on 'luci-lib-ipkg', which does not exist
make[1]: Entering directory '/home/u171785/lede'
make[2]: Entering directory '/home/u171785/lede'
+ mkdir -p /home/u171785/lede/staging_dir/target-mipsel_24kc_musl
+ cd /home/u171785/lede/staging_dir/target-mipsel_24kc_musl
+ mkdir -p bin lib stamp usr/include usr/lib
mkdir -p /home/u171785/lede/build_dir/target-mipsel_24kc_musl/stamp
touch /home/u171785/lede/staging_dir/target-mipsel_24kc_musl/.prepared
make[3]: Entering directory '/home/u171785/lede/toolchain/gdb'
CFLAGS="-O2 -I/home/u171785/lede/staging_dir/host/include " CPPFLAGS="-I/home/u171785/lede/staging_dir/host/include " CXXFLAGS="" LDFLAGS="-L/home/u171785/lede/staging_dir/host/lib " make  -C /home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1  
make[4]: Entering directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1'
make[5]: Entering directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1'
make[6]: Entering directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/libiberty'
make[7]: Entering directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/libiberty/testsuite'
make[7]: Nothing to be done for 'all'.
make[7]: Leaving directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/libiberty/testsuite'
make[6]: Leaving directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/libiberty'
make[6]: Entering directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/intl'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/intl'
make[6]: Entering directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/zlib'
true "AR_FLAGS=rc" "CC_FOR_BUILD=gcc" "CFLAGS=-O2 -I/home/u171785/lede/staging_dir/host/include     " "CXXFLAGS=-g -O2    " "CFLAGS_FOR_BUILD=-O2 -I/home/u171785/lede/staging_dir/host/include " "CFLAGS_FOR_TARGET=-g -O2 -minterlink-mips16" "INSTALL=/home/u171785/lede/staging_dir/host/bin/install -c" "INSTALL_DATA=/home/u171785/lede/staging_dir/host/bin/install -c -m 644" "INSTALL_PROGRAM=/home/u171785/lede/staging_dir/host/bin/install -c" "INSTALL_SCRIPT=/home/u171785/lede/staging_dir/host/bin/install -c" "LDFLAGS=-L/home/u171785/lede/staging_dir/host/lib " "LIBCFLAGS=-O2 -I/home/u171785/lede/staging_dir/host/include   " "LIBCFLAGS_FOR_TARGET=-g -O2 -minterlink-mips16" "MAKE=make" "MAKEINFO=/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/missing makeinfo --split-size=5000000 --split-size=5000000 " "PICFLAG=" "PICFLAG_FOR_TARGET=" "SHELL=/usr/bin/env bash" "EXPECT=expect" "RUNTEST=runtest" "RUNTESTFLAGS=" "exec_prefix=/home/u171785/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl" "infodir=/home/u171785/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/share/info" "libdir=/home/u171785/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib" "prefix=/home/u171785/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl" "tooldir=/home/u171785/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/mipsel-openwrt-linux-musl" "AR=ar " "AS=as" "CC=gcc" "CXX=g++" "LD=ld" "LIBCFLAGS=-O2 -I/home/u171785/lede/staging_dir/host/include   " "NM=nm" "PICFLAG=" "RANLIB=ranlib " "DESTDIR=" DO=all multi-do # make
make[6]: Leaving directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/zlib'
make[6]: Entering directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/bfd'
Making info in po
make[7]: Entering directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/bfd/po'
make[7]: Nothing to be done for 'info'.
make[7]: Leaving directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/bfd/po'
make[7]: Entering directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/bfd'
make[7]: Circular doc/bfd.stamp <- doc/bfd.texi dependency dropped.
make[7]: Leaving directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/bfd'
make  all-recursive
make[7]: Entering directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/bfd'
Making all in po
make[8]: Entering directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/bfd/po'
make[8]: Nothing to be done for 'all'.
make[8]: Leaving directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/bfd/po'
make[8]: Entering directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/bfd'
make[8]: Circular doc/bfd.stamp <- doc/bfd.texi dependency dropped.
make[8]: Leaving directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/bfd'
make[7]: Leaving directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/bfd'
make[6]: Leaving directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/bfd'
make[6]: Entering directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/opcodes'
make  all-recursive
make[7]: Entering directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/opcodes'
Making all in .
make[8]: Entering directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/opcodes'
make[8]: Leaving directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/opcodes'
Making all in po
make[8]: Entering directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/opcodes/po'
make[8]: Nothing to be done for 'all'.
make[8]: Leaving directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/opcodes/po'
[... 115 lines cut ...]
checking for shared library run path origin... done
checking for iconv... yes
checking for iconv declaration... 
         extern size_t iconv (iconv_t cd, char * *inbuf, size_t *inbytesleft, char * *outbuf, size_t *outbytesleft);
checking for library containing waddstr... no
configure: error: no enhanced curses library found; disable TUI
Makefile:11425: recipe for target 'configure-gdb' failed
make[5]: *** [configure-gdb] Error 1
make[5]: Leaving directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1'
Makefile:1001: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1'
Makefile:73: recipe for target '/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/.built' failed
make[3]: *** [/home/u171785/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gdb-12.1/.built] Error 2
make[3]: Leaving directory '/home/u171785/lede/toolchain/gdb'
time: toolchain/gdb/compile#3.71#3.80#13.83
    ERROR: toolchain/gdb failed to build.
toolchain/Makefile:95: recipe for target 'toolchain/gdb/compile' failed
make[2]: *** [toolchain/gdb/compile] Error 1
make[2]: Leaving directory '/home/u171785/lede'
toolchain/Makefile:93: recipe for target '/home/u171785/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/u171785/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/u171785/lede'
/home/u171785/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

S0SL, 2022-10-05:
依赖没安装吧    **curses**
configure: error: no enhanced curses library found; disable TUI

Destroy-God, 2022-10-05:
谢谢，编译问题已经解决了，但是现在还有一个问题，就是使用MTK的HWNAT时wan状态好长一段时间都是未连接，要等一段时间才能用，这是为什么啊
内核日志在这里：
`[    0.000000] Linux version 5.4.215 (runner@fv-az341-509) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r5209-758af54df)) #0 SMP Tue Oct 4 03:37:15 2022
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Xiaomi Mi Router 3G
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] On node 0 totalpages: 65536
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 14 pages/cpu s27184 r8192 d21968 u57344
[    0.000000] pcpu-alloc: s27184 r8192 d21968 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00000848
[    0.000000] Readback ErrCtl register=00000848
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 249828K/262144K available (6378K kernel code, 273K rwdata, 1320K rodata, 1292K init, 236K bss, 12316K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.007887] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.074058] pid_max: default: 32768 minimum: 301
[    0.078856] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.086057] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.096892] rcu: Hierarchical SRCU implementation.
[    0.102376] smp: Bringing up secondary CPUs ...
[    1.254391] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    1.254402] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    1.254415] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    1.254520] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.167178] Synchronize counters for CPU 1: done.
[    2.563069] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    2.563078] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    2.563086] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    2.563140] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.258080] Synchronize counters for CPU 2: done.
[    2.327024] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    2.327033] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    2.327041] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    2.327101] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.343276] Synchronize counters for CPU 3: done.
[    0.373144] smp: Brought up 1 node, 4 CPUs
[    0.382061] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.391848] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[    0.398884] pinctrl core: initialized pinctrl subsystem
[    0.406079] NET: Registered protocol family 16
[    0.423888] FPU Affinity set after 11720 emulations
[    0.444267] workqueue: max_active 576 requested for napi_workq is out of range, clamping between 1 and 512
[... 388 lines cut ...]
[   95.717450] device eth0 entered promiscuous mode
[   95.755077] mt7530 mdio-bus:1f lan2: configuring for phy/gmii link mode
[   95.762392] 8021q: adding VLAN 0 to HW filter on device lan2
[   95.771956] br-lan: port 2(lan2) entered blocking state
[   95.777263] br-lan: port 2(lan2) entered disabled state
[   95.783919] device lan2 entered promiscuous mode
[   95.808496] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
[   95.815753] 8021q: adding VLAN 0 to HW filter on device wan
[   99.316338] mt7530 mdio-bus:1f lan2: Link is Up - 1Gbps/Full - flow control rx/tx
[   99.323855] br-lan: port 2(lan2) entered blocking state
[   99.329135] br-lan: port 2(lan2) entered forwarding state
[... cut ...]
[  127.482609] NFSD: starting 10-second grace period (net f00000a0)
[  260.356051] mt7530 mdio-bus:1f wan: Link is Down
[  262.436155] mt7530 mdio-bus:1f lan2: Link is Down
[  262.441106] br-lan: port 2(lan2) entered disabled state
[  276.996236] mt7530 mdio-bus:1f lan2: Link is Up - 1Gbps/Full - flow control rx/tx
[  277.003756] br-lan: port 2(lan2) entered blocking state
[  277.009025] br-lan: port 2(lan2) entered forwarding state
[  295.716315] mt7530 mdio-bus:1f wan: Link is Up - 100Mbps/Full - flow control off
`

Destroy-God, 2022-10-05:
又发现一个问题，无线直接给干没了，就好像设备里根本就没有无线一样，什么错误也没有出现

xsj520, 2022-10-12:
没有无线的问题修复好了吗？

closed 2026-09-05 by coolsnowwolf