coolsnowwolf/lede #10234: 5.15内核cpu频率缺失

AXBNG, 2022-10-06:
ipq807x系（ax6），用5.15内核编译出的固件/sys/devices/system/cpu/cpufreq目录为空，因此cpu频率get不到，温度显示正常，5.10内核一切正常。
（虽说已不再支持米系，还是小小反馈一下，若能排查修复万分感谢，如有冒犯可直接关闭issue）

mitsukileung, 2022-10-06:
Adbyby Plus+ 也没法启动。5.10时没有问题

xiangfeidexiaohuo, 2022-10-06:
我也发现了，虽然也不觉得影响啥

lovehackintosh, 2022-10-06:
我发现5.15的CoreMark只有1017mhz的10682分
,5.10跑分是1382mhz的18万分
仅编译performance调度

AXBNG, 2022-10-06:
> 我发现5.15的CoreMark只有1017mhz的10682分 ,5.10跑分是1382mhz的18万分 仅编译performance调度

我这保持默认的5.15大概1.7万多些，比5.10稍低些倒也不会差太多

lovehackintosh, 2022-10-06:
> 我这保持默认的5.15大概1.7万多些，比5.10稍低些倒也不会差太多hhh

我直接锁定最低频率，也改不了

mitsukileung, 2022-10-06:
> > 我这保持默认的5.15大概1.7万多些，比5.10稍低些倒也不会差太多hhh
> 
> 我的直接锁最低频率，也改不了，不知道删掉这几行行不行
> 
> ```
> 		opp-1017600000 {
> 			opp-hz = /bits/ 64 <1017600000>;
> 			opp-microvolt = <1>;
> 			clock-latency-ns = <200000>;
> 		};
> ```

大佬，还有日志不断出现这些内容是什么？

Thu Oct  6 23:27:13 2022 daemon.notice hostapd: wlan1: BEACON-REQ-TX-STATUS a4:d1:8c:12:09:3c 156 ack=1
Thu Oct  6 23:27:13 2022 daemon.notice hostapd: wlan1: BEACON-RESP-RX a4:d1:8c:12:09:3c 156 00
Thu Oct  6 23:27:13 2022 daemon.notice hostapd: wlan1: BEACON-REQ-TX-STATUS fc:be:7b:3a:de:7d 159 ack=1
Thu Oct  6 23:27:13 2022 daemon.notice hostapd: wlan1: BEACON-REQ-TX-STATUS d2:82:53:32:0b:21 158 ack=1
Thu Oct  6 23:27:13 2022 daemon.notice hostapd: wlan1: BEACON-RESP-RX d2:82:53:32:0b:21 158 04
Thu Oct  6 23:27:13 2022 daemon.notice hostapd: wlan1: BEACON-REQ-TX-STATUS 84:8e:0c:9e:95:1e 154 ack=1
Thu Oct  6 23:27:13 2022 daemon.notice hostapd: wlan1: BEACON-RESP-RX 84:8e:0c:9e:95:1e 154 00
Thu Oct  6 23:27:14 2022 daemon.notice hostapd: wlan1: BEACON-REQ-TX-STATUS 78:7e:61:32:e0:5b 153 ack=0
Thu Oct  6 23:27:22 2022 daemon.notice hostapd: wlan2: BEACON-REQ-TX-STATUS f4:f5:db:cc:1e:59 61 ack=1
Thu Oct  6 23:27:22 2022 daemon.notice hostapd: wlan2: BEACON-REQ-TX-STATUS 9c:2e:a1:b2:61:19 59 ack=0
Thu Oct  6 23:27:22 2022 daemon.notice hostapd: wlan2: BEACON-RESP-RX f4:f5:db:cc:1e:59 61 00 510b0000000000000000000002c4008cdef9b46acd000000000001d6fedfbccf060000006400311400074d697473756b69010882848b960c12182403010b05040002010a0706555320010b1e2a010232043048606c30140100000fac040100000fac040100000fac020c000b050500530000460572000000003b0251002d1aef1103ffff0000000000000000000001000000000000000000003d160b0004000000000000000000000000000000000000007f0a04000802010001400040ff1a230d00189a401002604c887ec1839c010c00fafffaff191cc771ff0724f43f000cfcffff0e260008a9ff2fa9ff4575ff6575ff
Thu Oct  6 23:27:22 2022 daemon.notice hostapd: wlan2: BEACON-REQ-TX-STATUS de:aa:1a:8b:30:1a 60 ack=1
Thu Oct  6 23:27:22 2022 daemon.notice hostapd: wlan2: BEACON-RESP-RX de:aa:1a:8b:30:1a 60 4f
Thu Oct  6 23:27:23 2022 daemon.notice hostapd: wlan2: BEACON-REQ-TX-STATUS 90:b2:1f:c0:bd:b9 58 ack=1
Thu Oct  6 23:27:23 2022 daemon.notice hostapd: wlan2: BEACON-RESP-RX 90:b2:1f:c0:bd:b9 58 00
Thu Oct  6 23:27:23 2022 daemon.notice hostapd: wlan2: BEACON-REQ-TX-STATUS 70:47:e9:c3:76:1b 57 ack=1

lovehackintosh, 2022-10-07:
@mitsukileung wifi日志,不用管忽略即可

lovehackintosh, 2022-10-07:
关于频率缺失这个问题,可以尝试添加mhz这个包https://github.com/robimarko/openwrt/commit/30ee62ebab0fc45e4d14cac62fdc6cff91bac59a

AXBNG, 2022-10-07:
> 关于频率缺失这个问题,可以尝试添加mhz这个包[robimarko/openwrt@30ee62e](https://github.com/robimarko/openwrt/commit/30ee62ebab0fc45e4d14cac62fdc6cff91bac59a)

是的我也看到了这个，还没有尝试，但我想类似于netdata和cpufreq之类的软件应该仍然无法读取(╯︵╰)

coolsnowwolf, 2022-10-16:
[image]

已修复

lovehackintosh, 2022-10-17:
是这个commit修复的吗https://github.com/coolsnowwolf/luci/commit/2a380a31c99bce8f3ceb22645abeb33f09299b2e
我上次编译时确实选的Shortcut-FE ECM，cpu频率缺失

closed 2022-10-17 by aiamadeus

mitsukileung, 2022-10-17:
> 是这个commit修复的吗[coolsnowwolf/luci@2a380a3](https://github.com/coolsnowwolf/luci/commit/2a380a31c99bce8f3ceb22645abeb33f09299b2e) 我上次编译时确实选的Shortcut-FE ECM,cpu频率缺失 还是因为我几天前改了一下dts修复的吗？[AXBNG@84437ba](https://github.com/AXBNG/LEDE/commit/84437ba92a19a6d96cb710bc3cce816c281165fd)

好像还是一样，还是缺失

AXBNG, 2022-10-17:
> [image]
> 
> 已修复

哈哈感谢大佬，有空重新编下试试

lovehackintosh, 2022-10-17:
全新clone编译问题依旧<details>
<summary>kernel日志</summary>
<pre><code>  
&nbsp;
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.15.74 (clover@msi) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 12.2.0 r5240-1953d466f) 12.2.0, GNU ld (GNU Binutils) 2.39) #0 SMP Mon Oct 17 01:55:17 2022
[    0.000000] Machine model: Xiaomi Mi AIoT Router AX3600
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000040ffffff]
[    0.000000]   node   0: [mem 0x0000000041000000-0x000000004a3fffff]
[    0.000000]   node   0: [mem 0x000000004a400000-0x000000004a4fffff]
[    0.000000]   node   0: [mem 0x000000004a500000-0x000000004a5fffff]
[    0.000000]   node   0: [mem 0x000000004a600000-0x000000004e8fffff]
[    0.000000]   node   0: [mem 0x000000004e900000-0x000000005fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] psci: OSI mode supported.
[    0.000000] percpu: Embedded 17 pages/cpu s30296 r8192 d31144 u69632
[    0.000000] pcpu-alloc: s30296 r8192 d31144 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] Kernel command line: console=ttyMSM0,115200n8 ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 414592K/524288K available (8704K kernel code, 904K rwdata, 1988K rodata, 1280K init, 284K bss, 109696K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv2m: range[mem 0x0b00a000-0x0b00affc], SPI[448:479]
[    0.000000] arch_timer: cp15 and mmio timer(s) running at 19.20MHz (virt/virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000000] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000100] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000117] pid_max: default: 32768 minimum: 301
[    0.000266] Mount-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.000283] Mountpoint-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.001774] rcu: Hierarchical SRCU implementation.
[    0.001920] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.002387] smp: Bringing up secondary CPUs ...
[    0.002942] Detected VIPT I-cache on CPU1
[    0.003008] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.003576] Detected VIPT I-cache on CPU2
[    0.003616] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.004142] Detected VIPT I-cache on CPU3
[    0.004178] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.004244] smp: Brought up 1 node, 4 CPUs
[    0.004268] SMP: Total of 4 processors activated.
[    0.004275] CPU features: detected: 32-bit EL0 Support
[    0.004282] CPU features: detected: CRC32 instructions
[    0.004331] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[    0.004341] CPU: All CPU(s) started at EL1
[    0.004362] alternatives: patching kernel code
[    0.011666] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.011702] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.011859] pinctrl core: initialized pinctrl subsystem
[    0.012899] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[... 1311 lines cut ...]
Mon Oct 17 12:49:50 2022 kern.warn kernel: [  198.704927] mtdblock: MTD device 'rootfs' is NAND, please consider using UBI block devices instead.
Mon Oct 17 12:49:50 2022 kern.warn kernel: [  198.719763] mtdblock: MTD device 'rsvd0' is NAND, please consider using UBI block devices instead.
Mon Oct 17 12:49:50 2022 kern.warn kernel: [  198.734689] mtdblock: MTD device '0:qsee' is NAND, please consider using UBI block devices instead.
Mon Oct 17 12:49:50 2022 kern.warn kernel: [  198.748376] mtdblock: MTD device '0:devcfg' is NAND, please consider using UBI block devices instead.
Mon Oct 17 12:49:50 2022 kern.warn kernel: [  198.762058] mtdblock: MTD device '0:rpm' is NAND, please consider using UBI block devices instead.
[... cut ...]
Mon Oct 17 12:50:23 2022 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Mon Oct 17 12:50:23 2022 daemon.notice hostapd: nl80211: Failed to remove interface wlan0 from bridge br-lan: Invalid argument
Mon Oct 17 12:50:23 2022 kern.info kernel: [  231.375630] ath10k_pci 0000:01:00.0: mac flush null vif, drop 0 queues 0xffff
Mon Oct 17 12:50:34 2022 daemon.err uhttpd[3685]: cat: can't open '/sys/devices/system/cpu/cpufreq/policy0/cpuinfo_cur_freq': No such file or directory
Mon Oct 17 12:50:34 2022 daemon.err uhttpd[3685]: expr: syntax error
</code></pre>
</details>

xiangfeidexiaohuo, 2022-10-17:
依旧获取不到

AXBNG, 2022-10-17:
> [image]
> 
> 已修复

经测试仍旧无法读取频率，另外commit中也没看到相关修改，不知大佬是如何修复的，一脸懵 (×﹏×)

lovehackintosh, 2022-10-17:
添加了mhz https://github.com/robimarko/openwrt/commit/30ee62ebab0fc45e4d14cac62fdc6cff91bac59a
发现cpu频率只有788mhz
```bash
root@OpenWrt:~# mhz
count=169252 us50=10587 us250=53503 diff=42916 cpu_MHz=788.759
root@OpenWrt:~# /etc/coremark.sh && cat /etc/bench.log 
(CpuMark : 10570.126181 
 Scores)
```

lovehackintosh, 2022-10-17:
@AXBNG 建议reopen this issue

AXBNG, 2022-10-18:
> @AXBNG 建议reopen this issue

很抱歉，issue不是我关的我无法reopen......

我打算冒着生命危险新开一个issue（滑稽