coolsnowwolf/lede #10236: X86  5.15内核固件 在线升级有问题

xiangfeidexiaohuo, 2022-10-06:
[image]

[image]

实际固件有291M，但是放到在线升级里，就死活才222M。

你说固件有问题吧，我直接全新写入这个291M的固件 是没毛病的。

我还特意把这个291M的515内核固件，放在5.19内核固件的在线升级里去加载，大小完全正确。
[image]

所以x86 5.15内核固件 在线升级 这块 应该是有点毛病

xiangfeidexiaohuo, 2022-10-07:
> 个人验证，无此问题
> [image]
> [image]

小体积固件没毛病

sjkhsl, 2022-10-07:
我的也是，可能体积大的会这样？

mitsukileung, 2022-10-07:
有可能是分配的内存不够，我之前就试过，升级不了固件

yongman, 2022-10-07:
@sjkhsl @xiangfeidexiaohuo sysupgrade命令行升级是否可行有试过吗

yongman, 2022-10-07:
@xiangfeidexiaohuo 5.19内核固件用着怎么样，稳定吗

xiangfeidexiaohuo, 2022-10-07:
> @xiangfeidexiaohuo 5.19内核固件用着怎么样，稳定吗

我用pve整的 没感觉 一直很稳定

xiangfeidexiaohuo, 2022-10-07:
> 有可能是分配的内存不够，我之前就试过，升级不了固件

yes 确实是 内存分配不够，因为我是pve测试，就分了个512M内存，改成768M内存甚至更大就没问题了。

xiangfeidexiaohuo, 2022-10-07:
> 我的也是，可能体积大的会这样？

内存给的不够，768起步 就够了