coolsnowwolf/lede #10237: ipq40xx平台qca-ssdk-hnet编译出错

v8wzge, 2022-10-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
ipq40xx平台qca-ssdk-hnet编译出错
Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：选择kmod-qca-ssdk-hnet编译失败

### (2) 路由器型号和固件版本

A：cm520-79f 星际宝盒 master最新源码

### (3) 详细日志

A：
malibu_phy.c: In function 'malibu_phy_set_powersave':
malibu_phy.c:384:20: error: 'status' may be used uninitialized [-Werror=maybe-uninitialized]
  384 |                 if (status == A_FALSE) {
      |                    ^
cc1: all warnings being treated as errors
make[6]: *** [/home/ubuntu/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/qca-ssdk-hnat/qca-ssdk-2021-04-28-c9bc3bc3/make/target.mk:13: malibu_phy.
o] Error 1
make[6]: Leaving directory '/home/ubuntu/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/qca-ssdk-hnat/qca-ssdk-2021-04-28-c9bc3bc3/src/hsl/phy'
make[5]: *** [/home/ubuntu/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/qca-ssdk-hnat/qca-ssdk-2021-04-28-c9bc3bc3/make/target.mk:7: obj] Error 1
make[5]: Leaving directory '/home/ubuntu/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/qca-ssdk-hnat/qca-ssdk-2021-04-28-c9bc3bc3/src/hsl'
make[4]: *** [Makefile:27: kslib_o] Error 1
make[4]: Leaving directory '/home/ubuntu/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/qca-ssdk-hnat/qca-ssdk-2021-04-28-c9bc3bc3'
make[3]: *** [Makefile:127: /home/ubuntu/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/qca-ssdk-hnat/qca-ssdk-2021-04-28-c9bc3bc3/.built] Error 2
make[3]: Leaving directory '/home/ubuntu/lede/package/qca/qca-ssdk'
time: package/qca/qca-ssdk/hnat/compile#45.36#6.26#52.08
    ERROR: package/qca/qca-ssdk failed to build (build variant: hnat).
make[2]: *** [package/Makefile:116: package/qca/qca-ssdk/compile] Error 1
make[2]: Leaving directory '/home/ubuntu/lede'
make[1]: *** [package/Makefile:110: /home/ubuntu/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
make: *** [/home/ubuntu/lede/include/toplevel.mk:230: world] Error 2

v8wzge, 2022-10-06:
编译环境是Ubuntu20.04 lts，硅谷的服务器

lovehackintosh, 2022-10-06:
应用这个https://github.com/coolsnowwolf/lede/pull/10229

v8wzge, 2022-10-06:
> 应用这个#10229

大佬，你们讨论的太高级，看不懂，是要应用你给出的13号补丁吗

hxlls, 2022-10-06:
> 应用这个#10229

应用后编译还是一样报错

hzjnet, 2022-10-07:
> > 应用这个#10229
> 
> 应用后编译还是一样报错

大家测试过这个hnat有作用没有啊？

v8wzge, 2022-10-07:
> > > 应用这个#10229
> > 
> > 
> > 应用后编译还是一样报错
> 
> 大家测试过这个hnat有作用没有啊？

没有编译成功过

hzjnet, 2022-10-07:
> 

0716版本正常 可以试试

hzjnet, 2022-10-07:
> > > > 应用这个#10229
> > > 
> > > 
> > > 应用后编译还是一样报错
> > 
> > 
> > 大家测试过这个hnat有作用没有啊？
> 
> 没有编译成功过

0716正常，可以试试

lovehackintosh, 2022-10-07:
```shell
wget -O package/qca/qca-ssdk/patches/0013-fix-build-error-for-gcc-12.patch https://github.com/lovehackintosh/lede/raw/fix-qca-ssdk/package/qca/qca-ssdk/patches/0013-fix-build-error-for-gcc-12.patch
```
现已支持arm架构,执行以上命令再试一次

hxlls, 2022-10-07:
> ```shell
> wget -O package/qca/qca-ssdk/patches/0013-fix-build-error-for-gcc-12.patch https://github.com/lovehackintosh/lede/raw/fix-qca-ssdk/package/qca/qca-ssdk/patches/0013-fix-build-error-for-gcc-12.patch
> ```
> 
> 现已支持arm架构,执行以上命令再试一次

可以编译过去了

hzjnet, 2022-10-07:
> 

好像1号以后的源码R7800能正常编译，但是不能启动。上月源码正常！

hzjnet, 2022-10-07:
> > ```shell
> > wget -O package/qca/qca-ssdk/patches/0013-fix-build-error-for-gcc-12.patch https://github.com/lovehackintosh/lede/raw/fix-qca-ssdk/package/qca/qca-ssdk/patches/0013-fix-build-error-for-gcc-12.patch
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 现已支持arm架构,执行以上命令再试一次
> 
> 可以编译过去了

好像1号以后的源码R7800能正常编译，但是不能启动。上月源码正常！

hxlls, 2022-10-07:
> 好像1号以后的源码R7800能正常编译，但是不能启动。上月源码正常！

编译好了，能正常启动，测试应该是不生效，ACRH17 跑CPU上涨很快，应该靠的是cpu 进行的

v8wzge, 2022-10-07:
> 编译好了，能正常启动，测试应该是不生效，ACRH17 跑CPU上涨很快，应该靠的是cpu 进行的

v8wzge, 2022-10-07:
> 编译好了，能正常启动，测试应该是不生效，ACRH17 跑CPU上涨很快，应该靠的是cpu 进行的

编译成功，之前星际宝盒ipq4019没有qca-ssdk-hnet，开启offloading，iperf3测试wan-to-lan只有300-380mpbs
现在添加该插件，开启和关闭offloading测速结果都一样，340-370mpbs
这个为ipq40xx加速的插件在星际宝盒上基本没用

closed 2022-10-07 by aiamadeus