coolsnowwolf/lede #10238: 有大量daemon.notice miniupnpd

rrkai, 2022-10-07:
Wed Oct  5 16:38:55 2022 kern.info kernel: [   61.899371] igb 0000:02:00.0 eth1: igb: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
Wed Oct  5 16:38:56 2022 kern.info kernel: [   62.122784] br-lan: port 2(eth1) entered blocking state
Wed Oct  5 16:38:56 2022 kern.info kernel: [   62.126841] br-lan: port 2(eth1) entered forwarding state
Wed Oct  5 16:38:56 2022 daemon.notice netifd: Network device 'eth1' link is up
Wed Oct  5 16:38:56 2022 daemon.info procd: Instance haproxy::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Wed Oct  5 16:39:56 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 16:42:18 2022 daemon.err uhttpd[2943]: sh: out of range
Wed Oct  5 16:42:18 2022 daemon.err uhttpd[2943]: sh: out of range
Wed Oct  5 16:45:28 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 16:51:00 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 16:56:32 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 17:02:05 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 17:07:37 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 17:13:09 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 17:18:41 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 17:24:13 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 17:29:45 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 17:35:18 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 17:40:50 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 17:46:22 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 17:51:54 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 17:57:26 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 18:02:59 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 18:08:31 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 18:14:03 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 18:19:35 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 18:25:07 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 18:30:39 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 18:36:12 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 18:41:44 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 18:47:16 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 18:52:48 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 18:58:20 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 19:03:52 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 19:09:24 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 19:14:57 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 19:20:29 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 19:26:01 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 19:31:33 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 19:37:05 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 19:42:37 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 19:48:10 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 19:53:42 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 19:59:14 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 20:04:46 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 20:10:18 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 20:15:50 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 20:21:22 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 20:26:55 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 20:32:27 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 20:37:59 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 20:43:31 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 20:49:03 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 20:54:35 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 21:00:08 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 21:05:40 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 21:11:12 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 21:16:44 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 21:22:16 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 21:27:48 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 21:33:20 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 21:38:53 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 21:44:25 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 21:49:57 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 21:55:29 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 22:01:01 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 22:06:33 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 22:12:05 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 22:17:37 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Wed Oct  5 22:23:10 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
[... 112 lines cut ...]
Thu Oct  6 07:25:39 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 07:31:11 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 07:36:43 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 07:42:15 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 07:47:47 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 07:53:20 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 07:58:52 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 08:04:24 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 08:09:56 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 08:15:28 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 08:21:00 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 08:26:32 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 08:32:05 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 08:37:37 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 08:43:09 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 08:48:41 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 08:54:13 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 08:59:46 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
[... cut ...]
Thu Oct  6 09:10:50 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 09:16:22 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 09:21:54 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 09:27:26 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 09:32:59 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus
Thu Oct  6 09:38:31 2022 daemon.notice miniupnpd[6114]: SoapMethod: Unknown: GetFirewallStatus

rrkai, 2022-10-08:
5.15的内核

coolsnowwolf, 2022-10-08:
关闭 IPv6

rrkai, 2022-10-08:
> 关闭 IPv6

惊现大屌! IPV6我没编译进去..不过LAN口有个.我禁用试试

rrkai, 2022-10-08:
> 关闭 IPv6

关闭能找到的IPV6都一样不行.还是会出现.....

rrkai, 2022-10-09:
更换了5.10的内核还是一样会出现

VergilGao, 2022-10-31:
实在不行，不用upnp，手动做端口映射好了。

rrkai, 2022-11-01:
> 实在不行，不用upnp，手动做端口映射好了。

那要做废了....哈哈

closed 2026-09-05 by coolsnowwolf