coolsnowwolf/lede #10242: x86软路由系统日志出现这种错误，请大佬看看如何解决。。。

drw2005, 2022-10-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86软路由，OpenWrt R22.9.1，内核版本：5.15.70

### (3) 详细日志

A：
Sun Oct  9 19:35:57 2022 daemon.info urandom_seed[5050]: Seed saved (/etc/urandom.seed)
Sun Oct  9 19:35:57 2022 daemon.err nmbd[4347]: [2022/10/09 19:35:57.367811,  0] ../../lib/util/become_daemon.c:135(daemon_ready)
Sun Oct  9 19:35:57 2022 daemon.err nmbd[4347]:   daemon_ready: daemon 'nmbd' finished starting up and ready to serve connections
Sun Oct  9 19:35:57 2022 daemon.err smbd[4346]: [2022/10/09 19:35:57.459934,  0] ../../lib/util/become_daemon.c:135(daemon_ready)
Sun Oct  9 19:35:57 2022 daemon.err smbd[4346]:   daemon_ready: daemon 'smbd' finished starting up and ready to serve connections
Sun Oct  9 19:35:59 2022 kern.info kernel: [   50.410943] igb 0000:04:00.0 eth3: igb: eth3 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
Sun Oct  9 19:35:59 2022 daemon.notice netifd: Network device 'eth3' link is up
Sun Oct  9 19:35:59 2022 daemon.notice netifd: Interface 'wan' has link connectivity
Sun Oct  9 19:35:59 2022 daemon.notice netifd: Interface 'wan' is setting up now
Sun Oct  9 19:35:59 2022 daemon.notice netifd: wan (5159): uci: Entry not found
Sun Oct  9 19:35:59 2022 daemon.notice netifd: wan (5159): sh: out of range
Sun Oct  9 19:35:59 2022 daemon.info pppd[5166]: Plugin pppoe.so loaded.
Sun Oct  9 19:35:59 2022 daemon.info pppd[5166]: PPPoE plugin from pppd 2.4.9
Sun Oct  9 19:35:59 2022 daemon.notice pppd[5166]: pppd 2.4.9 started by root, uid 0
Sun Oct  9 19:36:01 2022 daemon.info procd: - init complete -
Sun Oct  9 19:36:05 2022 daemon.info pppd[5166]: PPP session is 64442
Sun Oct  9 19:36:05 2022 daemon.warn pppd[5166]: Connected to dc:ef:80:74:20:f1 via interface eth3
Sun Oct  9 19:36:05 2022 kern.info kernel: [   55.586870] pppoe-wan: renamed from ppp0
Sun Oct  9 19:36:05 2022 daemon.info pppd[5166]: Renamed interface ppp0 to pppoe-wan
Sun Oct  9 19:36:05 2022 daemon.info pppd[5166]: Using interface pppoe-wan
Sun Oct  9 19:36:05 2022 daemon.notice pppd[5166]: Connect: pppoe-wan <--> eth3
Sun Oct  9 19:36:07 2022 daemon.info pppd[5166]: syncppp not active
Sun Oct  9 19:36:07 2022 daemon.info pppd[5166]: Remote message: Authentication success,Welcome!
Sun Oct  9 19:36:07 2022 daemon.notice pppd[5166]: PAP authentication succeeded
Sun Oct  9 19:36:07 2022 daemon.notice pppd[5166]: peer from calling number DC:EF:80:74:20:F1 authorized
Sun Oct  9 19:36:07 2022 daemon.err pppd[5166]: IPv6 socket creation failed: Address family not supported by protocol
Sun Oct  9 19:36:07 2022 daemon.err pppd[5166]: IPv6 socket creation failed: Address family not supported by protocol
Sun Oct  9 19:36:07 2022 daemon.info : 14[KNL] interface pppoe-wan activated
Sun Oct  9 19:36:07 2022 daemon.info : 10[KNL] interface pppoe-wan deactivated
Sun Oct  9 19:36:07 2022 daemon.info : 12[KNL] 117.67.112.170 appeared on pppoe-wan
Sun Oct  9 19:36:07 2022 daemon.info : 13[KNL] 117.67.112.170 disappeared from pppoe-wan
Sun Oct  9 19:36:07 2022 daemon.notice pppd[5166]: local  IP address 117.67.112.170
Sun Oct  9 19:36:07 2022 daemon.notice pppd[5166]: remote IP address 117.67.112.1
Sun Oct  9 19:36:07 2022 daemon.notice pppd[5166]: primary   DNS address 202.102.213.68
Sun Oct  9 19:36:07 2022 daemon.notice pppd[5166]: secondary DNS address 61.132.163.68
Sun Oct  9 19:36:07 2022 daemon.info avahi-daemon[2440]: Joining mDNS multicast group on interface pppoe-wan.IPv4 with address 117.67.112.170.
Sun Oct  9 19:36:07 2022 daemon.info avahi-daemon[2440]: New relevant interface pppoe-wan.IPv4 for mDNS.
Sun Oct  9 19:36:07 2022 daemon.info avahi-daemon[2440]: Registering new address record for 117.67.112.170 on pppoe-wan.IPv4.
Sun Oct  9 19:36:07 2022 daemon.info : 08[KNL] 117.67.112.170 appeared on pppoe-wan
Sun Oct  9 19:36:07 2022 daemon.info : 14[KNL] interface pppoe-wan activated
Sun Oct  9 19:36:08 2022 daemon.notice netifd: Network device 'pppoe-wan' link is up
Sun Oct  9 19:36:08 2022 daemon.notice netifd: Interface 'wan' is now up
Sun Oct  9 19:36:08 2022 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Sun Oct  9 19:36:08 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan (pppoe-wan)
Sun Oct  9 19:36:09 2022 user.notice ddns-scripts[5526]: myddns_ipv4: PID '5526' started at 2022-10-09 19:36
Sun Oct  9 19:36:12 2022 daemon.err nmbd[4347]: [2022/10/09 19:36:12.580003,  0] ../../source3/nmbd/nmbd_become_lmb.c:532(become_local_master_browser)
Sun Oct  9 19:36:12 2022 daemon.err nmbd[4347]:   become_local_master_browser: Error - cannot find server OPENWRT-J1900 in workgroup WORKGROUP on subnet 192.168.100.1
Sun Oct  9 19:38:36 2022 daemon.notice ttyd[5010]: HTTP / - 192.168.100.220
Sun Oct  9 19:38:37 2022 daemon.notice ttyd[5010]: HTTP /token - 192.168.100.220
Sun Oct  9 19:38:37 2022 daemon.notice ttyd[5010]: WS   /ws - 192.168.100.220, clients: 1
[... 19 lines cut ...]
Sun Oct  9 20:12:07 2022 daemon.err nmbd[4347]:   become_local_master_browser: Error - cannot find server OPENWRT-J1900 in workgroup WORKGROUP on subnet 192.168.100.1
Sun Oct  9 20:16:19 2022 user.warn ddns-scripts[5526]: myddns_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Sun Oct  9 20:17:12 2022 daemon.err nmbd[4347]: [2022/10/09 20:17:12.937077,  0] ../../source3/nmbd/nmbd_become_lmb.c:532(become_local_master_browser)
Sun Oct  9 20:17:12 2022 daemon.err nmbd[4347]:   become_local_master_browser: Error - cannot find server OPENWRT-J1900 in workgroup WORKGROUP on subnet 192.168.100.1
Sun Oct  9 20:22:12 2022 daemon.err nmbd[4347]: [2022/10/09 20:22:12.445038,  0] ../../source3/nmbd/nmbd_become_lmb.c:532(become_local_master_browser)
Sun Oct  9 20:22:12 2022 daemon.err nmbd[4347]:   become_local_master_browser: Error - cannot find server OPENWRT-J1900 in workgroup WORKGROUP on subnet 192.168.100.1
Sun Oct  9 20:26:23 2022 user.warn ddns-scripts[5526]: myddns_ipv4: Updating IP at DDNS provider failed - starting retry 2/0
Sun Oct  9 20:27:17 2022 daemon.err nmbd[4347]: [2022/10/09 20:27:17.697422,  0] ../../source3/nmbd/nmbd_become_lmb.c:532(become_local_master_browser)
Sun Oct  9 20:27:17 2022 daemon.err nmbd[4347]:   become_local_master_browser: Error - cannot find server OPENWRT-J1900 in workgroup WORKGROUP on subnet 192.168.100.1
Sun Oct  9 20:32:22 2022 daemon.err nmbd[4347]: [2022/10/09 20:32:22.949099,  0] ../../source3/nmbd/nmbd_become_lmb.c:532(become_local_master_browser)
Sun Oct  9 20:32:22 2022 daemon.err nmbd[4347]:   become_local_master_browser: Error - cannot find server OPENWRT-J1900 in workgroup WORKGROUP on subnet 192.168.100.1
Sun Oct  9 20:37:22 2022 daemon.err nmbd[4347]: [2022/10/09 20:37:22.457192,  0] ../../source3/nmbd/nmbd_become_lmb.c:532(become_local_master_browser)
Sun Oct  9 20:37:22 2022 daemon.err nmbd[4347]:   become_local_master_browser: Error - cannot find server OPENWRT-J1900 in workgroup WORKGROUP on subnet 192.168.100.1
Sun Oct  9 20:42:27 2022 daemon.err nmbd[4347]: [2022/10/09 20:42:27.709153,  0] ../../source3/nmbd/nmbd_become_lmb.c:532(become_local_master_browser)
Sun Oct  9 20:42:27 2022 daemon.err nmbd[4347]:   become_local_master_browser: Error - cannot find server OPENWRT-J1900 in workgroup WORKGROUP on subnet 192.168.100.1
Sun Oct  9 20:47:32 2022 daemon.err nmbd[4347]: [2022/10/09 20:47:32.961428,  0] ../../source3/nmbd/nmbd_become_lmb.c:532(become_local_master_browser)
Sun Oct  9 20:47:32 2022 daemon.err nmbd[4347]:   become_local_master_browser: Error - cannot find server OPENWRT-J1900 in workgroup WORKGROUP on subnet 192.168.100.1
Sun Oct  9 20:52:32 2022 daemon.err nmbd[4347]: [2022/10/09 20:52:32.469133,  0] ../../source3/nmbd/nmbd_become_lmb.c:532(become_local_master_browser)
Sun Oct  9 20:52:32 2022 daemon.err nmbd[4347]:   become_local_master_browser: Error - cannot find server OPENWRT-J1900 in workgroup WORKGROUP on subnet 192.168.100.1
Sun Oct  9 20:57:37 2022 daemon.err nmbd[4347]: [2022/10/09 20:57:37.721172,  0] ../../source3/nmbd/nmbd_become_lmb.c:532(become_local_master_browser)
Sun Oct  9 20:57:37 2022 daemon.err nmbd[4347]:   become_local_master_browser: Error - cannot find server OPENWRT-J1900 in workgroup WORKGROUP on subnet 192.168.100.1
Sun Oct  9 21:02:42 2022 daemon.err nmbd[4347]: [2022/10/09 21:02:42.973139,  0] ../../source3/nmbd/nmbd_become_lmb.c:532(become_local_master_browser)
Sun Oct  9 21:02:42 2022 daemon.err nmbd[4347]:   become_local_master_browser: Error - cannot find server OPENWRT-J1900 in workgroup WORKGROUP on subnet 192.168.100.1
Sun Oct  9 21:07:42 2022 daemon.err nmbd[4347]: [2022/10/09 21:07:42.481189,  0] ../../source3/nmbd/nmbd_become_lmb.c:532(become_local_master_browser)
Sun Oct  9 21:07:42 2022 daemon.err nmbd[4347]:   become_local_master_browser: Error - cannot find server OPENWRT-J1900 in workgroup WORKGROUP on subnet 192.168.100.1

ecrasy, 2022-10-10:
这是samba4的服务日志
正常的是这样的：

> Sun` Oct  9 21:46:55 2022 daemon.err nmbd[4694]: [2022/10/09 21:46:55.370215,  0] ../../source3/nmbd/nmbd_become_lmb.c:398(become_local_master_stage2)
Sun Oct  9 21:46:55 2022 daemon.err nmbd[4694]:   *****
Sun Oct  9 21:46:55 2022 daemon.err nmbd[4694]:
Sun Oct  9 21:46:55 2022 daemon.err nmbd[4694]:   Samba name server OPENWRT is now a local master browser for workgroup WORKGROUP on subnet 192.168.2.1
Sun Oct  9 21:46:55 2022 daemon.err nmbd[4694]:
Sun Oct  9 21:46:55 2022 daemon.err nmbd[4694]:   *****

你是不是配置错了网络？

drw2005, 2022-10-10:
> 这是samba4的服务日志 正常的是这样的：
>
>
> 你是不是配置错了网络？

谢谢回复，问题是我没有勾选这个插件，怎么会出现这个错误呢。

mitsukileung, 2022-10-10:
X86默认是勾选了SAMBA的呀

mitsukileung, 2022-10-10:
> 谢谢回复，问题是我没有勾选这个插件，怎么会出现这个错误呢。

X86默认是勾选了SAMBA的呀

drw2005, 2022-10-10:
> X86默认是勾选了SAMBA的呀

编译时我去掉了这个勾选。

ecrasy, 2022-10-10:
Extra packages  --->里面的autosamba是依赖samba4的
而x86是默认勾选了autosamba的
你只有在Extra packages  --->里去掉了autosamba才能去掉samba4
samba和samba4是两个类似功能的不同组件。

drw2005, 2022-10-10:
> Extra packages --->里面的autosamba是依赖samba4的 而x86是默认勾选了autosamba的 你只有在Extra packages --->里去掉了autosamba才能去掉samba4 samba和samba4是两个类似功能的不同组件。

是的，这个我知道，正是这样去掉的，从来没勾选过这个插件。

ifurther, 2022-10-18:
請問您有檢查.config檔案

closed 2022-10-28 by drw2005