coolsnowwolf/lede #10248: orangepi-r1-plus-lts编译到uboot失败。

QYKing233, 2022-10-12:
反馈bug/问题模板，提建议请删除

## 1.orangepi-r1-plus-lts编译到uboot失败

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

[uboot-rockchipremove prebuilt of-platdata：](https://github.com/coolsnowwolf/lede/pull/9747#issue-1303074384)
天灵提交修改后编译orangepi-r1-plus-lts失败。

### (2) 路由器型号和固件版本

A：orangepi-r1-plus-lts，2022/10/12主线版本

### (3) 详细日志

A：drivers/ram/rockchip/sdram_rk3328.c: In function 'rk3328_dmc_init':
drivers/ram/rockchip/sdram_rk3328.c:525:55: error: 'struct dtd_rockchip_rk3328_dmc' has no member named 'rockchip_sdram_params'
  525 |                                         (void *)dtplat->rockchip_sdram_params;
      |                                                       ^~
make[8]: *** [scripts/Makefile.build:258: tpl/drivers/ram/rockchip/sdram_rk3328.o] Error 1
make[7]: *** [scripts/Makefile.build:398: tpl/drivers/ram/rockchip] Error 2
make[6]: *** [scripts/Makefile.build:398: tpl/drivers/ram] Error 2
make[5]: *** [scripts/Makefile.spl:531: tpl/drivers] Error 2
make[4]: *** [Makefile:2109: tpl/u-boot-tpl] Error 2
make[4]: Leaving directory '/media/king/Datas/lts/build_dir/target-aarch64_generic_musl/u-boot-orangepi-r1-plus-lts-rk3328/u-boot-2022.07'
make[3]: *** [Makefile:247: /media/king/Datas/lts/build_dir/target-aarch64_generic_musl/u-boot-orangepi-r1-plus-lts-rk3328/u-boot-2022.07/.built] Error 2
make[3]: Leaving directory '/media/king/Datas/lts/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/orangepi-r1-plus-lts-rk3328/compile#2.74#0.88#44.16
    ERROR: package/boot/uboot-rockchip failed to build (build variant: orangepi-r1-plus-lts-rk3328).
make[2]: *** [package/Makefile:116: package/boot/uboot-rockchip/compile] Error 1
make[2]: Leaving directory '/media/king/Datas/lts'
make[1]: *** [package/Makefile:110: /media/king/Datas/lts/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/media/king/Datas/lts'
make: *** [/media/king/Datas/lts/include/toplevel.mk:230：world] 错误 2

coolsnowwolf, 2022-10-13:
sudo apt update -y
sudo apt full-upgrade -y
sudo apt install -y ack antlr3 aria2 asciidoc autoconf automake autopoint binutils bison build-essential \
bzip2 ccache cmake cpio curl device-tree-compiler fastjar flex gawk gettext gcc-multilib g++-multilib \
git gperf haveged help2man intltool libc6-dev-i386 libelf-dev libglib2.0-dev libgmp3-dev libltdl-dev \
libmpc-dev libmpfr-dev libncurses5-dev libncursesw5-dev libreadline-dev libssl-dev libtool lrzsz \
mkisofs msmtp nano ninja-build p7zip p7zip-full patch pkgconf python2.7 python3 python3-pip libpython3-dev qemu-utils \
rsync scons squashfs-tools subversion swig texinfo uglifyjs upx-ucl unzip vim wget xmlto xxd zlib1g-dev

QYKing233, 2022-10-13:
重新安装依赖后拉取主线版本编译出现同issue中同样的错误。
20220505--R22.5.5-x64-Release稳定版编译成功。

hong0980, 2022-10-30:
> 用20220505版本5.10内核编译没问题，先别用主线版本。

commit是多少？

QYKing233, 2022-10-30:
> > 用20220505版本5.10内核编译没问题，先别用主线版本。
> 
> commit是多少？

如果你想编译的话，在tags找到稳定版的发行版，20220505那个日期。

hong0980, 2022-10-30:
> > > 用20220505版本5.10内核编译没问题，先别用主线版本。
> > 
> > 
> > commit是多少？
> 
> 如果你想编译的话，在tags找到稳定版的发行版，20220505那个日期。

谢谢！找到了，git reset --hard 3516c9aa8a3a19ed717bc888905f549dd7b98289

Leeleeyn, 2022-11-04:
用20220505版本编译成功，但yt8531网口不能工作，未找到解决方法。

QYKing233, 2022-11-04:
> 用20220505版本编译成功，但yt8531网口不能工作，未找到解决方法。

需要使用5.10内核，默认5.15内核yt8531确实有问题。另外我的仓库里有云编译的固件，在action可以试试。

closed 2022-12-13 by aiamadeus