coolsnowwolf/lede #10249: X86 出现daemon.err uhttpd[2999]: sh: out of range

rrkai, 2022-10-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

X86 5.10.147 R22.10.10

### (3) 详细日志

Thu Oct 13 00:59:58 2022 daemon.notice netifd: Network device 'pppoe-wan' link is up
Thu Oct 13 00:59:58 2022 daemon.notice netifd: Interface 'wan' is now up
Thu Oct 13 00:59:58 2022 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Thu Oct 13 00:59:58 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan (pppoe-wan)
Thu Oct 13 00:59:58 2022 daemon.err miniupnpd[5491]: could not open lease file: /var/upnp.leases
Thu Oct 13 00:59:58 2022 daemon.notice miniupnpd[5491]: HTTP listening on port 5000
Thu Oct 13 00:59:58 2022 daemon.notice miniupnpd[5491]: Listening for NAT-PMP/PCP traffic on port 5351
Thu Oct 13 01:00:00 2022 cron.err crond[2804]: USER root pid 5612 cmd echo "" > /tmp/vssr.log
Thu Oct 13 01:00:01 2022 daemon.notice netifd: Network device 'eth1' link is down
Thu Oct 13 01:00:01 2022 kern.info kernel: [   52.172319] igb 0000:02:00.0 eth1: igb: eth1 NIC Link is Down
Thu Oct 13 01:00:01 2022 kern.info kernel: [   52.174962] br-lan: port 2(eth1) entered disabled state
Thu Oct 13 01:00:04 2022 kern.info kernel: [   55.457327] igb 0000:02:00.0 eth1: igb: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
Thu Oct 13 01:00:05 2022 daemon.notice netifd: Network device 'eth1' link is up
Thu Oct 13 01:00:05 2022 kern.info kernel: [   55.706582] br-lan: port 2(eth1) entered blocking state
Thu Oct 13 01:00:05 2022 kern.info kernel: [   55.710763] br-lan: port 2(eth1) entered forwarding state
Thu Oct 13 01:02:57 2022 daemon.err uhttpd[2999]: sh: out of range
Thu Oct 13 01:02:57 2022 daemon.err uhttpd[2999]: sh: out of range

换5.15内核就不会.配置文件一样

l4326769, 2024-10-21:
arm 6.6内核了，跑了半个月也出现这个问题，重启解决了，就怕什么时候再出现

coolsnowwolf, 2026-09-05:
已核对当前 LEDE：uhttpd 核心及 LuCI 源码中没有该日志对应的固定错误路径；仓库内也没有 issue 日志中的 vssr.log 生成脚本。原始日志同时出现第三方 vssr 定时任务（cron 写入 /tmp/vssr.log），且缺少触发 URL、CGI/脚本内容和完整回溯，无法归因于 LEDE 当前源码或提出可验证的修复。若仍可复现，请补充请求路径、对应 CGI/脚本、完整 uhttpd 日志和最小配置。基于现有证据关闭为 not planned。

closed 2026-09-05 by coolsnowwolf