coolsnowwolf/lede #10251: R4S 状态/概览显示不全

ZZMHA227, 2022-10-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
10月13日编译的R4S固件。如果做旁路由，删掉WAN口，则概览显示不全。再添加WAN口后则正常。看到之前的issues里，R2S也有这样的问题
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
[image]
[image]
[image]
[image]

### (2) 路由器型号和固件版本

A：R4S（R2S也有这可能)

### (3) 详细日志

A：

coolsnowwolf, 2022-10-13:
没必要删除 WAN 口，又不妨碍你

felixxiao99, 2022-10-13:
> 

兄弟，你也一样啊

edmondfr, 2022-10-18:
这个问题搞得旁路由强迫症患者茶饭不思，夜不能寐

closed 2022-10-25 by aiamadeus