coolsnowwolf/lede #10253: python3 host 编译失败

ecrasy, 2022-10-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
python3 host 编译失败

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

最近两天编译发现这个编译错误。
对比官方源码v22.03.1的编译log，  
使用差不多的config文件，  
官方v22.03.1没有这个python3/host这个编译。   

仔细查看config文件，发现并没有勾选python3。
不知为何LEDE源码会编译这个python3/host，
并且编译出现错误无法通过。  

### (2) 路由器型号和固件版本

A：x86_64  
B: 树莓派4B  

### (3) 详细日志
2022-10-14T01:25:06.2931113Z building '_hashlib' extension
2022-10-14T01:25:06.2938727Z gcc -pthread -fPIC -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -O2 -I/home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/staging_dir/host/include -I/home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/staging_dir/hostpkg/include -I/home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/staging_dir/target-aarch64_cortex-a72_musl/host/include -O2 -I/home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/staging_dir/host/include -I/home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/staging_dir/hostpkg/include -I/home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/staging_dir/target-aarch64_cortex-a72_musl/host/include -std=c99 -Wextra -Wno-unused-result -Wno-unused-parameter -Wno-missing-field-initializers -Werror=implicit-function-declaration -fvisibility=hidden -I./Include/internal -I/home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/staging_dir/hostpkg/include -I./Include -I. -I/home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/staging_dir/host/include -I/home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/staging_dir/hostpkg/include -I/usr/include/x86_64-linux-gnu -I/usr/local/include -I/home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/build_dir/hostpkg/Python-3.9.12/Include -I/home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/build_dir/hostpkg/Python-3.9.12 -c /home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/build_dir/hostpkg/Python-3.9.12/Modules/_hashopenssl.c -o build/temp.linux-x86_64-3.9/home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/build_dir/hostpkg/Python-3.9.12/Modules/_hashopenssl.o
2022-10-14T01:25:06.4001524Z In file included from /home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/staging_dir/host/include/openssl/bio.h:69,
2022-10-14T01:25:06.4002188Z                  from /home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/staging_dir/host/include/openssl/evp.h:67,
2022-10-14T01:25:06.4003197Z                  from /home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/build_dir/hostpkg/Python-3.9.12/Modules/_hashopenssl.c:33:
2022-10-14T01:25:06.4004008Z /home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/build_dir/hostpkg/Python-3.9.12/Modules/_hashopenssl.c: In function 'HMAC_CTX_new':
2022-10-14T01:25:06.4006841Z /home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/build_dir/hostpkg/Python-3.9.12/Modules/_hashopenssl.c:54:43: error: invalid application of 'sizeof' to incomplete type 'HMAC_CTX' {aka 'struct hmac_ctx_st'}
2022-10-14T01:25:06.4007487Z    54 |     HMAC_CTX *ctx = OPENSSL_malloc(sizeof(HMAC_CTX));
2022-10-14T01:25:06.4007819Z       |                                           ^~~~~~~~
2022-10-14T01:25:06.4008502Z /home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/staging_dir/host/include/openssl/crypto.h:303:48: note: in definition of macro 'OPENSSL_malloc'
2022-10-14T01:25:06.4009031Z   303 | #define OPENSSL_malloc(num) CRYPTO_malloc((int)num,NULL,0)
2022-10-14T01:25:06.4009377Z       |                                                ^~~
2022-10-14T01:25:06.4012425Z /home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/build_dir/hostpkg/Python-3.9.12/Modules/_hashopenssl.c:56:31: error: invalid application of 'sizeof' to incomplete type 'HMAC_CTX' {aka 'struct hmac_ctx_st'}
2022-10-14T01:25:06.4013009Z    56 |         memset(ctx, 0, sizeof(HMAC_CTX));
2022-10-14T01:25:06.4013310Z       |                               ^~~~~~~~
2022-10-14T01:25:06.4049644Z /home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/build_dir/hostpkg/Python-3.9.12/Modules/_hashopenssl.c:57:9: error: implicit declaration of function 'HMAC_CTX_init'; did you mean 'HMAC_CTX_new'? [-Werror=implicit-function-declaration]
2022-10-14T01:25:06.4050340Z    57 |         HMAC_CTX_init(ctx);
2022-10-14T01:25:06.4050587Z       |         ^~~~~~~~~~~~~
2022-10-14T01:25:06.4050842Z       |         HMAC_CTX_new
2022-10-14T01:25:06.4136532Z /home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/build_dir/hostpkg/Python-3.9.12/Modules/_hashopenssl.c: In function 'HMAC_CTX_free':
2022-10-14T01:25:06.4138072Z /home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/build_dir/hostpkg/Python-3.9.12/Modules/_hashopenssl.c:66:9: error: implicit declaration of function 'HMAC_CTX_cleanup' [-Werror=implicit-function-declaration]
2022-10-14T01:25:06.4139304Z    66 |         HMAC_CTX_cleanup(ctx);
2022-10-14T01:25:06.4139782Z       |         ^~~~~~~~~~~~~~~~
2022-10-14T01:25:06.4141423Z /home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/build_dir/hostpkg/Python-3.9.12/Modules/_hashopenssl.c: In function 'HMAC_CTX_get_md':
2022-10-14T01:25:06.4142896Z /home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/build_dir/hostpkg/Python-3.9.12/Modules/_hashopenssl.c:74:15: error: dereferencing pointer to incomplete type 'HMAC_CTX' {aka 'const struct hmac_ctx_st'}
2022-10-14T01:25:06.4144010Z    74 |     return ctx->md;
2022-10-14T01:25:06.4144966Z       |               ^~
2022-10-14T01:25:06.4257791Z /home/runner/work/BuildOpenWrt/BuildOpenWrt/workspace/openwrt/build_dir/hostpkg/Python-3.9.12/Modules/_hashopenssl.c:75:1: warning: control reaches end of non-void function [-Wreturn-type]
2022-10-14T01:25:06.4258656Z    75 | }
2022-10-14T01:25:06.4259330Z       | ^
2022-10-14T01:25:06.4334314Z cc1: some warnings being treated as errors
[1_Build.txt](https://github.com/coolsnowwolf/lede/files/9781834/1_Build.txt)

ecrasy, 2022-10-14:
补充一下，
从我三个多月前接触编译openwrt开始，
到现在，
这个默认编译的包臃肿了很多，
单单是编译出来的toolchain就比以前大了400+M。
而这个python/host的编译是不是没必要加进去的？
官方源码v22.03.1编译差不多的config文件，
编译LOG里没有看到python/host的编译。
这玩意又占编译时间又占空间，
对于actions编译很不友好。

binge8, 2022-10-14:
如果你的编译环境还是用的这个
apt-get -qq install $(curl -fsSL git.io/depends-ubuntu-2004)
这已经过时了，就会报这个错误，用大雕仓库里的不就好了？
apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs gcc-multilib g++-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler antlr3 gperf swig

ecrasy, 2022-10-14:
我用的L大源码readme的
我先试下你的看看

closed 2022-10-17 by ecrasy

ecrasy, 2022-10-19:
python3/host 编译失败的原因是
platform.py 解析version字符串的时候失败了
触发了异常
打个补丁就好了
[lib-platform-sys-version.zip](https://github.com/coolsnowwolf/lede/files/9819890/lib-platform-sys-version.zip)