coolsnowwolf/lede #10256: ubuntu20.04本地编译出错，请大佬帮助解读一下看看有没有解决办法，感谢。。。

drw2005, 2022-10-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志
Enabling sysfsutils
Enabling sysntpd
Enabling system
Enabling ttyd
Enabling turboacc
Enabling turboacc_fss
Enabling ucitrack
Enabling uhttpd
Enabling umount
Enabling urandom_seed
Enabling urngd
Enabling wpad
sed -i "s/Installed-Time: .*/Installed-Time: 1664995310/" /home/drw/openwrt/build_dir/target-i386_pentium4_musl/root-x86/usr/lib/opkg/status
rm -rf /home/drw/openwrt/build_dir/target-i386_pentium4_musl/root-x86/boot /home/drw/openwrt/build_dir/target-i386_pentium4_musl/root-x86/tmp/* /home/drw/openwrt/build_dir/target-i386_pentium4_musl/root-x86/usr/lib/opkg/info/*.postinst* /home/drw/openwrt/build_dir/target-i386_pentium4_musl/root-x86/usr/lib/opkg/lists/* /home/drw/openwrt/build_dir/target-i386_pentium4_musl/root-x86/var/lock/*.lock
find /home/drw/openwrt/build_dir/target-i386_pentium4_musl/root-x86/ -mindepth 1 -execdir touch -hcd "@1664995310" "{}" +
find: The relative path 'Files' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find.  Please remove that entry from $PATH
make[2]: *** [package/Makefile:73: package/install] Error 1
make[2]: Leaving directory '/home/drw/openwrt'
make[1]: *** [package/Makefile:111: /home/drw/openwrt/staging_dir/target-i386_pentium4_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/drw/openwrt'
make: *** [/home/drw/openwrt/include/toplevel.mk:230: world] Error 2

hcym, 2022-10-15:
2004肯定可以，无非是插件和什么下载问题
2204和deb11都可以的，现在系统的兼容性相当好

reputati0n, 2022-10-15:
关闭 Libraries->SSL->libmbedtls->Enable use of the ARMv8 Crypto Extensions
我不知道具体是哪出了问题，但是我跟你一样的报错关了之后就编译成功了

aiamadeus, 2022-10-15:
`Please remove that entry from $PATH` 你就不会百度一下

closed 2022-10-15 by aiamadeus