coolsnowwolf/lede #10261: x86编译失败

xnxy2012, 2022-10-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：5.15内科可以编译完成，5.19及6.0编译失败，相同的错误日志

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：
make[3]: *** [modules/video.mk:1127: /workdir/openwrt/bin/targets/x86/64/packages/kmod-drm-i915_6.0.2-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/workdir/openwrt/package/kernel/linux'
time: package/kernel/linux/compile#2.50#0.15#2.64
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2

xiangfeidexiaohuo, 2022-10-16:
重新拉配置文件，我编译602的x86，都用了好久，没问题

xnxy2012, 2022-10-16:
> 重新拉配置文件，我编译602的x86，都用了好久，没问题

我试试，之前5.19一直很好，就是最近一个版本更新的时候出现的，然后6.0也是一样

hcym, 2022-10-16:
根本没有什么区别，估计十五代cpu有提高

closed 2022-10-17 by coolsnowwolf