coolsnowwolf/lede #10266: socat is different from the conffile in the new package

kl9800, 2022-10-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述
* resolve_conffiles: 
Existing conffile 
/home/FF/lede/build_dir/target-x86_64_musl/root-x86/etc/config/socat is different from the conffile in the new package. 
The new conffile will be placed at /home/FF/lede/build_dir/target-x86_64_musl/root-x86/etc/config/socat-opkg.

### (1) 具体问题
编译时候出现冲突，之前看到过有相关提问，但是没看懂怎么弄，本人小白，求个具体解决办法，例如要编辑或者修改哪些文件之类的具体步骤，多谢大佬们
A：

### (2) 路由器型号和固件版本

A：J4105

### (3) 详细日志

A：

ecrasy, 2022-10-18:
这是配置文件的路径有2个
新的会被placed at ……
这没提示是错误吧？
我路由现在跑的系统：

> root@OpenWrt:~# ls /etc/config/socat*
/etc/config/socat       /etc/config/socat-opkg

kl9800, 2022-10-18:
可以正常编译，但是，IPV6只能获取IP，实际不能用，测试IPV6的时候提示无法使用IPV6

kl9800, 2022-10-18:
而且也提示了error，只是我没贴上来。。