coolsnowwolf/lede #10267: luci “网络-接口”页面缺失“设备”标签页

lengran, 2022-10-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

luci 后台，网络-接口，下面没有“设备”页。请问是虚拟机环境设置问题，还是固件编译的时候漏选了什么组件？

### (2) 路由器型号和固件版本

1. x86-64，自己编译 PVE 环境，git-22.279.28429-0cb5c5c
2. x86-64，VBox 环境，release 页下载的 openwrt-x64-R22.10.1

### (3) 详细日志

A：无

NothingNA, 2022-10-17:
啥意思？在wan或者lan点修改里物理设置么？lede的luci不是那个新版本的，不是设备那个地方

不知道你说的是不是这个意思 好久没用那个新的luci了 我印象里设备选项是改这个的……

lengran, 2022-10-17:
> 

不完全一样，“物理设置”的选项卡可以调的比“设备”那里可调的要少一些。我是在配置 VLAN 的时候发现的。试过按 openwrt 的文档直接编辑 /etc/config/network，但是刷新页面再看会有 “不支持的协议类型”，“修改”按钮也是无法点击的状态。如果能有办法编译 luci 到支持“设备”设置的版本会比较方便。

coolsnowwolf, 2022-10-17:
这个是老版本的 luci ，不是基于 dev 而是基于 ifname 的，所以没有 dev 的选项

lengran, 2022-10-17:
> 这个是老版本的 luci ，不是基于 dev 而是基于 ifname 的，所以没有 dev 的选项

您好！请问 VLAN 的配置页面有办法启用吗？openwrt 19 的老 luci 配置页面我也没有找到。谢谢！

aoxueos, 2022-11-11:
> > 这个是老版本的 luci ，不是基于 dev 而是基于 ifname 的，所以没有 dev 的选项
> 
> 您好！请问 VLAN 的配置页面有办法启用吗？openwrt 19 的老 luci 配置页面我也没有找到。谢谢！

直接设备后面加个点，eg. 在eth0上创建一个41vlan的设备 eth0.41
Add a point after the device you need, eg. eth0.41 for user the vlan 41 on eth0

lengran, 2022-11-11:
> 直接设备后面加个点，eg. 在eth0上创建一个41vlan的设备 eth0.41 Add a point after the device you need, eg. eth0.41 for user the vlan 41 on eth0

这样只能创建 vlan 接口但是不能给流量打标？给不同接口设置 tagged / untagged / - 的那个界面貌似没找到。

aoxueos, 2022-11-11:
> ifname

这些功能在基于ifname的旧版本luci上并不存在，请使用luci19.07或更高
There isn‘t this on the old luci by ifname， please use the luci19.07 or higher

lengran, 2022-11-11:
> 这些功能在基于ifname的旧版本luci上并不存在，请使用luci19.07或更高 There isn‘t this on the old luci by ifname， please use the luci19.07 or higher

openwrt 19 是包含这个功能的，在网络 - 交换机页面。是的，它不存在于 lede 使用的 luci 上。

aoxueos, 2022-11-11:
> > 这些功能在基于ifname的旧版本luci上并不存在，请使用luci19.07或更高 There isn‘t this on the old luci by ifname， please use the luci19.07 or higher
> 
> openwrt 19 是包含这个功能的，在网络 - 交换机页面。是的，它不存在于 lede 使用的 luci 上。

冷知识，Lede使用的是Luci 18.07 修补版
The trivia,Lede is based on Luci 18.07 fix and plus

closed 2026-09-05 by coolsnowwolf