coolsnowwolf/lede #10270: 5.15频率缺失问题，再次打扰

AXBNG, 2022-10-18:
之前的issue https://github.com/coolsnowwolf/lede/issues/10234 中，感谢大佬的回复以及排查修复，但以我重新编译的结果以及评论区反馈来看，cpu频率依旧缺失，不知大佬能否赐教问题所在，万分感谢~

此外@lovehackintosh 还反馈cpu频率似乎有问题https://github.com/coolsnowwolf/lede/issues/10234#issuecomment-1281632638，这点我倒是没有注意，大佬有空也请看下，辛苦~

xiangfeidexiaohuo, 2022-10-18:
你这是冒着生命危险 重开了这个问题啊。确实没修复，我这边也试过了，5.15.74  依旧无解。

lovehackintosh, 2022-10-18:
不显示频率是小事,更重要的是添加了mhz https://github.com/robimarko/openwrt/commit/30ee62ebab0fc45e4d14cac62fdc6cff91bac59a
发现cpu频率只有788mhz
```bash
root@OpenWrt:~# mhz
count=169252 us50=10587 us250=53503 diff=42916 cpu_MHz=788.759
root@OpenWrt:~# /etc/coremark.sh && cat /etc/bench.log 
(CpuMark : 10570.126181 
 Scores)
```

AXBNG, 2022-10-18:
> 不显示频率是小事,更重要的是添加了mhz [robimarko/openwrt@30ee62e](https://github.com/robimarko/openwrt/commit/30ee62ebab0fc45e4d14cac62fdc6cff91bac59a) 发现cpu频率只有788mhz
> 
> ```shell
> root@OpenWrt:~# mhz
> count=169252 us50=10587 us250=53503 diff=42916 cpu_MHz=788.759
> root@OpenWrt:~# /etc/coremark.sh && cat /etc/bench.log 
> (CpuMark : 10570.126181 
>  Scores)
> ```
> 
> 建议还是使用5.10内核吧，高版本内核的特性普通人用不到，QSDK11还在用4.4.60呢

目前来说也只能维持在5.10了..........  ┐(￣ヘ￣)┌

lovehackintosh, 2022-10-18:
你还可以使用这个分支https://github.com/robimarko/openwrt/tree/ipq807x-5.15-pr

mitsukileung, 2022-10-18:
5.10还是不错的，觉得挺稳定

AXBNG, 2022-10-18:
> 你还可以使用这个分支https://github.com/robimarko/openwrt/tree/ipq807x-5.15-pr

算了添加插件又稍麻烦，如你所说其实5.15和5.10的差别感知不强，暂时不折腾了.......    乁(ᴗ ͜ʖ ᴗ)ㄏ

AXBNG, 2022-10-18:
> 5.10还是不错的，觉得挺稳定

系的5.10用着一直也蛮舒服........   ┌|ﾟзﾟ|┘

coolsnowwolf, 2022-10-18:
5.15 的 NSS ECM 还是有些小问题，暂时先不 PR 了

AXBNG, 2022-10-18:
> 5.15 的 NSS ECM 还是有些小问题，暂时先不 PR 了

soga，哈哈，坐等大佬修复（小白永远在坐等...........    ∠( ᐛ 」∠)＿

closed 2022-10-18 by AXBNG