coolsnowwolf/lede #10283: 最新源码make defconfig 失败

ecrasy, 2022-10-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
最新源码make defconfig 失败
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
### (1) 具体问题
源码无配置
直接命令make defconfig 失败

A：最新源码make defconfig 失败

### (2) 路由器型号和固件版本

A：无

### (3) 详细日志

A：Collecting package info: done
Collecting target info: done
Config.in:9: syntax error
Config.in:8: invalid statement
Config.in:9: invalid statement
Config.in:10: invalid statement
make: *** [/home/carbon/code/openwrt/include/toplevel.mk:120：defconfig] 错误 1

mitsukileung, 2022-10-19:
一样的问题+1

coolsnowwolf, 2022-10-19:
index 需要重新生成缓存

rm -rf ./tmp && rm -rf ./feeds && rm -rf ./pacakges/feeds

重新运行

mitsukileung, 2022-10-19:
> index 需要重新生成缓存
> 
> rm -rf ./tmp && rm -rf ./feeds && rm -rf ./pacakges/feeds
> 
> 重新运行

执行完上面的命令，重新运行后还是出现错误

Config.in:9: syntax error
Config.in:8: invalid statement
Config.in:9: invalid statement
Config.in:10: invalid statement
can't open file 'tmp/.diffconfig.stage1' at ./scripts/kconfig.pl line 32.
Config.in:9: syntax error
Config.in:8: invalid statement
Config.in:9: invalid statement
Config.in:10: invalid statement
can't open file 'tmp/.diffconfig.stage2' at ./scripts/kconfig.pl line 32

lovehackintosh, 2022-10-20:
重新生成缓存后还需要
```bash
./scripts/feeds update -a
./scripts/feeds install -a
```
如果还不行的话,删除重新clone，亲测可以成功。

ecrasy, 2022-10-20:
重新clone之后
测试：
./scripts/feeds update -a  
./scripts/feeds install -a  
再：
make defconfig  
通过

closed 2022-10-20 by ecrasy