coolsnowwolf/lede #10284: 同步最新源码后，保存配置出现以下错误

mitsukileung, 2022-10-19:
X86_64

执行 rm -rf ./tmp && rm -rf .config
        git pull
        ./scripts/feeds update -a && ./scripts/feeds install -a
        make menuconfig 配置完后
        ./scripts/diffconfig.sh > 64 保存配置时出现以下错误

Config.in:9: syntax error
Config.in:8: invalid statement
Config.in:9: invalid statement
Config.in:10: invalid statement
Config.in:9: syntax error
Config.in:8: invalid statement
Config.in:9: invalid statement
Config.in:10: invalid statement

coolsnowwolf, 2022-10-19:
index 需要重新生成缓存

rm -rf ./tmp && rm -rf ./feeds && rm -rf ./pacakges/feeds

重新运行

mitsukileung, 2022-10-19:
> index 需要重新生成缓存
> 
> rm -rf ./tmp && rm -rf ./feeds && rm -rf ./pacakges/feeds
> 
> 重新运行

还是会出错

执行完上面的命令，重新运行后还是出现错误

Config.in:9: syntax error
Config.in:8: invalid statement
Config.in:9: invalid statement
Config.in:10: invalid statement
can't open file 'tmp/.diffconfig.stage1' at ./scripts/kconfig.pl line 32.
Config.in:9: syntax error
Config.in:8: invalid statement
Config.in:9: invalid statement
Config.in:10: invalid statement
can't open file 'tmp/.diffconfig.stage2' at ./scripts/kconfig.pl line 32

lovehackintosh, 2022-10-20:
重新生成缓存后还需要
```bash
./scripts/feeds update -a
./scripts/feeds install -a
```
如果还不行的话,删除重新clone，亲测可以成功。

closed 2022-10-20 by mitsukileung

bi4tms, 2022-10-20:
> > 重新生成缓存后还需要
> > ```shell
> > ./scripts/feeds update -a
> > ./scripts/feeds install -a
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 如果还不行的话,删除重新clone，亲测可以成功。
> 
> 重clone后可以了，谢谢

请教下如何重新clone