coolsnowwolf/lede #10286: h68k-f 刷入自己编译固件无法启动

zhangguanzhang, 2022-10-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

```
CONFIG_TARGET_rockchip=y
CONFIG_TARGET_rockchip_armv8=y

CONFIG_TARGET_DEVICE_rockchip_armv8_DEVICE_hinlink_opc-h68k=y

CONFIG_TARGET_KERNEL_PARTSIZE=18
CONFIG_TARGET_ROOTFS_PARTSIZE=656
# CONFIG_TARGET_ROOTFS_TARGZ is not set
CONFIG_TARGET_ROOTFS_EXT4FS=y
CONFIG_TARGET_ROOTFS_SQUASHFS=y

```

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

TendernessRadif, 2022-10-20:
EMMC启动 群禁言了 直接EMMC刷入且启动就行了

aiamadeus, 2022-10-20:
emmc 有固件的话要先格式化掉，问就是 rk 的启不动主线 uboot（sd 卡正常启动如下图：
[image]

closed 2022-10-20 by aiamadeus