coolsnowwolf/lede #10287: Config.in 更新后，两次make defconfig触发错误

zhangguanzhang, 2022-10-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

https://github.com/coolsnowwolf/lede/commit/66d86d23716fc71e71a459f5352eee3a9caf749b#diff-a17fb9f46d91e0e8f467f63ac557a1bafa8f4b98c2d85b226a3f7179108d36c3L8
这个提交的第八行的 option 删除后
两次 make defconfig 就会报错
```
Config.in:9: syntax error
Config.in:8: invalid statement
Config.in:9: invalid statement
Config.in:10: invalid statement
make: *** [/home/guanzhang/lede/include/toplevel.mk:120: defconfig] Error 1
```

mitsukileung, 2022-10-20:
重CLONE后没问题

zhangguanzhang, 2022-10-20:
> 重CLONE后没问题

感谢，试了下是这样的

closed 2022-10-20 by zhangguanzhang

bi4tms, 2022-10-20:
> 重CLONE后没问题

请教下如何重clone

lovehackintosh, 2022-10-20:
[就是本仓库README的编译命令](https://github.com/coolsnowwolf/lede#%E7%BC%96%E8%AF%91%E5%91%BD%E4%BB%A4) 
   ```bash
   git clone https://github.com/coolsnowwolf/lede
   cd lede
   ./scripts/feeds update -a
   ./scripts/feeds install -a
   make menuconfig
   ```