coolsnowwolf/lede #10288: 拉取最新源码后make defconfig无法通过

bi4tms, 2022-10-20:
make clean后拉取最新源码，然后rm -rf ./tmp && rm -rf ./feeds && rm -rf ./pacakges/feeds，接着./scripts/feeds update -a && ./scripts/feeds install -a，最后make defconfig依旧不通过，提示：
Collecting target info: done
Config.in:9: syntax error
Config.in:8: invalid statement
Config.in:9: invalid statement
Config.in:10: invalid statement

lovehackintosh, 2022-10-20:
删掉重新clone,亲测可行

bi4tms, 2022-10-20:
> 删掉重新clone,亲测可行

请教下，删掉哪些内容，如何重新克隆

bi4tms, 2022-10-20:
> 删掉重新clone,亲测可行

重新clone是相当于删掉项目然后重建吗

lovehackintosh, 2022-10-20:
对，就是删除lede文件夹，然后
   ```bash
   git clone https://github.com/coolsnowwolf/lede
   cd lede
   ./scripts/feeds update -a
   ./scripts/feeds install -a
   make menuconfig
   ```

zhangguanzhang, 2022-10-20:
https://github.com/coolsnowwolf/lede/issues/10287
重复issue

bi4tms, 2022-10-20:
> 对，就是删除lede文件夹，然后
> 
> ```shell
> git clone https://github.com/coolsnowwolf/lede
> cd lede
> ./scripts/feeds update -a
> ./scripts/feeds install -a
> make menuconfig
> make download -j8
> make -j$(nproc)
> ```

感谢

closed 2022-10-20 by bi4tms