coolsnowwolf/lede #10293: x86编译出错

ookinasiawase, 2022-10-21:
反馈bug/问题模板，提建议请删除
```bash
checking whether mkfifo rejects trailing slashes... yes
checking whether mkfifoat rejects trailing slashes... yes
checking whether mknod can create fifo without root privileges... configure: error: in `/root/openwrt/lede/build_dir/host/tar-1.34':
configure: error: you should not run configure as root (set FORCE_UNSAFE_CONFIGURE=1 in environment to bypass this check)
See `config.log' for more details
make[3]: *** [Makefile:32: /root/openwrt/lede/build_dir/host/tar-1.34/.configured] Error 1
make[3]: Leaving directory '/root/openwrt/lede/tools/tar'
time: tools/tar/compile#12.71#4.42#20.69
    ERROR: tools/tar failed to build.
make[2]: *** [tools/Makefile:173: tools/tar/compile] Error 1
make[2]: Leaving directory '/root/openwrt/lede'
make[1]: *** [tools/Makefile:169: /root/openwrt/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnyyynynyyyyyyynyynnyynynyyyyyyyyyyyyyyyyyynynnyyyyyyyyyn] Error 2
make[1]: Leaving directory '/root/openwrt/lede'
make: *** [/root/openwrt/lede/include/toplevel.mk:230: world] Error 2
```

## 有人遇到上面问题吗，求解

lovehackintosh, 2022-10-21:
不要用root用户编译

ecrasy, 2022-10-21:
configure: error: you should not run configure as root

1582130940, 2022-10-21:
configure: error: you should not run configure as root (set FORCE_UNSAFE_CONFIGURE=1 in environment to bypass this check)
不要用root用户，要不就设置FORCE_UNSAFE_CONFIGURE=1到环境

shenyz, 2022-10-27:
> 反馈bug/问题模板，提建议请删除
> 
> ```shell
> checking whether mkfifo rejects trailing slashes... yes
> checking whether mkfifoat rejects trailing slashes... yes
> checking whether mknod can create fifo without root privileges... configure: error: in `/root/openwrt/lede/build_dir/host/tar-1.34':
> configure: error: you should not run configure as root (set FORCE_UNSAFE_CONFIGURE=1 in environment to bypass this check)
> See `config.log' for more details
> make[3]: *** [Makefile:32: /root/openwrt/lede/build_dir/host/tar-1.34/.configured] Error 1
> make[3]: Leaving directory '/root/openwrt/lede/tools/tar'
> time: tools/tar/compile#12.71#4.42#20.69
>     ERROR: tools/tar failed to build.
> make[2]: *** [tools/Makefile:173: tools/tar/compile] Error 1
> make[2]: Leaving directory '/root/openwrt/lede'
> make[1]: *** [tools/Makefile:169: /root/openwrt/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnyyynynyyyyyyynyynnyynynyyyyyyyyyyyyyyyyyynynnyyyyyyyyyn] Error 2
> make[1]: Leaving directory '/root/openwrt/lede'
> make: *** [/root/openwrt/lede/include/toplevel.mk:230: world] Error 2
> ```
> 
> ## 有人遇到上面问题吗，求解
> ### 执行 make V=s -j$(nproc) 出错。

exprot FORCE_UNSAFE_CONFIGURE=1  就可以编译了

ookinasiawase, 2022-10-27:
> exprot FORCE_UNSAFE_CONFIGURE=1

exprot FORCE_UNSAFE_CONFIGURE=1 加在哪里？

shenyz, 2022-10-27:
> > exprot FORCE_UNSAFE_CONFIGURE=1
> 
> exprot FORCE_UNSAFE_CONFIGURE=1 加在哪里？

直接在编译的界面运行这条命令就可以了

billylicn, 2024-02-21:
感谢，另外改正一下命令是`export FORCE_UNSAFE_CONFIGURE=1
`

coolsnowwolf, 2026-09-05:
这是 GNU tar 的 configure 主动禁止 root 用户构建（`you should not run configure as root`），不是 x86/LEDE 源码错误。应使用普通用户在工作目录构建；若构建环境确实只能使用 root，可显式设置 `FORCE_UNSAFE_CONFIGURE=1`，但这会绕过上游安全检查，不建议作为固件默认修复。该 issue 基于 2022 年环境，当前构建系统仍遵循这一规则，关闭。

closed 2026-09-05 by coolsnowwolf