coolsnowwolf/lede #10294: 本地编译一直没能解决的问题

Shaco-Ma, 2022-10-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：X86编译release的代码,编译到libselinux的时候,install对应的库失败

### (2) 路由器型号和固件版本

A：X86_64

### (3) 详细日志

A：CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2=libselinux-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/shaco/My_data/lede-20221001/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/shaco/My_data/lede-20221001/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/shaco/My_data/lede-20221001/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2=libselinux-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/shaco/My_data/lede-20221001/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/shaco/My_data/lede-20221001/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/shaco/My_data/lede-20221001/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " LDFLAGS="-L/home/shaco/My_data/lede-20221001/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/shaco/My_data/lede-20221001/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro " make -j1 -C /home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/. AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2=libselinux-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" FTS_LDLIBS=-lfts SHLIBDIR=/usr/lib OS=Linux all;
make[4]: Entering directory '/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2'
make[5]: Entering directory '/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/include'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/include'
make[5]: Entering directory '/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/src'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/src'
make[5]: Entering directory '/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/utils'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/utils'
make[5]: Entering directory '/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/man'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/man'
make[4]: Leaving directory '/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2'
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2=libselinux-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/shaco/My_data/lede-20221001/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/shaco/My_data/lede-20221001/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/shaco/My_data/lede-20221001/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2=libselinux-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/shaco/My_data/lede-20221001/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/shaco/My_data/lede-20221001/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/shaco/My_data/lede-20221001/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " LDFLAGS="-L/home/shaco/My_data/lede-20221001/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/shaco/My_data/lede-20221001/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro " make -C /home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/. AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2=libselinux-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" FTS_LDLIBS=-lfts SHLIBDIR=/usr/lib OS=Linux DESTDIR="/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/ipkg-install"  install;
make[4]: Entering directory '/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2'
make[5]: Entering directory '/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/include'
test -d /home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/ipkg-install/usr/include/selinux || install -m 755 -d /home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/ipkg-install/usr/include/selinux
install -m 644 selinux/get_context_list.h selinux/get_default_type.h selinux/avc.h selinux/context.h selinux/restorecon.h selinux/selinux.h selinux/label.h /home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/ipkg-install/usr/include/selinux
make[5]: Leaving directory '/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/include'
make[5]: Entering directory '/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/src'
test -d /home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/ipkg-install/usr/lib || install -m 755 -d /home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/ipkg-install/usr/lib
install: error while loading shared libraries: libpcre.so.1: cannot open shared object file: No such file or directory

make[4]: Entering directory '/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2'
make[5]: Entering directory '/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/include'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/include'
make[5]: Entering directory '/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/src'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/src'
make[5]: Entering directory '/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/utils'
make[5]: Nothing to be done for 'all'.
[... cut ...]
test -d /home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/ipkg-install/usr/lib || install -m 755 -d /home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/ipkg-install/usr/lib
install: error while loading shared libraries: libpcre.so.1: 错误是执行install命令错误,但是编译主机已经装好pcre,openssl,zlib这些
另外发现在/home/shaco/My_data/lede-20221001/build_dir/target-x86_64_musl/libselinux-3.2/src这个目录下,执行ls都会失败,同样提示
缺少libpcre.so.1动态库,实在是想不明白

另外,我看了不少的issues都有提到过这个问题,但是都是没有解决的,都被关闭了
恳求大佬们解决一下!

lovehackintosh, 2022-10-21:
是用ubuntu 18.06 编译的吗？

lovehackintosh, 2022-10-21:
可以尝试执行以下命令
```bash
wget https://github.com/lovehackintosh/lede/commit/4de74b255b5e8088c1ef2797935454ed2c00e4bf.patch
git apply 4de74b255b5e8088c1ef2797935454ed2c00e4bf.patch
```

closed 2022-10-22 by aiamadeus

Shaco-Ma, 2022-10-24:
> 可以尝试执行以下命令
> 
> ```shell
> wget https://github.com/lovehackintosh/lede/commit/49f4013e761387c0cd647fb8877e70926e90a2e3.patch
> git apply 49f4013e761387c0cd647fb8877e70926e90a2e3.patch
> ```

切换到3.3也是一样的问题,问题根本不在这里,是没有生成对应的库,或者说selinux的文件夹内,执行install失败

lovehackintosh, 2022-10-24:
`HOST_BUILD_DEPENDS:=libsepol/host pcre/host`
有pcre这个依赖呀，我没遇见过这个问题，你的编译环境是什么

Shaco-Ma, 2022-10-24:
> `HOST_BUILD_DEPENDS:=libsepol/host pcre/host` 有pcre这个依赖呀，我没遇见过这个问题，你的编译环境是什么

找到问题了,是因为我在.bashrc设置了QT的环境,把LD_LIBRARY_PATH重定向了,但是原始的LD_LIBRARY_PATH又没有值,导致编译失败了,去掉QT的LD_LIBRARY_PATH重定向就可以了

PR #10300 (2022-10-22): libselinux: bump to 3.3 and sync with upstream
Fixes: #10294 

Q：你知道这是`pull request`吗？(使用 "x" 选择)
* [x] 我知道
commit e19b441c (2022-10-22): libselinux: sync with upstream (#10300)

Fixes: #10294