coolsnowwolf/lede #10301: 编译X86出错

yishan2019, 2022-10-22:
如下
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
make[2]: Leaving directory '/workdir/openwrt'
make[1]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:111: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

yishan2019, 2022-10-22:
多次通不过

yishan2019, 2022-10-22:
云编译。以前都没问题。这次没用在线配置，关闭了SSH。上传的本地的.config配置

EnnawYang, 2022-10-22:
能不能往上翻翻
[image]

AXBNG, 2022-10-22:
你这日志跟没发一样.........  (ノ-_-)ノ~┻━┻

楼上给你截图了，有冲突

closed 2022-10-23 by yishan2019