coolsnowwolf/lede #10308: syntax error: operand expected (error token is "*1024*1024")

openrouter, 2022-10-23:
make[2]: Leaving directory '/home/tony/master/scripts/config'
bash: *1024*1024: syntax error: operand expected (error token is "*1024*1024")
bash: *1024*1024: syntax error: operand expected (error token is "*1024*1024")
time: target/linux/prereq#0.42#0.04#0.45

lovehackintosh, 2022-10-23:
issue 2日志没发全,执行`make tools/install V=s -j1`把完整日志贴上来

openrouter, 2022-10-23:
> issue 2日志没发全,执行`make tools/install V=s -j1`把完整日志贴上来

after updated cmake issue 2 shutdown, tks.

how is issue1 ?

lovehackintosh, 2022-10-23:
please run `make target/linux/{clean,compile} V=s -j1` and provide full log

openrouter, 2022-10-23:
full logfile  attached 
[make.log](https://github.com/coolsnowwolf/lede/files/9847791/make.log)

lovehackintosh, 2022-10-24:
Please use ubuntu or debina to compile,don't use wsl

openrouter, 2022-10-24:
> Please use ubuntu or debina to compile,don't use wsl

Host OS is 18.04.2-Ubuntu

lovehackintosh, 2022-10-24:
Please use Ubuntu-20.04

yuzhe1129, 2022-10-26:
This is the case with me. I don't know what happened

yuzhe1129, 2022-10-26:
> Refer

> 请使用 Ubuntu-20.04

我也是使用20.04啊，也是出错这个错误 到底是什么引起的啊

schidler, 2022-10-27:
我这也是一样，我是用docker措的编译环境，是否对什么工具的版本有要求？

bash: *1024*1024: syntax error: operand expected (error token is "*1024*1024")

确认是这里引起
/include/image.mk
167 ROOTFS_PARTSIZE=$(shell echo $$(($(CONFIG_TARGET_ROOTFS_PARTSIZE)*1024*1024)))

yuzhe1129, 2022-10-27:
> 

> 我这也是一样，我是用docker的编译环境，对什么工具的版本有要求吗？
> 
> bash：_1024_ 1024：语法错误：预期操作数（错误标记为“ _1024_ 1024”）
> 
> 确认是这里引发 /include/image.mk 167 ROFS_PARTSIZE=$(shell echo $$(($(CONFIG_TARGET_ROOTFS_PARTSIZE) _1024_ 1024)))

是什么意思？我不怎么懂，我只会编译，能否解释简单点怎么解决

schidler, 2022-10-27:
> > 
> 
> > 我这也是一样，我是用docker的编译环境，对什么工具的版本有要求吗？
> > bash：_1024_ 1024：语法错误：预期操作数（错误标记为“ _1024_ 1024”）
> > 确认是这里引发 /include/image.mk 167 ROFS_PARTSIZE=$(shell echo $$(($(CONFIG_TARGET_ROOTFS_PARTSIZE) _1024_ 1024)))
> 
> 是什么意思？我不怎么懂，我只会编译，能否解释简单点怎么解决

`ROOTFS_PARTSIZE := $(shell expr $$(( $(CONFIG_TARGET_ROOTFS_PARTSIZE) * 1024 * 1024)))
`
可以这样试下

yuzhe1129, 2022-10-27:
> `ROOTFS_PARTSIZE := $(shell expr $$(( $(CONFIG_TARGET_ROOTFS_PARTSIZE) * 1024 * 1024))) ` 可以这样试下

:~$ cd lede
xxoo@ubuntu:~/lede$ ROOTFS_PARTSIZE := $(shell expr $$(( $(CONFIG_TARGET_ROOTFS_PARTSIZE) * 1024 * 1024)))
-bash: command substitution: 行 3: 未预期的符号“(”附近有语法错误
-bash: command substitution: 行 3: `shell expr $$(( $(CONFIG_TARGET_ROOTFS_PARTSIZE) * 1024 * 1024)))'

错误？

schidler, 2022-10-27:
> > > > 
> > > 
> > > 
> > > > 我这也是一样，我是用docker的编译环境，对什么工具的版本有要求吗？
> > > > bash：_1024_ 1024：语法错误：预期操作数（错误标记为“ _1024_ 1024”）
> > > > 确认是这里引发 /include/image.mk 167 ROFS_PARTSIZE=$(shell echo $$(($(CONFIG_TARGET_ROOTFS_PARTSIZE) _1024_ 1024)))
> > > 
> > > 
> > > 是什么意思？我不怎么懂，我只会编译，能否解释简单点怎么解决
> > 
> > 
> > `ROOTFS_PARTSIZE := $(shell expr $$(( $(CONFIG_TARGET_ROOTFS_PARTSIZE) * 1024 * 1024))) ` 可以这样试下
> 
> ROOTFS_PARTSIZE=$(shell echo $$(($(CONFIG_TARGET_ROOTFS_PARTSIZE)_1024_1024)))
> 
> 看了下， 原来是echo 改了expr就行了？

+ifdef CONFIG_TARGET_ROOTFS_PARTSIZE
 ROOTFS_PARTSIZE=$(shell echo $$(($(CONFIG_TARGET_ROOTFS_PARTSIZE)*1024*1024)))
+endif

估计是CONFIG_TARGET_ROOTFS_PARTSIZE这个没有定义
增加下判断

closed 2023-02-28 by aiamadeus