coolsnowwolf/lede #10317: 重启后文件只读了，但是执行  mount -o remount rw / 可解决

aiastia, 2022-10-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：新编译的 openwrt 重启后 系统变只读 Read-only file system
执行 
mount -o remount rw /
后 又可以读写了

### (2) 路由器型号和固件版本

A：x86 软路由 <html>
<body>

OpenWrt R22.10.20 / LuCI Master (git-22.289.58375-2a380a3)
--

<br class="Apple-interchange-newline">
</body>
</html>
 
### (3) 详细日志

aiastia, 2022-10-26:
内核日志

````
[    0.000000] Linux version 5.15.74 (runner@fv-az741-725) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r5317-063f51c24) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP Tue Oct 25 04:40:31 2022
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=9a260c83-02 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[3]:  960, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]: 1024, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1b, context size is 1088 bytes, using 'standard' format.
[    0.000000] signal: max sigframe size: 1616
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003ffdffff] usable
[    0.000000] BIOS-e820: [mem 0x000000003ffe0000-0x000000003fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (i440FX + PIIX, 1996), BIOS Virtualization-2.5.1-11303-200428-0-gfff70e7 04/01/2014
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: cpu 0, msr 248b001, primary cpu clock
[    0.000002] kvm-clock: using sched offset of 3784402776301 cycles
[    0.000006] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000012] tsc: Detected 1996.800 MHz processor
[    0.000901] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000906] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000912] last_pfn = 0x3ffe0 max_arch_pfn = 0x400000000
[    0.000952] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.009209] found SMP MP-table at [mem 0x000f5b80-0x000f5b8f]
[    0.009237] Using GB pages for direct mapping
[    0.009472] ACPI: Early table checksum verification disabled
[    0.009478] ACPI: RSDP 0x00000000000F5990 000014 (v00 BOCHS )
[    0.009492] ACPI: RSDT 0x000000003FFE1549 000030 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.009501] ACPI: FACP 0x000000003FFE1435 000074 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.009510] ACPI: DSDT 0x000000003FFE0040 0013F5 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.009516] ACPI: FACS 0x000000003FFE0000 000040
[    0.009520] ACPI: APIC 0x000000003FFE14A9 000078 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.009526] ACPI: WAET 0x000000003FFE1521 000028 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.009530] ACPI: Reserving FACP table memory at [mem 0x3ffe1435-0x3ffe14a8]
[    0.009533] ACPI: Reserving DSDT table memory at [mem 0x3ffe0040-0x3ffe1434]
[    0.009534] ACPI: Reserving FACS table memory at [mem 0x3ffe0000-0x3ffe003f]
[    0.009536] ACPI: Reserving APIC table memory at [mem 0x3ffe14a9-0x3ffe1520]
[    0.009538] ACPI: Reserving WAET table memory at [mem 0x3ffe1521-0x3ffe1548]
[    0.009598] Zone ranges:
[    0.009601]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.009604]   DMA32    [mem 0x0000000001000000-0x000000003ffdffff]
[    0.009606]   Normal   empty
[    0.009608] Movable zone start for each node
[    0.009610] Early memory node ranges
[    0.009611]   node   0: [mem 0x0000000000001000-0x000000000009efff]
[    0.009613]   node   0: [mem 0x0000000000100000-0x000000003ffdffff]
[    0.009615] Initmem setup node 0 [mem 0x0000000000001000-0x000000003ffdffff]
[    0.009637] On node 0, zone DMA: 1 pages in unavailable ranges
[    0.009776] On node 0, zone DMA: 97 pages in unavailable ranges
[    0.017599] On node 0, zone DMA32: 32 pages in unavailable ranges
[    0.018154] ACPI: PM-Timer IO Port: 0x608
[    0.018171] ACPI: LAPIC_NMI (acpi_id[0xff] dfl dfl lint[0x1])
[    0.018216] IOAPIC[0]: apic_id 0, version 17, address 0xfec00000, GSI 0-23
[    0.018222] ACPI: INT_SRC_OVR (bus 0 bus_irq 0 global_irq 2 dfl dfl)
[    0.018225] ACPI: INT_SRC_OVR (bus 0 bus_irq 5 global_irq 5 high level)
[    0.018227] ACPI: INT_SRC_OVR (bus 0 bus_irq 9 global_irq 9 high level)
[    0.018229] ACPI: INT_SRC_OVR (bus 0 bus_irq 10 global_irq 10 high level)
[    0.018231] ACPI: INT_SRC_OVR (bus 0 bus_irq 11 global_irq 11 high level)
[    0.018236] ACPI: Using ACPI (MADT) for SMP configuration information
[... 997 lines cut ...]
Wed Oct 26 08:34:05 2022 daemon.err nginx[6989]: 2022/10/26 08:34:04 [emerg] 6989#0: bind() to 0.0.0.0:80 failed (98: Address in use)
Wed Oct 26 08:34:05 2022 daemon.err nginx[6989]: 2022/10/26 08:34:04 [emerg] 6989#0: bind() to [::]:80 failed (98: Address in use)
Wed Oct 26 08:34:05 2022 daemon.err nginx[6989]: 2022/10/26 08:34:04 [emerg] 6989#0: bind() to 0.0.0.0:443 failed (98: Address in use)
[... cut ...]
Wed Oct 26 08:34:47 2022 daemon.notice ttyd[4901]: started process, pid: 9023
Wed Oct 26 08:34:48 2022 daemon.notice ttyd[4901]: WS closed from 192.168.0.167, clients: 0
Wed Oct 26 08:34:48 2022 daemon.notice ttyd[4901]: killing process 9023 with signal: 1 (SIGHUP)
Wed Oct 26 08:34:48 2022 daemon.notice ttyd[4901]: process killed with signal 1 (SIGHUP), pid: 9023
Wed Oct 26 08:35:21 2022 kern.debug kernel: [  124.781471] ata2.01: NODEV after polling detection
Wed Oct 26 08:36:04 2022 daemon.err uhttpd[2012]: mv: can't rename '/root/node.sh': Read-only file system
````

aiastia, 2022-10-26:
````
config文件
# 编译x64固件:
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_Generic=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_GRUB_IMAGES=y
CONFIG_TARGET_ROOTFS_EXT4FS=y
# 内核
# CONFIG_LINUX_5_10=y
CONFIG_LINUX_5_15=y
# 固件压缩
CONFIG_TARGET_IMAGES_GZIP=y
CONFIG_TARGET_KERNEL_PARTSIZE=512
CONFIG_TARGET_ROOTFS_PARTSIZE=1024
#编译UEFI固件:
CONFIG_EFI_IMAGES=y
#IPv6支持:
CONFIG_PACKAGE_dnsmasq_full_dhcpv6=y
CONFIG_PACKAGE_ipv6helper=y
#多文件系统支持:
CONFIG_PACKAGE_kmod-fs-nfs=y
CONFIG_PACKAGE_kmod-fs-nfs-common=y
CONFIG_PACKAGE_kmod-fs-nfs-v3=y
CONFIG_PACKAGE_kmod-fs-nfs-v4=y
CONFIG_PACKAGE_kmod-fs-ntfs=y
CONFIG_PACKAGE_kmod-fs-squashfs=y
#USB3.0支持:
CONFIG_PACKAGE_kmod-usb-ohci=y
CONFIG_PACKAGE_kmod-usb-ohci-pci=y
CONFIG_PACKAGE_kmod-usb2=y
CONFIG_PACKAGE_kmod-usb2-pci=y
CONFIG_PACKAGE_kmod-usb3=y
#常用LuCI插件选择:
CONFIG_PACKAGE_luci-app-adbyby-plus=y
CONFIG_PACKAGE_luci-app-aria2=y
CONFIG_PACKAGE_luci-app-baidupcs-web=y
CONFIG_PACKAGE_luci-app-frpc=y
CONFIG_PACKAGE_luci-app-frps=y
CONFIG_PACKAGE_luci-app-nps=y
CONFIG_PACKAGE_luci-i18n-frpc-zh-cn=y
CONFIG_PACKAGE_luci-i18n-frps-zh-cn=y
CONFIG_PACKAGE_luci-i18n-nps-zh-cn=y
CONFIG_PACKAGE_luci-app-kodexplorer=y
CONFIG_PACKAGE_luci-app-minidlna=y
CONFIG_PACKAGE_luci-app-openvpn=y
CONFIG_PACKAGE_luci-app-openvpn-server=y
CONFIG_PACKAGE_luci-app-qbittorrent=y
CONFIG_PACKAGE_luci-app-ssr-plus=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Kcptun=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Server=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Socks=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_V2ray=y
CONFIG_PACKAGE_luci-app-ttyd=y
CONFIG_PACKAGE_luci-app-v2ray-server=y
CONFIG_PACKAGE_luci-app-verysync=y
CONFIG_PACKAGE_luci-app-webadmin=y
CONFIG_PACKAGE_luci-app-wrtbwmon=y
CONFIG_PACKAGE_luci-app-koolproxyR=y
CONFIG_PACKAGE_luci-app-passwall=y
CONFIG_PACKAGE_luci-i18n-passwall-zh-cn=y
CONFIG_PACKAGE_luci-app-passwall_Transparent_Proxy=y
CONFIG_PACKAGE_luci-app-netdata=y
CONFIG_PACKAGE_luci-i18n-netdata-zh-cn=y
CONFIG_PACKAGE_luci-app-adguardhome=y
#CONFIG_PACKAGE_luci-app-smartdns=y
CONFIG_PACKAGE_luci-proto-ipv6=y
#CONFIG_PACKAGE_luci-app-diskman=y
CONFIG_PACKAGE_luci-app-alist=y
#LuCI主题:
CONFIG_PACKAGE_luci-theme-argon=y
#常用软件包:
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_htop=y
CONFIG_PACKAGE_nano=y
CONFIG_PACKAGE_screen=y
CONFIG_PACKAGE_tree=y
CONFIG_PACKAGE_vim-fuller=y
CONFIG_PACKAGE_wget=y
CONFIG_PACKAGE_open-vm-tools=y
CONFIG_PACKAGE_snmpd=y
CONFIG_PACKAGE_tcping=y
CONFIG_PACKAGE_unzip=y
CONFIG_PACKAGE_urngd=y
CONFIG_PACKAGE_luci-app-rclone=y
CONFIG_PACKAGE_python3=y
CONFIG_PACKAGE_python3-speedtest-cli=y
#取消编译VMware镜像以及镜像填充 (不要删除被缩进的注释符号):
# CONFIG_TARGET_IMAGES_PAD is not set
# CONFIG_VMDK_IMAGES is not set
CONFIG_PACKAGE_wireguard-tools=y
CONFIG_DEFAULT_luci-app-wireguard=y
CONFIG_PACKAGE_kmod-wireguard=y
CONFIG_PACKAGE_luci-app-wireguard=y
CONFIG_PACKAGE_luci-proto-wireguard=y
CONFIG_PACKAGE_luci-i18n-wireguard-zh-cn=y
CONFIG_PACKAGE_luci-app-softethervpn=y
CONFIG_PACKAGE_luci-app-advanced=y
CONFIG_PACKAGE_luci-app-uugamebooster=y

hcym, 2022-10-26:
遇到，还有这种事，扔掉几个旧U盘，😂

zhangguanzhang, 2022-10-26:
你用的是ext4 的固件吧，机器是不是之前突然拔电之类断电过

aiastia, 2022-10-26:
> 你用的是ext4 的固件吧，机器是不是之前突然拔电之类断电过

对 ext4 的 我是虚拟机里面跑到 不存在拔电啊 ， 我配置好虚拟机 然后 进去配置 都ok 然后 输入reboot
就出现这个情况了

kousyougi, 2022-10-26:
我遇到一样的问题。都是每次开机都会变成read-only，输入mount -o remount rw / 可以正常
一台跑在esxi的x86跟一台R4S都是这样的状况，刷完开机可以读写，但重开机后就变read-only。
比较奇怪的是，有另一台R4S刷同一个固件，同一个型号的TF卡，没有重开机变read-only的问题。

aiastia, 2022-10-26:
我是虚拟机 我都这样…… 但是我发现我9月30 编译的没问题

aiastia, 2022-10-26:
10🈷️15编译的好像也没有问题。我 在找找有么有更近时间编译的

aiastia, 2022-10-26:
10.22编译的就会这样 重启就read-only

ecrasy, 2022-10-27:
系统->挂载点
这个模块是block-mount应用负责的
好像是block-mount挂载失败了
就把根系统分区重新挂载成read-only
其实没有失败，根分区已经挂载成功并且可读写
我把block-mount配置里要挂载根分区的项删掉了
系统启动自己会挂载根分区
然后问题就没了
这个蠢应用block-mount惹的祸

ecrasy, 2022-10-27:
我在官方源码master分支编译的固件也是同样的问题
估计就是这个block-mount导致的。

hcym, 2022-10-28:
block-mount不编译，好了
谢谢分享

jxph1123, 2022-11-05:
> block-mount

我也是变只读，请问改哪里的配置，谢谢

jxph1123, 2022-11-05:
> 系统->挂载点 这个模块是block-mount应用负责的 好像是block-mount挂载失败了 就把根系统分区重新挂载成read-only 其实没有失败，根分区已经挂载成功并且可读写 我把block-mount配置里要挂载根分区的项删掉了 系统启动自己会挂载根分区 然后问题就没了 这个蠢应用block-mount惹的祸

我也是变只读，请问改哪里的配置，谢谢

nidbCN, 2022-11-14:
> > 系统->挂载点 这个模块是block-mount应用负责的 好像是block-mount挂载失败了 就把根系统分区重新挂载成read-only 其实没有失败，根分区已经挂载成功并且可读写 我把block-mount配置里要挂载根分区的项删掉了 系统启动自己会挂载根分区 然后问题就没了 这个蠢应用block-mount惹的祸
> 
> 我也是变只读，请问改哪里的配置，谢谢

luci: 

在“系统” “挂载点” 中删除或取消勾选挂载点为 `/` 的记录。

sean908, 2022-11-17:
> > block-mount不编译，好了 谢谢分享
> 
> 非常感谢大佬的分享。我想要问下，我这里 `make menuconfig` 中 `block-mount` 默认勾选并且不能取消（为 `-*-`）。大佬是怎么不编译这个的？还是说我需要取消某些依赖于他的软件包？

估计是有依赖, 我的可以取消
这个根目录只读的问题前几天我遇到了, 当时没头绪, 直接给刷回旧版本去了...

lyrl, 2022-11-22:
默认 -*- 无法取消

nkym233, 2022-11-25:
11.25编译还是这样 ，block-mount默认选择无法取消。有解决方法吗

lyrl, 2022-11-25:
> 11.25编译还是这样 ，block-mount默认选择无法取消。有解决方法吗

mount -o remount rw /
 添加到启动脚本里就行了

hcym, 2022-11-25:
就两个地方，删掉完事了，x86里面还有一个，其他架构就是一个

991547436, 2022-12-19:
> 11.25编译还是这样 ，block-mount默认选择无法取消。有解决方法吗
按顺序取消选中
Extra packages ---> automount
Base system ---> block-mount

aiastia, 2023-02-14:
这个bug 还没修好吗（

happyghw, 2023-03-11:
这个问题目前还是存在，我折腾了好久。第一次开机正常，重启一下就只读。

991547436, 2023-03-16:
> 这个问题目前还是存在，我折腾了好久。第一次开机正常，重启一下就只读。

看我之前的回答

happyghw, 2023-03-18:
> 

按顺序取消选中
Extra packages ---> automount
Base system ---> block-mount
实测还是不能取消。

MinChoul0, 2023-04-10:
还没解决么？我刷几次也遇到首次可以，重启之后就只读了，后续在挂载点里面不要点生成配置，重启了几次，暂时没有异常，另外开机脚本加mount -r后是可以解决么

991547436, 2023-04-10:
> 还没解决么？我刷几次也遇到首次可以，重启之后就只读了，后续在挂载点里面不要点生成配置，重启了几次，暂时没有异常，另外开机脚本加mount -r后是可以解决么
block-mount的问题
按顺序取消选中
Extra packages ---> automount
Base system ---> block-mount
重新打包即可 我是这样 如果不行看看那个包是否确实未安装

baopengsun, 2023-10-04:
还没解决么？我刷几次也都遇到了，首次可以，重启之后就只读了 …………

nidbCN, 2023-10-08:
> 还没解决么？我刷几次也都遇到了，首次可以，重启之后就只读了 …………

第一次启动后手动配置下挂载点就好了，或者写在初始化脚本里面

helllkz, 2023-11-21:
之前通过这种方式都还能解决，目前貌似都没法解决了。。
root@OpenWrt:~# mount -o remount rw /
mount: /: cannot remount rw read-write, is write-protected.
       dmesg(1) may have more information after failed mount system call.

TheLastManT1, 2023-12-31:
今天编译也遇到了，希望可以修复

bull1012, 2024-02-11:
确实遇到了一样的问题，/dev/root on / type ext4 (ro,noatime)  无法解决

vpeu, 2024-02-19:
block-mount 引起问题的可能性很大。

igmkgu, 2024-10-06:
我也遇到了，按楼上说的挂载点里取消勾选挂载/就好了

coolsnowwolf, 2026-09-05:
现有讨论已确认根因是 block-mount/automount 为根分区生成了不合适的挂载配置，导致启动后根文件系统被重新挂成只读；删除“挂载点”中针对 `/` 的记录，或不选择 `automount`/`block-mount` 后即可恢复。该行为不是内核或 LEDE 根文件系统代码缺陷，当前 issue 无需源码补丁。

closed 2026-09-05 by coolsnowwolf