coolsnowwolf/lede #10320: x86 编译 报错

maoguojun, 2022-10-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86 

### (3) 详细日志

A：
touch /home/mao/下载/lede/build_dir/target-x86_64_musl/runc-1.1.2/.configured_1cd727ac7b9ebbe4ee453b5ddb8504b3
rm -f /home/mao/下载/lede/build_dir/target-x86_64_musl/runc-1.1.2/.built
touch /home/mao/下载/lede/build_dir/target-x86_64_musl/runc-1.1.2/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/mao/下载/lede/build_dir/target-x86_64_musl/runc-1.1.2=runc-1.1.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/mao/下载/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/mao/下载/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/mao/下载/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/mao/下载/lede/build_dir/target-x86_64_musl/runc-1.1.2=runc-1.1.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/mao/下载/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/mao/下载/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/mao/下载/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " LDFLAGS="-L/home/mao/下载/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/mao/下载/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro " GOOS="linux" GOARCH="amd64" GO386="" GOAMD64="v1" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/mao/下载/lede/build_dir/target-x86_64_musl/runc-1.1.2=runc-1.1.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/mao/下载/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/mao/下载/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/mao/下载/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/mao/下载/lede/build_dir/target-x86_64_musl/runc-1.1.2=runc-1.1.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/mao/下载/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/mao/下载/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro" GOPATH="/home/mao/下载/lede/build_dir/target-x86_64_musl/runc-1.1.2/.go_work/build" GOCACHE="/home/mao/下载/lede/tmp/go-build" GOMODCACHE="/home/mao/下载/lede/dl/go-mod-cache" GOENV=off make  -C /home/mao/下载/lede/build_dir/target-x86_64_musl/runc-1.1.2/.go_work/build/src/github.com/opencontainers/runc AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/mao/下载/lede/build_dir/target-x86_64_musl/runc-1.1.2=runc-1.1.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" COMMIT=a916309fff0f838eb94e928713dbc3c0d0ac7aa4 BUILDTAGS='seccomp' ;
make[4]: Entering directory '/home/mao/下载/lede/build_dir/target-x86_64_musl/runc-1.1.2/.go_work/build/src/github.com/opencontainers/runc'
go build -trimpath "-buildmode=pie"  -tags "seccomp" -ldflags "-X main.gitCommit=a916309fff0f838eb94e928713dbc3c0d0ac7aa4 -X main.version=1.1.2 " -o runc .
go build github.com/seccomp/libseccomp-golang: invalid flag in go:cgo_ldflag: -znow
make[4]: *** [Makefile:33: runc] Error 1
make[4]: Leaving directory '/home/mao/下载/lede/build_dir/target-x86_64_musl/runc-1.1.2/.go_work/build/src/github.com/opencontainers/runc'
make[3]: *** [Makefile:67: /home/mao/下载/lede/build_dir/target-x86_64_musl/runc-1.1.2/.built] Error 2
make[3]: Leaving directory '/home/mao/下载/lede/feeds/packages/utils/runc'
time: package/feeds/packages/runc/compile#24.60#12.52#13.25
    ERROR: package/feeds/packages/runc failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/runc/compile] Error 1
make[2]: Leaving directory '/home/mao/下载/lede'
make[1]: *** [package/Makefile:110: /home/mao/下载/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/mao/下载/lede'
make: *** [/home/mao/下载/lede/include/toplevel.mk:230：world] 错误 2

maoguojun, 2022-10-27:
编译环境 虚拟机 daban 11

ecrasy, 2022-10-27:
建议切换到Ubuntu 20.04
并且尽量不要使用中文路径
可以节省你很多摸索的时间

closed 2022-10-27 by aiamadeus