coolsnowwolf/lede #10323: x86 编译报错 ，请大佬看看啥情况

maoguojun, 2022-10-28:
[ -f /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/generic-kernel.bin -a -f /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/root.squashfs ]
[ -f /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/generic-kernel.bin -a -f /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/root.squashfs ]
dd if=/home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/root.squashfs >> /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz
rm -fR /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.boot
install -d -m0755 /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.boot/boot/grub
sed -e 's#@SERIAL_CONFIG@#serial --unit=0 --speed=115200 --word=8 --parity=no --stop=1 --rtscts=off#g' -e 's#@TERMINAL_CONFIG@#terminal_input console serial; terminal_output console serial#g' -e 's#@ROOTPART@#root=PARTUUID=6d89b569-02 rootwait#g' -e 's#@GPT_ROOTPART@#root=PARTUUID=6d89b569-0eb7-4077-0830-1037767ba702 rootwait#g' -e 's#@CMDLINE@# console=tty0 console=ttyS0,115200n8#g' -e 's#@TIMEOUT@#0#g' -e 's#@TITLE@#OpenWrt#g' ./grub-efi.cfg > /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.boot/boot/grub/grub.cfg
cp -fpR /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/bzImage /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.boot/boot/vmlinuz
cp -fpR /home/mao/lede/staging_dir/target-x86_64_musl/root-x86/boot/. /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.boot/boot/
cp: cannot stat '/home/mao/lede/staging_dir/target-x86_64_musl/root-x86/boot/.': No such file or directory
make[5]: [Makefile:164: /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz] Error 1 (ignored)
cp -fpR /home/mao/lede/staging_dir/target-x86_64_musl/image/grub2/boot.img /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.boot/boot/grub/
cp -fpR /home/mao/lede/staging_dir/target-x86_64_musl/image/grub2/gpt-core.img /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.boot/boot/grub/core.img
install -d -m0755 /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.boot/efi/boot
cp -fpR /home/mao/lede/staging_dir/target-x86_64_musl/image/grub2/bootx64.efi /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.boot/efi/boot/
PADDING="1" SIGNATURE="6d89b569" GUID="6d89b569-0eb7-4077-0830-1037767ba700" /home/mao/lede/scripts/gen_image_generic.sh /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz 160 /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.boot 4000 /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/root.squashfs 256
+ [ 6 -ne 5 ]
+ [ 6 -ne 6 ]
+ OUTPUT=/home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz
+ KERNELSIZE=160
+ KERNELDIR=/home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.boot
+ ROOTFSSIZE=4000
+ ROOTFSIMAGE=/home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/root.squashfs
+ ALIGN=256
+ rm -f /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz
+ head=16
+ sect=63
+ ptgen -o /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz -h 16 -s 63 -g -p 160m -p 4000m -l 256 -S 0x6d89b569 -G 6d89b569-0eb7-4077-0830-1037767ba700
part 0 163840
part 0 4096000
+ set 262144 167772160 168034304 4194304000
+ KERNELOFFSET=512
+ KERNELSIZE=167772160
+ ROOTFSOFFSET=328192
+ ROOTFSSIZE=8192000
+ [ -n 1 ]
+ dd if=/dev/zero of=/home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz bs=512 seek=328192 conv=notrunc count=8192000
84046+1 records in
84046+1 records out
43031651 bytes (43 MB, 41 MiB) copied, 0.102324 s, 421 MB/s
dd if=/home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz of=/home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz.new bs=4194304000 conv=sync
make[5]: *** [Makefile:164: /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz] Killed
make[5]: *** Deleting file '/home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz'
make[5]: *** Waiting for unfinished jobs....
8192000+0 records in
8192000+0 records out
4194304000 bytes (4.2 GB, 3.9 GiB) copied, 106.423 s, 39.4 MB/s
+ dd if=/home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/root.squashfs of=/home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz bs=512 seek=328192 conv=notrunc
84046+1 records in
84046+1 records out
43031651 bytes (43 MB, 41 MiB) copied, 13.8217 s, 3.1 MB/s
+ [ -n 6d89b569-0eb7-4077-0830-1037767ba700 ]
+ [ -n 1 ]
+ dd if=/dev/zero of=/home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz bs=512 seek=8520192 conv=notrunc count=63
63+0 records in
63+0 records out
32256 bytes (32 kB, 32 KiB) copied, 0.000109757 s, 294 MB/s
+ mkfs.fat --invariant -n kernel -C /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.kernel -S 512 163840
mkfs.fat 4.2 (2021-01-31)
mkfs.fat: Warning: lowercase labels might not work properly on some systems
+ LC_ALL=C dos_dircopy /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.boot /
+ local entry
+ local baseentry
+ [ -f /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.boot/boot ]
+ [ -d /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.boot/boot ]
+ basename /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.boot/boot
+ baseentry=boot
+ mmd -i /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.kernel ::/boot
+ dos_dircopy /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.boot/boot /boot/
+ local entry
+ local baseentry
[... 37 lines cut ...]
167772160 bytes (168 MB, 160 MiB) copied, 0.318411 s, 527 MB/s
+ rm -f /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.kernel
rm -fR /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2
install -d -m0755 /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2
cp -fpR /home/mao/lede/staging_dir/target-x86_64_musl/image/grub2/boot.img /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2/
cp -fpR /home/mao/lede/staging_dir/target-x86_64_musl/image/grub2/gpt-core.img /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2/core.img
echo '(hd0) /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz' > /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2/device.map
/home/mao/lede/staging_dir/host/bin/grub-bios-setup -m "/home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2/device.map" -d "/home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-combined-efi.img.gz.grub2" -r "hd0,gpt1" /home/mao/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-gen [... cut ...]
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/mao/lede/target/linux'
time: target/linux/install#77.99#105.26#239.70
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/home/mao/lede'
make[1]: *** [target/Makefile:24: /home/mao/lede/staging_dir/target-x86_64_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/mao/lede'
make: *** [/home/mao/lede/include/toplevel.mk:230：world] 错误 2
有人遇到上面问题吗，求解

ohbaby30, 2022-10-28:
root filesystem size太小 设大一点

maoguojun, 2022-10-31:
多谢大佬已解决

closed 2026-09-05 by coolsnowwolf