coolsnowwolf/lede #10329: mac80211 failed to build,  kmod-rtw88 is missing usbcore.ko

zhangguanzhang, 2022-10-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：n1 编译失败
```
CONFIG_TARGET_armvirt=y
CONFIG_TARGET_armvirt_64=y
CONFIG_TARGET_armvirt_64_Default=y

CONFIG_PACKAGE_kmod-mac80211=y
CONFIG_PACKAGE_kmod-usb-core=y
CONFIG_PACKAGE_kmod-rtw88=y
```

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

```
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.backports-5.15.33-1.tar.xz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "backports-5.15.33-1.tar.xz" "1b6b3bded4c81814ebebe2d194c2f8966d2399005b85ebb0557285b6e73f5422" "" "@KERNEL/linux/kernel/projects/backports/stable/v5.15.33/"    '
mkdir -p /workdir/openwrt/bin/targets/armvirt/64/packages /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/backports-5.15.33-1/ipkg-aarch64_cortex-a53/kmod-rtw88/CONTROL /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo
mkdir -p /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/backports-5.15.33-1/ipkg-aarch64_cortex-a53/kmod-rtw88/etc/modules.d; ( echo "rtw88_8723de";   echo "rtw88_8821cu";   echo "rtw88_8822be";   echo "rtw88_8822ce"; ) > /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/backports-5.15.33-1/ipkg-aarch64_cortex-a53/kmod-rtw88/etc/modules.d/rtw88; 
find /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/backports-5.15.33-1/ipkg-aarch64_cortex-a53/kmod-rtw88 -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package kmod-rtw88 is missing dependencies for the following libraries:
usbcore.ko
make[2]: *** [Makefile:563: /workdir/openwrt/bin/targets/armvirt/64/packages/kmod-rtw88_5.10.149+5.15.33-1-1_aarch64_cortex-a53.ipk] Error 1
make[2]: Leaving directory '/workdir/openwrt/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#0.50#0.16#0.60
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[1]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: package/compile] Error 2
```

minlang112, 2022-10-30:
USB网卡驱动ko依赖usbcore.ko，不加载这个就无法安装

zhangguanzhang, 2022-10-30:
> USB网卡驱动ko依赖usbcore.ko，不加载这个就无法安装

大兄弟，这是编译

minlang112, 2022-10-30:
> > USB网卡驱动ko依赖usbcore.ko，不加载这个就无法安装
> 
> 大兄弟，这是编译

我意思是看日志是kmod-rtw88_5.10.149+5.15.33-1-1_aarch64_cortex-a53.ipk这个包在打包的时候缺失了usbcore.ko，大概是kmod-rtw88的makefile里写漏了依赖吧，已经有task了应该是有人去修了

zhangguanzhang, 2022-10-31:
只能等了🤔，不过有几位contributer提交修复的时候没有link到issue，我只能过一段时间观察下commit

zhangguanzhang, 2022-11-04:
> > > USB网卡驱动ko依赖usbcore.ko，不加载这个就无法安装
> > 
> > 
> > 大兄弟，这是编译
> 
> 我意思是看日志是kmod-rtw88_5.10.149+5.15.33-1-1_aarch64_cortex-a53.ipk这个包在打包的时候报错了，缺失了usbcore.ko，大概是kmod-rtw88的makefile里写漏了依赖吧，已经有task了应该是有人去修了

老哥。是 Makefile的FILES里加ko的依赖吗

zhangguanzhang, 2022-11-06:
暂时的hack 解决，估计要有人makefile里解决该问题
```
    depend_line=$(awk '/KernelPackage\/rtw88*$/{flag=1}flag{flag++}flag>0&&flag<=7&&/^\s*DEPENDS/&& $0!~"kmod-usb-core"{print NR;exit}' ./package/kernel/mac80211/realtek.mk)
    if [ -n "$depend_line" ];then
        sed -ri "${depend_line}"'s#$# +kmod-usb-core#' ./package/kernel/mac80211/realtek.mk
    fi
```

closed 2022-12-29 by aiamadeus