coolsnowwolf/lede #10330: 电犀牛 R66s 编译遇到 mbedtls 错误

miaoermua, 2022-10-29:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

在使用最新的源码编译电犀牛 R66s 时其他基础插件都编译和了，准备过一边 Proxy 类插件，把 ssrp 默认勾上就会出现 [issues 8922](https://github.com/coolsnowwolf/lede/issues/8922) 相似的问题

```
/home/by/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.28.1/library/armv8ce_aes.c:61:17: note: called from here
```

在看下面说修改 mbedtls 的 PKG_VERSION 和 PKG_HASH 是没有这个问题了，但也出现新的问题，求解。

如果 issues 不合适请关闭。

```
……
make[3]: Leaving directory '/home/by/lede/feeds/luci/themes/luci-theme-bootstrap'
time: package/feeds/luci/luci-theme-bootstrap/compile#0.23#0.11#0.31
make[3]: Entering directory '/home/by/lede/feeds/luci/libs/rpcd-mod-rrdns'
mkdir -p /home/by/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
SHELL= flock /home/by/lede/tmp/.root-copy.flock -c 'cp -fpR /home/by/lede/build_dir/target-aarch64_generic_musl/rpcd-mod-rrdns-20170710/.pkgdir/rpcd-mod-rrdns/. /home/by/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /home/by/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.rpcd-mod-rrdns_installed
echo "rpcd-mod-rrdns" >> /home/by/lede/staging_dir/target-aarch64_generic_musl/pkginfo/rpcd-mod-rrdns.default.install
make[3]: Leaving directory '/home/by/lede/feeds/luci/libs/rpcd-mod-rrdns'
time: package/feeds/luci/rpcd-mod-rrdns/compile#0.15#0.06#0.19
make[3]: Entering directory '/home/by/lede/package/libs/mbedtls'
mkdir -p /home/by/lede/dl
SHELL= flock /home/by/lede/tmp/.mbedtls-2.16.10.tar.gz.flock -c '       /home/by/lede/scripts/download.pl "/home/by/lede/dl" "mbedtls-2.16.10.tar.gz" "96257bb03b30300b2f35f861ffe204ed957e9fd0329d80646fe57fc49f589b29" "" "https://codeload.github.com/ARMmbed/mbedtls/tar.gz/v2.16.10?"    '
touch /home/by/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.16.10/.prepared_676153cac07242d33d200240f7586ef0_6664517399ebbbc92a37c5bb081b5c53_check
. /home/by/lede/include/shell.sh; gzip -dc /home/by/lede/dl/mbedtls-2.16.10.tar.gz | tar -C /home/by/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.16.10/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/by/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.16.10

Applying ./patches/100-Implements-AES-and-GCM-with-ARMv8-Crypto-Extensions.patch using plaintext:
patching file ChangeLog.d/armv8_crypto_extensions.txt
patching file include/mbedtls/armv8ce_aes.h
patching file include/mbedtls/check_config.h
Hunk #1 succeeded at 95 (offset 23 lines).
Hunk #2 succeeded at 776 (offset -125 lines).
patching file include/mbedtls/config.h
Hunk #1 succeeded at 73 (offset 27 lines).
Hunk #2 succeeded at 1906 (offset -408 lines).
patching file library/aes.c
Hunk #1 succeeded at 69 (offset 30 lines).
Hunk #2 succeeded at 1054 (offset 53 lines).
patching file library/armv8ce_aes.c
patching file library/CMakeLists.txt
Hunk #1 succeeded at 7 (offset -8 lines).
patching file library/gcm.c
Hunk #1 succeeded at 71 (offset 30 lines).
Hunk #2 succeeded at 144 with fuzz 2 (offset 53 lines).
Hunk #3 succeeded at 258 (offset 52 lines).
patching file library/Makefile
Hunk #1 FAILED at 74.
1 out of 1 hunk FAILED -- saving rejects to file library/Makefile.rej
patching file library/version_features.c
Hunk #1 succeeded at 586 (offset -38 lines).
Patch failed!  Please fix ./patches/100-Implements-AES-and-GCM-with-ARMv8-Crypto-Extensions.patch!
make[3]: *** [Makefile:180: /home/by/lede/build_dir/target-aarch64_generic_musl/mbedtls-2.16.10/.prepared_676153cac07242d33d200240f7586ef0_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/by/lede/package/libs/mbedtls'
time: package/libs/mbedtls/compile#0.21#0.29#0.40
    ERROR: package/libs/mbedtls failed to build.
make[2]: *** [package/Makefile:116: package/libs/mbedtls/compile] Error 1
make[2]: Leaving directory '/home/by/lede'
make[1]: *** [package/Makefile:110: /home/by/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/by/lede'
make: *** [/home/by/lede/include/toplevel.mk:230: world] Error 2
by@by:~/lede$ make V=s -j1
^X^Cmake[2]: *** [include/scan.mk:102: /home/by/lede/tmp/info/.files-packageinfo.stamp] Interrupt
make[1]: *** [/home/by/lede/include/toplevel.mk:79: prepare-tmpinfo] Interrupt
make: *** [/home/by/lede/include/toplevel.mk:228: world] Interrupt
```

### (2) 路由器型号和固件版本

电犀牛 R66s

### (3) 详细日志

如上

zhangguanzhang, 2022-10-30:
```
Patch failed!  Please fix ./patches/100-Implements-AES-and-GCM-with-ARMv8-Crypto-Extensions.patch!
```

aiamadeus, 2022-10-31:
> SHELL= flock /home/by/lede/tmp/.mbedtls-2.16.10.tar.gz.flock -c '

你这也不是最新的代码，另外实测编译没有任何问题

closed 2022-10-31 by aiamadeus

miaoermua, 2022-11-17:
> > SHELL= flock /home/by/lede/tmp/.mbedtls-2.16.10.tar.gz.flock -c '
> 
> 你这也不是最新的代码，另外实测编译没有任何问题

找到问题了，不能编译 [bypass](https://github.com/kiddin9/openwrt-bypass) 和 passwall，openclash 和 ssrp 倒没事