coolsnowwolf/lede #10332: 发现ssh 拒绝连接，web UI正常登录

yuhongwei380, 2022-10-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
ssh相关问题
Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：近期编译固件发现新版本的固件编译出来，我qm import 到pve上，发现op发现ssh 拒绝连接，web UI正常登录；

### (2) 路由器型号和固件版本

A：X86版本；虚拟机，5.15.78

### (3) 详细日志

A：ssh连接，提示拒绝连接；

zhangguanzhang, 2022-10-30:
你都有pve了，pve的console进去看看日志和系统日志不行吗，以及consol里ssh -vvv localhost试试

yuhongwei380, 2022-10-31:
目前发现是xshell 无法ssh 连接到自编译新版本 的问题，用cmd 和 terminal可以ssh访问；case 先关了

closed 2022-10-31 by yuhongwei380

zhangguanzhang, 2022-10-31:
应该是ssh的密钥算法和op里的不匹配

yuhongwei380, 2022-10-31:
害，我也不知道哪儿出问题了；新编了一个版本，发现又没法 cmd 和terminal访问了；感觉是网络层面的问题或者是ssh 这边的问题，之前遇到了centos8 ssh 无法ssh，是因为ipqos 影响到。我们是访问ipsec 对端的 op；

zhangguanzhang, 2022-10-31:
你找个局域网的大型linux，或者你的pve，ssh -vvvv 看下调试信息

yuhongwei380, 2022-10-31:
root@OpenWrt:~# ssh localhost
ssh: Connection to root@localhost:22 exited: Connect failed: Connection refused

yuhongwei380, 2022-10-31:
大局域网内是可以ssh 的，跨ipsec 的访问就有问题了

yuhongwei380, 2022-10-31:
我真的醉了，刚装好新OP那会，配置好网络等参数没法连，等我跟你回复完消息又可以连接了 ；

zhangguanzhang, 2022-10-31:
你这是ipsec 的段冲突了吧。。。，你ssh的不是你的op

yuhongwei380, 2022-10-31:
ispec 段没冲突的哈；我是用的同局域网的ssh 到这台15.X的OP上，-vvv打印信息；

yuhongwei380, 2022-10-31:
这样子吗

yuhongwei380, 2022-10-31:
[image]

yuhongwei380, 2022-11-01:
我把op的版本降级回我们今年22-3月编译的版本，直接本地就ssh上了

yuhongwei380, 2022-11-01:
[image]

zhangguanzhang, 2022-11-01:
🤔你用git 的ssh试试呢
```
ssh -o HostKeyAlgorithms=+ssh-rsa -o PubkeyAcceptedKeyTypes=+ssh-rsa  xxx
```

yuhongwei380, 2022-11-01:
[image]

yuhongwei380, 2022-11-01:
还是一样的

zhangguanzhang, 2022-11-01:
```
ssh -o HostKeyAlgorithms=+ssh-rsa -o PubkeyAcceptedKeyTypes=+ssh-rsa -vvvv  xxx
```

yuhongwei380, 2022-11-01:
[image]
mu'

zhangguanzhang, 2022-11-02:
🤔也就是说搞了个另类的qos吗

zhangguanzhang, 2022-11-02:
🤔也就是说搞了个另类的qos吗

yuhongwei380, 2022-11-02:
算是吧，你搜 IPQos  ssh 这种可能会更有相关信息；

yuhongwei380, 2022-11-03:
我到dropbear 那个项目下提出这个问题了，作者表示后续会出一个选项，用来避免这样的情况。https://github.com/mkj/dropbear/issues/193

zhangguanzhang, 2022-11-03:
👍很棒