coolsnowwolf/lede #10335: 问题 libxslt 源未找到 ftp://xmlsoft.org/。

GenzPN, 2022-10-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：我有下载源 ftp://xmlsoft.org/ 的问题
[image]

### (2) 路由器型号和固件版本

A：p2w_r619ac-128m

### (3) 详细日志

A：
[build.log](https://github.com/coolsnowwolf/lede/files/9895782/build.log)

hzjnet, 2022-10-30:
> 反馈bug/问题模板，提建议请删除
> 
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
> 
> * [x]  没有类似的issue
> 
> ## 2. 详细叙述
> ### (1) 具体问题
> A：我有下载源 ftp://xmlsoft.org/ 的问题 [image]
> 
> ### (2) 路由器型号和固件版本
> A：p2w_r619ac-128m
> 
> ### (3) 详细日志
> A： [build.log](https://github.com/coolsnowwolf/lede/files/9895782/build.log)

原因是版本高了，退回上一个版本就行了：
PKG_NAME:=libxml2
PKG_VERSION:=2.9.9
PKG_RELEASE:=3

PKG_NAME:=libxslt
PKG_VERSION:=1.1.33
PKG_RELEASE:=4

GenzPN, 2022-11-04:
Still error i think

closed 2023-02-28 by aiamadeus