coolsnowwolf/lede #10338: 最新版SSR-PLUS的问题，可以顺利打开youtube视频列表，但打不开视频

ecco2005, 2022-10-31:
有1年多没编译新固件了，昨天心血来潮，编译了一个最新版的。ssr-plus遇到问题。

打开youtube首页很顺利，速度也很快。点开某个视频，右侧的视频列表也很快就出来了。

但是视频窗口一直转圈，黑屏。查看统计信息，网速那一栏是一个固定值，而且是不变化的。感觉就是没有流量。

SSR-PLUS设置里的各种选项也都试过了，比如运行模式，DNS节点，DNS解析方式，都试过了。

换了2个完全不一样的节点，协议都是SSR，问题依旧。这2个节点是绝对没问题的，而且在我昨天换新编译的固件之前，也一直是稳定运行的。

请问问题出在哪？

AXBNG, 2022-10-31:
DNS的问题，你谷歌应该也打不开.........

iseeyou, 2022-10-31:
关闭电脑的ipv6试试
我的也一样个别视频一直转，关闭电脑的ipv6秒开

ecco2005, 2022-10-31:
> DNS节点，DNS解析方式

> 关闭电脑的ipv6试试 我的也一样个别视频一直转，关闭电脑的ipv6秒开

DNS节点和DNS解析方式都换过了呀，客户端的DNS也换了好多个，试了无数种组合，都不行呀。

ecco2005, 2022-10-31:
问题解决了。事情是这样的，昨天我编译完新固件以后，想偷懒，于是直接把编译出来的luci-app-ssr-plus.ipk装到以前的固件里了，装完以后就不正常了。于是，我又在以前的固件里把新编译的固件丢进网页升级，升级也没报错。升完后看版本号，也的确是更新了。表面看起来一切正常，没想到SSR-PLUS出了问题。

刚刚我把硬盘删了，重新建了一个，再导入新固件，一切正常了

closed 2022-10-31 by ecco2005