coolsnowwolf/lede #10340: mac80211 failed to build rt2x00

zhangguanzhang, 2022-10-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：doornet2 编译报错
```
CONFIG_PACKAGE_kmod-mac80211=y

CONFIG_PACKAGE_kmod-rt2500-usb=y
CONFIG_PACKAGE_kmod-rt2800-lib=y
CONFIG_PACKAGE_kmod-rt2800-usb=y
CONFIG_PACKAGE_kmod-rt2x00-lib=n
CONFIG_PACKAGE_kmod-rt2x00-usb=n
```

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

```
make[3]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.10.68-1'
make[4]: 'Kconfig.versions' is up to date.
make[6]: 'Kconfig.versions' is up to date.
make[7]: 'conf' is up to date.
boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
#
# configuration written to .config
#
Building backport-include/backport/autoconf.h ... done.
  CC [M]  /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.10.68-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.o
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.10.68-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.c: In function 'rt2x00lib_set_mac_address':
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.10.68-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.c:999:9: error: too many arguments to function 'of_get_mac_address'
  999 |         of_get_mac_address(rt2x00dev->dev->of_node, eeprom_mac_addr);
      |         ^~~~~~~~~~~~~~~~~~
In file included from /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.10.68-1/backport-include/linux/of_net.h:3,
                 from /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.10.68-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.c:19:
./include/linux/of_net.h:14:20: note: declared here
   14 | extern const void *of_get_mac_address(struct device_node *np);
      |                    ^~~~~~~~~~~~~~~~~~
make[10]: *** [scripts/Makefile.build:262: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.10.68-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.o] Error 1
make[9]: *** [scripts/Makefile.build:497: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.10.68-1/drivers/net/wireless/ralink/rt2x00] Error 2
make[8]: *** [scripts/Makefile.build:497: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.10.68-1/drivers/net/wireless/ralink] Error 2
make[7]: *** [scripts/Makefile.build:497: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.10.68-1/drivers/net/wireless] Error 2
make[6]: *** [Makefile:1734: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.10.68-1] Error 2
make[5]: *** [Makefile.build:13: modules] Error 2
make[4]: *** [Makefile.real:93: modules] Error 2
make[3]: *** [Makefile:121: modules] Error 2
make[2]: *** [Makefile:568: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.10.68-1/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.10.68-1'
make[2]: Leaving directory '/workdir/openwrt/package/kernel/mac80211'
time: package/kernel/mac80211/compile#2.09#0.57#2.75
    ERROR: package/kernel/mac80211 failed to build.
make[1]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: package/compile] Error 2
```

aiamadeus, 2022-10-31:
你这明显不是最新的代码

closed 2022-10-31 by aiamadeus

zhangguanzhang, 2022-10-31:
> 你这明显不是最新的代码

今天下午的还不是最新的代码吗