coolsnowwolf/lede #10344: 请问如何默认关闭lan口的dhcp服务

viviczh1, 2022-11-01:
路由一直都是作为旁路网关使用，想在编译时默认关闭lan口的dhcp功能，请问如何修改配置？

zhangguanzhang, 2022-11-01:
```
# 通过 opkg search 配置文件找到属于dnsmasq的包
$ opkg search /etc/config/dhcp
dnsmasq-full - 2.86-13
# 源码里搜
$ find -type d -name 'dnsmasq*'
./package/network/services/dnsmasq
# 进目录找到文件
$ cat package/network/services/dnsmasq/files/dhcp.conf
```

主要是
```
config dhcp 'lan'
	option interface 'lan'
	option ignore '1'
```

另一种方法是用 `uci-defaults`，也就是源码目录存放 `files/etc/uci-defaults/zzz-default-settings.sh` 一个shell文件，名字的字母让排序是最后，里面写
```
uci set dhcp.lan.ignore='1'
uci commit dhcp
```
当然，最人性化的是 uci-defaults 的脚本里上面的逻辑外面加一个判断，如果只有一个网口才取消lan的dhcp

viviczh1, 2022-11-01:
> 
非常感谢，我这就测试一下

adamgonglj, 2022-11-02:
> ```
> # 通过 opkg search 配置文件找到属于dnsmasq的包
> $ opkg search /etc/config/dhcp
> dnsmasq-full - 2.86-13
> # 源码里搜
> $ find -type d -name 'dnsmasq*'
> ./package/network/services/dnsmasq
> # 进目录找到文件
> $ cat [...]

您好呀，前面的我通过学习openwrt的官方文档我会了。但是我就是不会最后一个，就是，逻辑判断，人性化地判断网口的数量。

我是这样设置的

```shell
# 设置自定义dhcp
cat > $lede/lede/package/base-files/files/etc/uci-defaults/80_customized << "EOF"

uci add_list dhcp.lan.dhcp_option="6,119.29.29.29,223.5.5.5"

uci add dhcp domain
uci set dhcp.@domain[-1].name="xxxxxxx.com"
uci set dhcp.@domain[-1].ip="192.168.50.1"
uci delete dhcp.@dnsmasq[0].filter_aaaa

uci commit dhcp

## uci delete dhcp.dnsmasq..option.filter_aaaa 这是自己的学习的记录

exit 0

EOF
```

zhangguanzhang, 2022-11-02:
> 您好呀，前面的我通过学习openwrt的官方文档我会了。但是我就是不会最后一个，就是，逻辑判断，人性化地判断网口的数量。
>
> 我是这样设置的
>
> ```shell
> # 设置自定义dhcp
> cat > $lede/lede/package/base-files/files/etc/uci-defaults/80_customized << "EOF"
>
> uci add_list [...]

```
if xxx;then
    uci set dhcp.lan.ignore='1'
    uci commit dhcp
fi
```
至于判断逻辑，不依靠op编译的命令的话，可以 `/sys/class/net` 目录信息作为依据，下面这个命令就是打印网口信息
```
ls -l /sys/class/net/ | awk '$NF~/\/devices\/(platform|pci)/{print $9}'
```

adamgonglj, 2022-11-02:
> > > ```
> > > # 通过 opkg search 配置文件找到属于dnsmasq的包
> > > $ opkg search /etc/config/dhcp
> > > dnsmasq-full - 2.86-13
> > > # 源码里搜
> > > $ find -type d -name 'dnsmasq*'
> > > ./package/network/services/dnsmasq
> > > # 进目录找到文件
> > > $ cat package/network/services/dnsmasq/files/dhcp.conf
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > 主要是
> > > ```
> > > config dhcp 'lan'
> > > 	option interface 'lan'
> > > 	option ignore '1'
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > 另一种方法是用 `uci-defaults`，也就是源码目录存放 `files/etc/uci-defaults/zzz-default-settings.sh` 一个shell文件，名字的字母让排序是最后，里面写
> > > ```
> > > uci set dhcp.lan.ignore='1'
> > > uci commit dhcp
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > 当然，最人性化的是 uci-defaults 的脚本里上面的逻辑外面加一个判断，如果只有一个网口才取消lan的dhcp
> > 
> > 
> > 您好呀，前面的我通过学习openwrt的官方文档我会了。但是我就是不会最后一个，就是，逻辑判断，人性化地判断网口的数量。
> > 我是这样设置的
> > ```shell
> > # 设置自定义dhcp
> > cat > $lede/lede/package/base-files/files/etc/uci-defaults/80_customized << "EOF"
> > 
> > uci add_list dhcp.lan.dhcp_option="6,119.29.29.29,223.5.5.5"
> > 
> > uci add dhcp domain
> > uci set dhcp.@domain[-1].name="xxxxxxx.com"
> > uci set dhcp.@domain[-1].ip="192.168.50.1"
> > uci delete dhcp.@dnsmasq[0].filter_aaaa
> > 
> > uci commit dhcp
> > 
> > ## uci delete dhcp.dnsmasq..option.filter_aaaa 这是自己的学习的记录
> > 
> > exit 0
> > 
> > EOF
> > ```
> 
> ```
> if xxx;then
>     uci set dhcp.lan.ignore='1'
>     uci commit dhcp
> fi
> ```
> 
> 至于判断逻辑，不依靠op编译的命令的话，可以 `/sys/class/net` 目录信息作为依据，下面这个命令就是打印网口信息
> 
> ```
> ls -l /sys/class/net/ | awk '$NF~/\/devices\/(platform|pci)/{print $9}'
> ```
> 
> 不过例如N1，情况特殊
> 
> ```
> root@OpenWrt:~# ls -l /sys/class/net/ | awk '$NF~/\/devices\/(platform|pci)/{print $9}'
> eth0
> wlan0
> root@OpenWrt:~# ls -l /sys/class/net/
> lrwxrwxrwx    1 root     root             0 Oct  1 05:03 bond0 -> ../../devices/virtual/net/bond0
> -rw-r--r--    1 root     root          4096 Oct  8 20:56 bonding_masters
> lrwxrwxrwx    1 root     root             0 Oct  1 05:03 docker0 -> ../../devices/virtual/net/docker0
> lrwxrwxrwx    1 root     root             0 Oct  1 05:03 dummy0 -> ../../devices/virtual/net/dummy0
> lrwxrwxrwx    1 root     root             0 Jan  1  1970 eth0 -> ../../devices/platform/soc/c9410000.ethernet/net/eth0
> lrwxrwxrwx    1 root     root             0 Jan  1  1970 ip6tnl0 -> ../../devices/virtual/net/ip6tnl0
> lrwxrwxrwx    1 root     root             0 Jan  1  1970 lo -> ../../devices/virtual/net/lo
> lrwxrwxrwx    1 root     root             0 Jan  1  1970 sit0 -> ../../devices/virtual/net/sit0
> lrwxrwxrwx    1 root     root             0 Oct  1 05:03 tunl0 -> ../../devices/virtual/net/tunl0
> lrwxrwxrwx    1 root     root             0 Oct  8 20:56 wg0 -> ../../devices/virtual/net/wg0
> lrwxrwxrwx    1 root     root             0 Oct  1 05:03 wlan0 -> ../../devices/platform/soc/d0000000.apb/d0070000.mmc/mmc_host/mmc0/mmc0:0001/mmc0:0001:1/net/wlan0
> ```
> 
> 下面是我的sft1200的信息
> 
> ```
> root@xxx:~# ls -l /sys/class/net/
> lrwxrwxrwx    1 root     root             0 Nov  2 09:37 br-lan -> ../../devices/virtual/net/br-lan
> lrwxrwxrwx    1 root     root             0 Nov  2 09:37 eth0 -> ../../devices/platform/10800000.gmac/net/eth0
> lrwxrwxrwx    1 root     root             0 Nov  2 09:37 eth0.1 -> ../../devices/virtual/net/eth0.1
> lrwxrwxrwx    1 root     root             0 Nov  2 09:37 eth0.2 -> ../../devices/virtual/net/eth0.2
> lrwxrwxrwx    1 root     root             0 Nov  2 09:37 ifb0 -> ../../devices/virtual/net/ifb0
> lrwxrwxrwx    1 root     root             0 Nov  2 09:37 ifb1 -> ../../devices/virtual/net/ifb1
> lrwxrwxrwx    1 root     root             0 Nov  2 09:37 lo -> ../../devices/virtual/net/lo
> lrwxrwxrwx    1 root     root             0 Nov  2 09:37 teql0 -> ../../devices/virtual/net/teql0
> lrwxrwxrwx    1 root     root             0 Nov  2 09:37 wg0 -> ../../devices/virtual/net/wg0
> lrwxrwxrwx    1 root     root             0 Nov  2 09:37 wlan0 -> ../../devices/platform/17800000.wifi-hb/net/wlan0
> lrwxrwxrwx    1 root     root             0 Nov  2 09:37 wlan1 -> ../../devices/platform/11000000.wifi-lb/net/wlan1
> root@xxx:~# ls -l /sys/class/net/ | awk '$NF~/\/devices\/(platform|pci)/{print $9}'
> eth0
> wlan0
> wlan1
> ```

就是这一段里面的xxx如何写，我不会xxx的这个表达式。
```shell
if xxx;then
    uci set dhcp.lan.ignore='1'
    uci commit dhcp
fi
```

zhangguanzhang, 2022-11-02:
linux的wc命令能统计行数，去学下shell

adamgonglj, 2022-11-02:
好的，感谢，我去查一下wc这个命令

adamgonglj, 2022-11-02:
可以用 lspci | grep eth命令

adamgonglj, 2022-11-02:
if [ lspci |grep -i eth | wc -l ]==1

adamgonglj, 2022-11-02:
if [ $(lspci | grep -i eth | wc -l) -eq 1 ] ; then
    uci set dhcp.lan.ignore='1'
    uci commit dhcp
fi

zhangguanzhang, 2022-11-02:
那如果固件没 `lspci` 命令呢

adamgonglj, 2022-11-02:
这个我再研究一下，实在没有，那就用openwrt的基本命令喽。

adamgonglj, 2022-11-02:
```shell
if [ $(ifconfig | grep eth | wc -l) -eq 1 ] ; then
uci set dhcp.lan.ignore='1'
uci commit dhcp
fi
```

zhangguanzhang, 2022-11-02:
如果网卡名不叫eth呢，可能是ens啥的

adamgonglj, 2022-11-02:
> 如果网卡名不叫eth呢，可能是ens啥的

那。。。。。哎，我就不懂了。

adamgonglj, 2022-11-02:
```shell
if [ $(ifconfig | grep -E "(eth|ens)" | wc -l) -eq 1 ] ; then
uci set dhcp.lan.ignore='1'
uci commit dhcp
fi
```

closed 2026-09-05 by coolsnowwolf