coolsnowwolf/lede #10346: 友善R5S LAN1不可用的问题

ufk119, 2022-11-01:
## 1.友善R5S，使用内核5.15（具体目前是5.15.74）编译后，在插上内置nvme固态后LAN1不可用，不差nvme时LAN1工作正常！ 官方固件是可以正常使用的。已经重新编译以及重刷多次，问题依旧。

ufk119, 2022-11-02:
麻烦大佬帮忙看下，开机日志：
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 5.15.74 (runner@fv-az377-726) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r5343-10f2bed40) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP PREEMPT Sat Oct 29 13:33:49 2022
[    0.000000] Machine model: FriendlyElec NanoPi R5S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 8 MiB at 0x00000000ef000000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 18 pages/cpu s33432 r8192 d32104 u73728
[    0.000000] pcpu-alloc: s33432 r8192 d32104 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] alternatives: patching kernel code
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 967176
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xfe660000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 3829924K/3930112K available (8704K kernel code, 2376K rwdata, 2468K rodata, 1664K init, 725K bss, 91996K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 320 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: Distributor has no Range Selector support
[    0.000000] GICv3: MBI range [296:319]
[    0.000000] GICv3: Using MBI frame 0x00000000fd410000
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: 16 PPIs implemented
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fd460000
[    0.000000] ITS: No ITS available, not enabling LPIs
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000000] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001384] Console: colour dummy device 80x25
[    0.001873] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.002925] pid_max: default: 32768 minimum: 301
[    0.003625] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.004382] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.007523] rcu: Hierarchical SRCU implementation.
[    0.008278] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.009807] smp: Bringing up secondary CPUs ...
[    0.010982] Detected VIPT I-cache on CPU1
[    0.011032] GICv3: CPU1: found redistributor 100 region 0:0x00000000fd480000
[    0.011092] CPU1: Booted secondary processor 0x0000000100 [0x412fd050]
[    0.011896] Detected VIPT I-cache on CPU2
[    0.011934] GICv3: CPU2: found redistributor 200 region 0:0x00000000fd4a0000
[... 457 lines cut ...]
[   39.707708] br-lan: port 1(eth1) entered disabled state
[   39.708370] device eth1 entered promiscuous mode
[   39.715355] eth2: 0xffff800009d10000, 62:57:de:50:73:6b, IRQ 60
[   39.772949] br-lan: port 2(eth2) entered blocking state
[   39.773431] br-lan: port 2(eth2) entered disabled state
[   39.774195] device eth2 entered promiscuous mode
[   39.779336] rk_gmac-dwmac fe2a0000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   39.871166] rk_gmac-dwmac fe2a0000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211F Gigabit Ethernet] (irq=67)
[   39.879118] dwmac4: Master AXI performs any burst length
[   39.879622] rk_gmac-dwmac fe2a0000.ethernet eth0: No Safety Features support found
[... cut ...]
[   66.349991] appfilter version:5.0
[   66.351463] init oaf sysctl...ok
[   84.205938] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[   84.206666] NFSD: Using legacy client tracking operations.
[   84.207153] NFSD: starting 10-second grace period (net f0000000)
[  111.217787] r8125: eth2: link up
[  111.218155] br-lan: port 2(eth2) entered blocking state
[  111.218627] br-lan: port 2(eth2) entered forwarding state
[  111.219726] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready

xiangfeidexiaohuo, 2022-11-03:
确实存在这个问题，等修复

iKenzo, 2022-11-03:
当固件使用 [DSA](https://openwrt.org/docs/guide-user/network/dsa/start) 时正常

ufk119, 2022-11-04:
> 当固件使用 [DSA](https://openwrt.org/docs/guide-user/network/dsa/start) 时正常

确实，试过多个固件，友善官方给的固件5.10内核的没问题，不过感觉其他方面不是很稳定，也不像自己编译的功能直接到位。 还试过两个内核4.几的，也没问题，但是内核感觉有点老，还行想自己编译。所以等修复。

iKenzo, 2022-11-06:
目前 iStoreOS 工作正常

ufk119, 2022-11-06:
> 目前 iStoreOS 工作正常

iStoreOS我试过，正常是正常，功能上不喜欢iStoreOS的某些功能，并且它的内核版本也是4点几的。

ufk119, 2022-11-16:
请各位大佬给点思路，硬件肯定是支持的，因为不插nvme时LAN1可用，使用iStoreOS和友善官方固件是没问题的，他们的内核4.19和5.10。 我尝试着编译5.10，好像脚本有一定问题，提示缺少dtb东西。

具体现象是插上网线，两头的黄灯会亮，绿灯不闪。获取不到ip，手动设置ip也ping不通。  需要更多日志我去搞。

qingtian110, 2022-11-16:
你可以试试编译6.0 内核。我现在在R5S上用的固件就是6.0 内核的，使用或者不使用nvme硬盘的情况下，三个网口都可以工作。

你可以测试下这个看看是不是你机器本身有问题  https://r5s.cooluc.com  Mainline 版本 6.0 内核的，虽然写着是试验，不过用着好像没啥问题

[image]
[image]
[image]
[image]

qingtian110, 2022-11-16:
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 6.0.8 (admin@cooluc.com) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r19803-9a599fee93) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP PREEMPT Fri Oct 14 22:44:41 2022
[    0.000000] Machine model: FriendlyElec NanoPi R5S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 6 MiB at 0x00000000ef200000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 20 pages/cpu s41000 r8192 d32728 u81920
[    0.000000] pcpu-alloc: s41000 r8192 d32728 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: kernel page table isolation forced OFF by mitigations=off
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: patching kernel code
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 967176
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xfe660000 root=PARTUUID=97a05a23-02 rw rootwait mitigations=off
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 3829196K/3930112K available (10432K kernel code, 2474K rwdata, 2936K rodata, 1920K init, 826K bss, 94772K reserved, 6144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 30 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 320 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: MBI range [296:319]
[    0.000000] GICv3: Using MBI frame 0x00000000fd410000
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fd460000
[    0.000000] ITS: No ITS available, not enabling LPIs
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000001] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001534] Console: colour dummy device 80x25
[    0.002027] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=80000)
[    0.003040] pid_max: default: 32768 minimum: 301
[    0.003712] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.004472] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.006377] spectre-v4 mitigation disabled by command-line option
[    0.007031] cacheinfo: Unable to detect cache hierarchy for CPU 0
[    0.008512] cblist_init_generic: Setting adjustable number of callback queues.
[    0.009270] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.009991] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.010842] rcu: Hierarchical SRCU implementation.
[    0.011317] rcu: 	Max phase no-delay instances is 1000.
[    0.012122] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[... 474 lines cut ...]
[   22.716892] rk_gmac-dwmac fe2a0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   22.717799] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   25.485514] appfilter version:5.0
[   25.487146] init oaf sysctl...ok
[  140.606453] rknic 0000:01:00.0 eth1: Link is Down
[  140.607087] br-lan: port 1(eth1) entered disabled state
[  143.317644] rknic 0001:01:00.0 eth2: Link is Up - 1Gbps/Full - flow control rx/tx
[  143.318420] br-lan: port 2(eth2) entered blocking state
[  143.318903] br-lan: port 2(eth2) entered forwarding state
[  157.846171] rknic 0001:01:00.0 eth2: Link is Down
[... cut ...]
[  166.205704] rknic 0001:01:00.0 eth2: Link is Up - 1Gbps/Full - flow control rx/tx
[  166.206477] br-lan: port 2(eth2) entered blocking state
[  166.206958] br-lan: port 2(eth2) entered forwarding state
[  167.747367] rknic 0001:01:00.0 eth2: Link is Down
[  167.748048] br-lan: port 2(eth2) entered disabled state
[  171.012666] rknic 0000:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[  171.013437] br-lan: port 1(eth1) entered blocking state
[  171.013918] br-lan: port 1(eth1) entered forwarding state
```

ufk119, 2022-11-17:
> 你可以试试编译6.0 内核。我现在在R5S上用的固件就是6.0 内核的，使用或者不使用nvme硬盘的情况下，三个网口都可以工作。
> 
> 你可以测试下这个看看是不是你机器本身有问题 https://r5s.cooluc.com Mainline 版本 6.0 内核的，虽然写着是试验，不过用着好像没啥问题
> 

好的，谢谢，我去试试。

ufk119, 2022-11-29:
目前使用lean大的源码直接编译，5.15和6.0内核下，这个问题依旧。我看楼上说的https://r5s.cooluc.com/里面的固件拿来测试了一下，这个问题确实是没有的，应该是用的openwrt源码编译的。这个bug虽然不太影响使用，但是觉得稍微有点膈应。

ufk119, 2022-11-29:
以下是ethtool 得到的信息：

root@OpenWrt:~# ethtool eth1
Settings for eth1:
	Supported ports: [ TP ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseT/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseT/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Full
	Auto-negotiation: on
	Port: Twisted Pair
	PHYAD: 0
	Transceiver: internal
	MDI-X: Unknown
	Supports Wake-on: pumbg
	Wake-on: g
        Current message level: 0x00000033 (51)
                               drv probe ifdown ifup
	Link detected: no
//-----------------------------------------------------------
root@OpenWrt:~# ethtool eth2
Settings for eth2:
	Supported ports: [ TP ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseT/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseT/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	                                     1000baseT/Full
	Link partner advertised pause frame use: Symmetric Receive-only
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Full
	Auto-negotiation: on
	Port: Twisted Pair
	PHYAD: 0
	Transceiver: internal
	MDI-X: Unknown
	Supports Wake-on: pumbg
	Wake-on: g
        Current message level: 0x00000033 (51)
                               drv probe ifdown ifup
	Link detected: yes

ufk119, 2022-11-29:
以下是ethtool     eth1 和eht2的信息，（eth1不可用，eth2 可用）

root@OpenWrt:~# ethtool eth1
Settings for eth1:
	Supported ports: [ TP ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseT/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseT/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Full
	Auto-negotiation: on
	Port: Twisted Pair
	PHYAD: 0
	Transceiver: internal
	MDI-X: Unknown
	Supports Wake-on: pumbg
	Wake-on: g
        Current message level: 0x00000033 (51)
                               drv probe ifdown ifup
	Link detected: no
root@OpenWrt:~# ethtool eth2
Settings for eth2:
	Supported ports: [ TP ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseT/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseT/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	                                     1000baseT/Full
	Link partner advertised pause frame use: Symmetric Receive-only
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Full
	Auto-negotiation: on
	Port: Twisted Pair
	PHYAD: 0
	Transceiver: internal
	MDI-X: Unknown
	Supports Wake-on: pumbg
	Wake-on: g
        Current message level: 0x00000033 (51)
                               drv probe ifdown ifup
	Link detected: yes

以下是lspci信息：
root@OpenWrt:~# lspci 
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0000:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0001:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0001:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0002:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0002:01:00.0 Non-Volatile memory controller: MAXIO Technology (Hangzhou) Ltd. NVMe SSD Controller MAP1202 (rev 01)

以下是ifconfig 的信息
eth1      Link encap:Ethernet  HWaddr 8A:B3:0F:B0:6A:6A  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth2      Link encap:Ethernet  HWaddr 8A:B3:0F:B0:6A:6A  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:13006793 errors:0 dropped:601 overruns:0 frame:0
          TX packets:22357948 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:6977131310 (6.4 GiB)  TX bytes:26165153585 (24.3 GiB)
          Interrupt:60

ufk119, 2022-11-29:
以下是ethtool eth1 和eht2的信息（eth1不可用，eth2 可用）

root@OpenWrt:  ethtool eth1
Settings for eth1:
Supported ports: [ TP ]
Supported link modes: 10baseT/Half 10baseT/Full
100baseT/Half 100baseT/Full
1000baseT/Full
2500baseT/Full
Supported pause frame use: Symmetric Receive-only
Supports auto-negotiation: Yes
Supported FEC modes: Not reported
Advertised link modes: 10baseT/Half 10baseT/Full
100baseT/Half 100baseT/Full
1000baseT/Full
2500baseT/Full
Advertised pause frame use: Symmetric Receive-only
Advertised auto-negotiation: Yes
Advertised FEC modes: Not reported
Speed: 1000Mb/s
Duplex: Full
Auto-negotiation: on
Port: Twisted Pair
PHYAD: 0
Transceiver: internal
MDI-X: Unknown
Supports Wake-on: pumbg
Wake-on: g
Current message level: 0x00000033 (51)
drv probe ifdown ifup
Link detected: no
root@OpenWrt:  ethtool eth2
Settings for eth2:
Supported ports: [ TP ]
Supported link modes: 10baseT/Half 10baseT/Full
100baseT/Half 100baseT/Full
1000baseT/Full
2500baseT/Full
Supported pause frame use: Symmetric Receive-only
Supports auto-negotiation: Yes
Supported FEC modes: Not reported
Advertised link modes: 10baseT/Half 10baseT/Full
100baseT/Half 100baseT/Full
1000baseT/Full
2500baseT/Full
Advertised pause frame use: Symmetric Receive-only
Advertised auto-negotiation: Yes
Advertised FEC modes: Not reported
Link partner advertised link modes: 10baseT/Half 10baseT/Full
100baseT/Half 100baseT/Full
1000baseT/Full
Link partner advertised pause frame use: Symmetric Receive-only
Link partner advertised auto-negotiation: Yes
Link partner advertised FEC modes: Not reported
Speed: 1000Mb/s
Duplex: Full
Auto-negotiation: on
Port: Twisted Pair
PHYAD: 0
Transceiver: internal
MDI-X: Unknown
Supports Wake-on: pumbg
Wake-on: g
Current message level: 0x00000033 (51)
drv probe ifdown ifup
Link detected: yes

以下是lspci信息：
root@OpenWrt:~# lspci
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0000:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0001:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0001:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0002:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3568 Remote Signal Processor (rev 01)
0002:01:00.0 Non-Volatile memory controller: MAXIO Technology (Hangzhou) Ltd. NVMe SSD Controller MAP1202 (rev 01)

以下是ifconfig 的信息
eth1 Link encap:Ethernet HWaddr 8A:B3:0F:B0:6A:6A
UP BROADCAST MULTICAST MTU:1500 Metric:1
RX packets:0 errors:0 dropped:0 overruns:0 frame:0
TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000
RX bytes:0 (0.0 B) TX bytes:0 (0.0 B)

eth2 Link encap:Ethernet HWaddr 8A:B3:0F:B0:6A:6A
UP BROADCAST RUNNING MULTICAST MTU:1500 Metric:1
RX packets:13006793 errors:0 dropped:601 overruns:0 frame:0
TX packets:22357948 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000
RX bytes:6977131310 (6.4 GiB) TX bytes:26165153585 (24.3 GiB)
Interrupt:60

ufk119, 2022-12-07:
问题目前未解决，也没人关注，就此关闭。有人解决了叫我一声，谢谢啦。

ufk119, 2022-12-09:
> 当固件使用 [DSA](https://openwrt.org/docs/guide-user/network/dsa/start) 时正常

老哥，我试过openwrt原版是可以使用dsa的，我也尝试着在现有的版本中使用dsa去配置/etc/config/network,但是都不成功。
你有方法（编译或安装软件）能够在现有版本上使用上dsa吗？Thanks♪(･ω･)ﾉ

donbowman, 2022-12-13:
it might be the 40-net-smp-affinity script.

closed 2022-12-15 by ufk119

ufk119, 2023-01-01:
结案！  
今天2023-1-1编译的固件，在插上nvme的情况下，中间的lan1口也是可用的。
  大概率是从 本次修复后OK的。[c4be33b792e926c7c83ae41aa704d0010de0efb5](https://github.com/coolsnowwolf/lede/commit/c4be33b792e926c7c83ae41aa704d0010de0efb5)
谢谢大佬们。