coolsnowwolf/lede #10348: x86 编译失败,求指教

FormatToday, 2022-11-01:
## 1.关于要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：未复制完全，从后往前截取了一段

```log
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/__init__.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/_msvccompiler.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/archive_util.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/bcppcompiler.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/ccompiler.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/cmd.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/config.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/core.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/cygwinccompiler.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/debug.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/dep_util.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/dir_util.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/dist.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/errors.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/extension.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/fancy_getopt.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/file_util.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/filelist.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/log.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/msvc9compiler.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/msvccompiler.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/spawn.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/sysconfig.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/text_file.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/unixccompiler.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/util.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/version.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/versionpredicate.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/__init__.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/bdist.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/bdist_dumb.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/bdist_msi.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/bdist_rpm.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/bdist_wininst.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/build.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/build_clib.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/build_ext.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/build_py.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/build_scripts.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/check.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/clean.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/command_template /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/config.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/install.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/install_data.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/install_egg_info.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/install_headers.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/install_lib.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/install_scripts.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./Lib/distutils/command/register.py /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/distutils/command
[... 197 lines cut ...]
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 `cat pybuilddir.txt`/_sysconfigdata*.py \
	/home/gavin/lede/staging_dir/hostpkg/lib/python3.9; \
/home/gavin/lede/staging_dir/host/bin/install -c -m 644 ./LICENSE /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/LICENSE.txt
if test "xupgrade" != "xno"  ; then \
	case upgrade in \
		upgrade) ensurepip="--upgrade" ;; \
		install|*) ensurepip="" ;; \
	esac; \
	 ./python -E -m ensurepip \
		$ensurepip --root=/ ; \
fi
Looking in links: /home/gavin/lede/tmp/tmpog3_zya7
Requirement already satisfied: setuptools in /home/gavin/lede/staging_dir/hostpkg/lib/python3.9/site-packages (58.1.0)
[... cut ...]
touch /home/gavin/lede/build_dir/hostpkg/Python-3.9.15/.built
touch /home/gavin/lede/staging_dir/hostpkg/stamp/.python3_installed
make[3]: Leaving directory '/home/gavin/lede/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/host-compile#123.46#21.35#112.74
make[2]: Leaving directory '/home/gavin/lede'
make[1]: *** [package/Makefile:110: /home/gavin/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gavin/lede'
make: *** [/home/gavin/lede/include/toplevel.mk:230：world] 错误 2
```

ecrasy, 2022-11-01:
解决办法有两个，你可以试试：
1. 使用选项 IGNORE_ERRORS="m n" 忽略掉这个错误，用法：make -j$(nproc) IGNORE_ERRORS="m n"
2. 打补丁 [patch](https://github.com/ecrasy/BuildOpenWrt/blob/main/data/patches/lib-platform-sys-version.patch)  
  用法：cp lib-platform-sys-version.patch feeds/packages/lang/python/python3/patches/

aliuq, 2022-11-01:
@ecrasy 方法2试了无效，还是一样的错误提示

closed 2026-09-05 by coolsnowwolf