coolsnowwolf/lede #10350: 极路由B70 mt7621编译失败，求教

lee29, 2022-11-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志
`Run cd openwrt
 make[1] dirclean
2 thread compile
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'gcc'... ok.
Checking 'working-gcc'... ok.
Checking 'g++'... ok.
Checking 'working-g++'... ok.
Checking 'ncurses'... ok.
Checking 'perl-data-dumper'... ok.
Checking 'perl-findbin'... ok.
Checking 'perl-file-copy'... ok.
Checking 'perl-file-compare'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'xargs'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'egrep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'install'... ok.
Checking 'perl'... ok.
Checking 'python2-cleanup'... ok.
Checking 'python'... ok.
Checking 'python3'... ok.
Checking 'python3-distutils'... ok.
Checking 'git'... ok.
Checking 'file'... ok.
Checking 'rsync'... ok.
Checking 'which'... ok.
Checking 'ldconfig-stub'... ok.

make[2]: Entering directory '/workdir/openwrt/scripts/config'
cc -O2    -c -o conf.o conf.c
cc -O2    -c -o confdata.o confdata.c
cc -O2    -c -o expr.o expr.c
cc -O2 -I ./.   -c -o lexer.lex.o lexer.lex.c
cc -O2    -c -o menu.o menu.c
cc -O2 -I ./.   -c -o parser.tab.o parser.tab.c
cc -O2    -c -o preprocess.o preprocess.c
[... 40 lines cut ...]
touch /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.221/.configured
rm -f /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.221/vmlinux /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.221/System.map
make -C /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.221 KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Nov  1 14:08:09 2022" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.221 CC="mipsel-openwrt-linux-musl-gcc" olddefconfig
make[5]: Entering directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.221'
/bin/sh: 1: mipsel-openwrt-linux-musl-gcc: not found
/bin/sh: 1: mipsel-openwrt-linux-musl-gcc: not found
  LEX     scripts/kconfig/lexer.lex.c
flex: fatal internal error, exec of /workdir/openwrt/staging_dir/host/bin/m4 failed
flex: error writing output file scripts/kconfig/lexer.lex.c
make[6]: *** [scripts/Makefile.host:9: scripts/kconfig/lexer.lex.c] Error 1
make[5]: *** [Makefile:590: olddefconfig] Error 2
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.221'
make[4]: *** [Makefile:24: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.221/.modules] Error 2
make[4]: Leaving directory '/workdir/openwrt/target/linux/ramips'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/compile#0.[41](https://github.com/lee29/P3TERX-B70/actions/runs/3371410541/jobs/5593595872#step:13:42)#0.13#0.[52](https://github.com/lee29/P3TERX-B70/actions/runs/3371410541/jobs/5593595872#step:13:53)
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:23: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.`
A：

hzjnet, 2022-11-01:
仔细看看log:
> /bin/sh: 1: mipsel-openwrt-linux-musl-gcc: not found
> /bin/sh: 1: mipsel-openwrt-linux-musl-gcc: not found
明显没有按照readme执行

lee29, 2022-11-02:
> 仔细看看log:
> 
> > /bin/sh: 1: mipsel-openwrt-linux-musl-gcc: not found
> > /bin/sh: 1: mipsel-openwrt-linux-musl-gcc: not found
> > 明显没有按照readme执行

改了之后一样，最后发现用20221001这一版就没问题，master的就不行。。

coolsnowwolf, 2022-11-03:
彻底删除后重新 clone 就好了，缓存问题

closed 2022-11-13 by coolsnowwolf