coolsnowwolf/lede #10357: 各位德华彦祖帮看看有个网口老断流重启是怎么回事

zhangdp, 2022-11-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 有类似的issue但我不确定是否一样的问题

## 2. 详细叙述

### (1) 具体问题

A：i2250v B3有一个网口eth1经常性的断开重启，我应该怎么排查，难道这就是i225-v B3本身的断流问题？还是说是固件问题还是软件不兼容？有搜到关闭sfe的，尝试关闭了好像依旧。

### (2) 路由器型号和固件版本

A：j4125 4*i225-v B3
内核5.15.41

### (3) 详细日志

A：
系统日志：
Thu Nov  3 15:08:40 2022 kern.info kernel: [ 2125.350392] igc 0000:02:00.0 eth1: NIC Link is Down
Thu Nov  3 15:08:40 2022 kern.info kernel: [ 2125.355902] br-lan: port 1(eth1) entered disabled state
Thu Nov  3 15:08:40 2022 daemon.notice netifd: Network device 'eth1' link is down
Thu Nov  3 15:08:43 2022 kern.info kernel: [ 2128.314602] igc 0000:02:00.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
Thu Nov  3 15:08:44 2022 kern.info kernel: [ 2128.986349] br-lan: port 1(eth1) entered blocking state
Thu Nov  3 15:08:44 2022 kern.info kernel: [ 2128.991985] br-lan: port 1(eth1) entered forwarding state
Thu Nov  3 15:08:44 2022 daemon.notice netifd: Network device 'eth1' link is up
Thu Nov  3 15:12:26 2022 kern.info kernel: [ 2351.398229] igc 0000:02:00.0 eth1: NIC Link is Down
Thu Nov  3 15:12:26 2022 kern.info kernel: [ 2351.403776] br-lan: port 1(eth1) entered disabled state
Thu Nov  3 15:12:26 2022 daemon.notice netifd: Network device 'eth1' link is down
Thu Nov  3 15:12:29 2022 kern.info kernel: [ 2354.182439] igc 0000:02:00.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
Thu Nov  3 15:12:30 2022 kern.info kernel: [ 2354.846191] br-lan: port 1(eth1) entered blocking state
Thu Nov  3 15:12:30 2022 kern.info kernel: [ 2354.851803] br-lan: port 1(eth1) entered forwarding state
Thu Nov  3 15:12:30 2022 daemon.notice netifd: Network device 'eth1' link is up
Thu Nov  3 15:13:15 2022 kern.info kernel: [ 2400.297439] igc 0000:02:00.0 eth1: NIC Link is Down
Thu Nov  3 15:13:15 2022 kern.info kernel: [ 2400.302913] br-lan: port 1(eth1) entered disabled state
Thu Nov  3 15:13:15 2022 daemon.notice netifd: Network device 'eth1' link is down
Thu Nov  3 15:13:19 2022 kern.info kernel: [ 2403.749646] igc 0000:02:00.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
Thu Nov  3 15:13:19 2022 kern.info kernel: [ 2403.758875] br-lan: port 1(eth1) entered blocking state
Thu Nov  3 15:13:19 2022 kern.info kernel: [ 2403.764506] br-lan: port 1(eth1) entered forwarding state
Thu Nov  3 15:13:19 2022 daemon.notice netifd: Network device 'eth1' link is up
Thu Nov  3 15:13:49 2022 kern.info kernel: [ 2434.096903] igc 0000:02:00.0 eth1: NIC Link is Down
Thu Nov  3 15:13:49 2022 kern.info kernel: [ 2434.102374] br-lan: port 1(eth1) entered disabled state
Thu Nov  3 15:13:49 2022 daemon.notice netifd: Network device 'eth1' link is down
Thu Nov  3 15:13:52 2022 kern.info kernel: [ 2437.381113] igc 0000:02:00.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
Thu Nov  3 15:13:52 2022 kern.info kernel: [ 2437.390305] br-lan: port 1(eth1) entered blocking state
Thu Nov  3 15:13:52 2022 kern.info kernel: [ 2437.395904] br-lan: port 1(eth1) entered forwarding state
Thu Nov  3 15:13:52 2022 daemon.notice netifd: Network device 'eth1' link is up

内核日志：
[ 1954.618440] igc 0000:02:00.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[ 1954.627664] br-lan: port 1(eth1) entered blocking state
[ 1954.633284] br-lan: port 1(eth1) entered forwarding state
[ 1965.357922] igc 0000:02:00.0 eth1: NIC Link is Down
[ 1965.363383] br-lan: port 1(eth1) entered disabled state
[ 1968.210112] igc 0000:02:00.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[ 1968.881863] br-lan: port 1(eth1) entered blocking state
[ 1968.887442] br-lan: port 1(eth1) entered forwarding state
[ 2125.350392] igc 0000:02:00.0 eth1: NIC Link is Down
[ 2125.355902] br-lan: port 1(eth1) entered disabled state
[ 2128.314602] igc 0000:02:00.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[ 2128.986349] br-lan: port 1(eth1) entered blocking state
[ 2128.991985] br-lan: port 1(eth1) entered forwarding state
[ 2351.398229] igc 0000:02:00.0 eth1: NIC Link is Down
[ 2351.403776] br-lan: port 1(eth1) entered disabled state
[ 2354.182439] igc 0000:02:00.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[ 2354.846191] br-lan: port 1(eth1) entered blocking state
[ 2354.851803] br-lan: port 1(eth1) entered forwarding state
[ 2400.297439] igc 0000:02:00.0 eth1: NIC Link is Down
[ 2400.302913] br-lan: port 1(eth1) entered disabled state
[ 2403.749646] igc 0000:02:00.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[ 2403.758875] br-lan: port 1(eth1) entered blocking state
[ 2403.764506] br-lan: port 1(eth1) entered forwarding state
[ 2434.096903] igc 0000:02:00.0 eth1: NIC Link is Down
[ 2434.102374] br-lan: port 1(eth1) entered disabled state
[ 2437.381113] igc 0000:02:00.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[ 2437.390305] br-lan: port 1(eth1) entered blocking state
[ 2437.395904] br-lan: port 1(eth1) entered forwarding state

zhangdp, 2022-11-03:
要如何查看网卡是不是b1 b2冒充的b3版本？

coolsnowwolf, 2022-11-03:
225 B2 受害者

closed 2026-09-05 by coolsnowwolf