coolsnowwolf/lede #10362: xunlong_orangepi-r1-plus-lts 无法编译通过

yanght1001, 2022-11-03:
无法编译编译通过，
我的操作是仅仅选择了cpu架构（xunlong_orangepi-r1-plus-lts），其他都是默认，无法编译通过

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* 有相似，ID #9757，但是没有说明如何解决。

## 2. 详细叙述

### (1) 具体问题

A：编译无法通过

### (2) 路由器型号和固件版本

A：xunlong_orangepi-r1-plus-lts

### (3) 详细日志

A：
 LD      u-boot
  OBJCOPY u-boot.srec
  OBJCOPY u-boot-nodtb.bin
  RELOC   u-boot-nodtb.bin
  CAT     u-boot-dtb.bin
  COPY    u-boot.bin
  SYM     u-boot.sym
  MKIMAGE u-boot.img
  CC      tpl/drivers/ram/rockchip/sdram_rk3328.o
drivers/ram/rockchip/sdram_rk3328.c: In function 'rk3328_dmc_init':
drivers/ram/rockchip/sdram_rk3328.c:525:55: error: 'struct dtd_rockchip_rk3328_dmc' has no member named 'rockchip_sdram_params'
  525 |                                         (void *)dtplat->rockchip_sdram_params;
      |                                                       ^~
make[8]: *** [scripts/Makefile.build:258: tpl/drivers/ram/rockchip/sdram_rk3328.o] Error 1
make[7]: *** [scripts/Makefile.build:398: tpl/drivers/ram/rockchip] Error 2
make[6]: *** [scripts/Makefile.build:398: tpl/drivers/ram] Error 2
make[5]: *** [scripts/Makefile.spl:531: tpl/drivers] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/u-boot-orangepi-r1-plus-lts-rk3328/u-boot-2022.07'
make[3]: Leaving directory '/workdir/openwrt/package/boot/uboot-rockchip'
make[4]: *** [Makefile:2109: tpl/u-boot-tpl] Error 2
make[3]: *** [Makefile:258: /workdir/openwrt/build_dir/target-aarch64_generic_musl/u-boot-orangepi-r1-plus-lts-rk3328/u-boot-2022.07/.built] Error 2
time: package/boot/uboot-rockchip/orangepi-r1-plus-lts-rk3328/compile#2.73#0.95#3.55
    ERROR: package/boot/uboot-rockchip failed to build (build variant: orangepi-r1-plus-lts-rk3328).
make[2]: *** [package/Makefile:116: package/boot/uboot-rockchip/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

hong0980, 2022-11-09:
唉这个冷门机型大佬们不修复，我换了很多uboot-rockchip都不成功，10月底前用immortalwrt的还能用。

yanght1001, 2022-11-09:
> 唉这个冷门机型大佬们不修复，我换了很多uboot-rockchip都不成功，10月底前用immortalwrt的还能用。

我最后还是用immortalwrt编译成功了，大佬很忙，但是有错误要解决，要么在固件选择里面把这个删除了，以免误导呀。

1715173329, 2022-11-10:
My code was raped, so good luck.

BoBppy, 2022-11-30:
> My code was raped, so good luck.

what??

closed 2026-09-05 by coolsnowwolf