coolsnowwolf/lede #10366: 添加"OUI"后，编译出错了，请教解决方法！谢谢！！

Dboykey, 2022-11-04:
make[3]: Entering directory '/home/siva/lede/5.4/feeds/oui/oui-ui-core'
rm -f /home/siva/lede/5.4/build_dir/target-x86_64_musl/oui-ui-core-git-2022.306.12938-3a345e9/.built
touch /home/siva/lede/5.4/build_dir/target-x86_64_musl/oui-ui-core-git-2022.306.12938-3a345e9/.built_check
/home/siva/lede/5.4/staging_dir/hostpkg/bin/npm --prefix /home/siva/lede/5.4/build_dir/target-x86_64_musl/oui-ui-core-git-2022.306.12938-3a345e9/htdoc run build
npm info it worked if it ends with ok
npm info using npm@6.14.12
npm info using node@v12.22.1
npm info lifecycle oui-ui-core@~prebuild: oui-ui-core@
npm info lifecycle oui-ui-core@~build: oui-ui-core@

> oui-ui-core@ build /home/siva/lede/5.4/build_dir/target-x86_64_musl/oui-ui-core-git-2022.306.12938-3a345e9/htdoc
> vite build

file:///home/siva/lede/5.4/build_dir/target-x86_64_musl/oui-ui-core-git-2022.306.12938-3a345e9/htdoc/node_modules/vite/bin/vite.js:7
    await import('source-map-support').then((r) => r.default.install())
    ^^^^^

SyntaxError: Unexpected reserved word
    at Loader.moduleStrategy (internal/modules/esm/translators.js:140:18)
    at async link (internal/modules/esm/module_job.js:42:21)
npm info lifecycle oui-ui-core@~build: Failed to exec build script
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! oui-ui-core@ build: `vite build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the oui-ui-core@ build script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm timing npm Completed in 250ms

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/siva/.npm/_logs/2022-11-03T14_53_33_749Z-debug.log
make[3]: *** [Makefile:89: /home/siva/lede/5.4/build_dir/target-x86_64_musl/oui-ui-core-git-2022.306.12938-3a345e9/.built] Error 1
make[3]: Leaving directory '/home/siva/lede/5.4/feeds/oui/oui-ui-core'
time: package/feeds/oui/oui-ui-core/compile#0.69#0.17#1.01
    ERROR: package/feeds/oui/oui-ui-core failed to build.
make[2]: *** [package/Makefile:116: package/feeds/oui/oui-ui-core/compile] Error 1
make[2]: Leaving directory '/home/siva/lede/5.4'
make[1]: *** [package/Makefile:110: /home/siva/lede/5.4/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/siva/lede/5.4'
make: *** [/home/siva/lede/5.4/include/toplevel.mk:230：world] 错误 2

aiamadeus, 2022-11-04:
@zhaojh329

zhaojh329, 2022-11-04:
编译 Oui 需要用到 Node，而且版本不能低于 14.18。

Dboykey, 2022-11-04:
@zhaojh329
谢谢大佬的指点，我查了一下，我的nodejs版本是(12.22.12~dfsg-1~deb11u1)
应该就是这个问题了，感谢感谢

moonerick, 2022-11-05:
> @zhaojh329 谢谢大佬的指点，我查了一下，我的nodejs版本是(12.22.12 ~ dfsg-1 ~ deb11u1) 应该就是这个问题了，感谢感谢

解决了么，我用node -v查看，系统安装的版本是19.0.01，错误和你这个完全是一样的

zhaojh329, 2022-11-05:
https://github.com/coolsnowwolf/lede/pull/10375/files

zhaojh329, 2022-11-05:
选中
```
 [*] Use existing nodejs installation on the host system
```
然后
```
rm staging_dir/hostpkg/bin/node
```

Dboykey, 2022-11-06:
我是这样处理的：
$ make menuconfig
Languages -> Node.js -> Configuration -> Version Selection -> 勾选16或18

然后
$ rm staging_dir/hostpkg/bin/node

Dboykey, 2022-11-06:
@zhaojh329 大佬，请问“ [*] Use existing nodejs installation on the host system”这项在那层里面的？我怎么都找不到？

我在上面的哪里选了node 18后，编译哪里是过了，最后封装的时候过不了。。。

hxlls, 2022-11-06:
我一直有个疑问，支持是不是兼容之前的luci 的IPK？如果不兼容，这个是不是很多应用需要重写？我看了文档，好像没找到这个问题的答案

zhaojh329, 2022-11-06:
> 我一直有个疑问，支持是不是兼容之前的luci 的IPK？如果不兼容，这个是不是很多应用需要重写？我看了文档，好像没找到这个问题的答案

 不兼容 luci

zhaojh329, 2022-11-06:
> @zhaojh329 大佬，请问“ [*] Use existing nodejs installation on the host system”这项在那层里面的？我怎么都找不到？
> 
> 我在上面的哪里选了node 18后，编译哪里是过了，最后封装的时候过不了。。。

要么使用 主机自带的 node，要么使用 OpenWrt 的 node。
```
Oui  --->
    [*] Use existing nodejs installation on the host system
```
勾选这个表示使用主机自带的 Node。但如果 OpenWr 之前编译过 node，需要删除
```
rm staging_dir/hostpkg/bin/node
```

如果不勾选那个选项，即使用 OpenWrt 的 Node，可以这样编译：
```
export CONFIG_NODEJS_18=y
make -j4
```

zhaojh329, 2022-11-06:
我刚刚已经修复了 oui。如果 openwrt 之前编译过 node，不再需要删除已编译的 node 了。更新下 oui。
```
./scripts/feeds update oui
```

Dboykey, 2022-11-07:
@zhaojh329 大佬，看到你上面说“不兼容 luci”，我编译了几次都在最后封装那步失败，出错信息如下：
Collected errors:
 * check_data_file_clashes: Package oui-ui-core wants to install file /home/siva/lede/5.4/build_dir/target-x86_64_musl/root-x86/www/index.html
        But that file is already provided by package  * luci-base
 * opkg_install_cmd: Cannot install package oui-app-acl.
 * check_data_file_clashes: Package oui-ui-core wants to install file /home/siva/lede/5.4/build_dir/target-x86_64_musl/root-x86/www/index.html
        But that file is already provided by package  * luci-base
 * opkg_install_cmd: Cannot install package oui-app-backup.
 * check_data_file_clashes: Package oui-ui-core wants to install file /home/siva/lede/5.4/build_dir/target-x86_64_musl/root-x86/www/index.html
        But that file is already provided by package  * luci-base
 * opkg_install_cmd: Cannot install package oui-app-demo.
 * check_data_file_clashes: Package oui-ui-core wants to install file /home/siva/lede/5.4/build_dir/target-x86_64_musl/root-x86/www/index.html
        But that file is already provided by package  * luci-base
 * opkg_install_cmd: Cannot install package oui-app-dhcp-lease.
 * check_data_file_clashes: Package oui-ui-core wants to install file /home/siva/lede/5.4/build_dir/target-x86_64_musl/root-x86/www/index.html
        But that file is already provided by package  * luci-base
 * opkg_install_cmd: Cannot install package oui-app-home.
 * check_data_file_clashes: Package oui-ui-core wants to install file /home/siva/lede/5.4/build_dir/target-x86_64_musl/root-x86/www/index.html
        But that file is already provided by package  * luci-base
 * opkg_install_cmd: Cannot install package oui-app-layout.
 * check_data_file_clashes: Package oui-ui-core wants to install file /home/siva/lede/5.4/build_dir/target-x86_64_musl/root-x86/www/index.html
        But that file is already provided by package  * luci-base
 * opkg_install_cmd: Cannot install package oui-app-login.
 * check_data_file_clashes: Package oui-ui-core wants to install file /home/siva/lede/5.4/build_dir/target-x86_64_musl/root-x86/www/index.html
        But that file is already provided by package  * luci-base
 * opkg_install_cmd: Cannot install package oui-app-stations.
 * check_data_file_clashes: Package oui-ui-core wants to install file /home/siva/lede/5.4/build_dir/target-x86_64_musl/root-x86/www/index.html
        But that file is already provided by package  * luci-base
 * opkg_install_cmd: Cannot install package oui-app-system.
 * check_data_file_clashes: Package oui-ui-core wants to install file /home/siva/lede/5.4/build_dir/target-x86_64_musl/root-x86/www/index.html
        But that file is already provided by package  * luci-base
 * opkg_install_cmd: Cannot install package oui-app-upgrade.
 * check_data_file_clashes: Package oui-ui-core wants to install file /home/siva/lede/5.4/build_dir/target-x86_64_musl/root-x86/www/index.html
        But that file is already provided by package  * luci-base
 * opkg_install_cmd: Cannot install package oui-app-user.
 * check_data_file_clashes: Package oui-ui-core wants to install file /home/siva/lede/5.4/build_dir/target-x86_64_musl/root-x86/www/index.html
        But that file is already provided by package  * luci-base
 * opkg_install_cmd: Cannot install package oui-ui-core.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/siva/lede/5.4'
make[1]: *** [package/Makefile:111: /home/siva/lede/5.4/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/siva/lede/5.4'
make: *** [/home/siva/lede/5.4/include/toplevel.mk:230：world] 错误 2

请问需要怎样修改配置才可以通过编译？谢谢

zhaojh329, 2022-11-07:
不要勾选 luci-base

zhangguanzhang, 2022-11-23:
@zhaojh329 对于其他依赖node的，也可以使用主机自带的 node吗

zhaojh329, 2022-11-23:
> @zhaojh329 对于其他依赖node的，也可以使用主机自带的 node吗

oui 之所以可以使用主机自带得 node，是因为 oui 做了特别得处理。

g93920079, 2022-11-24:
> 不要勾选 luci-base 和 uhttpd

请问大佬，这个在哪边设置，找了半天没找到哪个位置。。

LXGMAX, 2023-02-03:
> > 不要勾选 luci-base 和 uhttpd
> 
> 请问大佬，这个在哪边设置，找了半天没找到哪个位置。。

LUCI那里，依赖luci的都要取消勾选，然后找到luci-base和uhttpd取消勾选

coolsnowwolf, 2026-09-05:
该错误来自 2022 年第三方 `oui-ui-core` 使用 Node.js 12 执行 Vite，新版 Node.js 包已提供更新版本；旧构建环境不足以证明当前仓库缺陷，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf