coolsnowwolf/lede #10372: Redmi AX6000 PPPOE断网重连多次才成功

ramondelee, 2022-11-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述
Redmi AX6000 PPPOE连接5天左右会断网重连多次才成功。（本地ISP是7天强制删除重连）
### (1) 具体问题

A：AX6000的PPPOE是否有问题、请分析一下？

### (2) 路由器型号和固件版本

A：Redmi AX6000 LEDE改兼容官方分区自编译

### (3) 详细日志

A：
Sun Oct 30 09:06:30 2022 daemon.notice netifd: Modem (18857): udhcpc: broadcasting discover
Sun Oct 30 09:06:30 2022 daemon.notice pppd[18866]: pppd 2.4.9 started by root, uid 0
Sun Oct 30 09:06:33 2022 daemon.notice netifd: Modem (18857): udhcpc: broadcasting discover
Sun Oct 30 09:06:35 2022 daemon.info pppd[18866]: PPP session is 48000
Sun Oct 30 09:06:35 2022 daemon.notice netifd: Modem (18857): udhcpc: broadcasting select for 192.168.1.2, server 192.168.1.1
Sun Oct 30 09:06:35 2022 daemon.warn pppd[18866]: Connected to 60:08:10:09:c8:30 via interface wan
Sun Oct 30 09:06:35 2022 kern.info kernel: [130370.497503] pppoe-wan: renamed from ppp0
Sun Oct 30 09:06:35 2022 daemon.notice netifd: Modem (18857): udhcpc: lease of 192.168.1.2 obtained from 192.168.1.1, lease time 259200
Sun Oct 30 09:06:35 2022 daemon.info pppd[18866]: Renamed interface ppp0 to pppoe-wan
Sun Oct 30 09:06:35 2022 daemon.info pppd[18866]: Using interface pppoe-wan
Sun Oct 30 09:06:35 2022 daemon.notice pppd[18866]: Connect: pppoe-wan <--> wan
Sun Oct 30 09:06:35 2022 daemon.notice netifd: Interface 'Modem' is now up
Sun Oct 30 09:06:38 2022 daemon.info pppd[18866]: syncppp not active
Sun Oct 30 09:06:38 2022 daemon.info pppd[18866]: CHAP authentication succeeded: Authentication success,Welcome!
Sun Oct 30 09:06:38 2022 daemon.notice pppd[18866]: CHAP authentication succeeded
Sun Oct 30 09:06:38 2022 daemon.notice pppd[18866]: peer from calling number 60:08:10:09:C8:30 authorized
Sun Oct 30 09:06:38 2022 daemon.notice pppd[18866]: local  LL address fe80::5d5b:d0cb:cb05:a6d3
Sun Oct 30 09:06:38 2022 daemon.notice pppd[18866]: remote LL address fe80::6208:10ff:fe09:c830
Sun Oct 30 09:06:38 2022 daemon.notice netifd: Network device 'pppoe-wan' link is up
Sun Oct 30 09:06:38 2022 daemon.notice netifd: Interface 'wan' is now up
Sun Oct 30 09:06:38 2022 daemon.notice pppd[18866]: local  IP address 113.205.238.30
Sun Oct 30 09:06:38 2022 daemon.notice pppd[18866]: remote IP address 113.205.236.1
Sun Oct 30 09:06:38 2022 daemon.notice pppd[18866]: primary   DNS address 221.7.92.98
Sun Oct 30 09:06:38 2022 daemon.notice pppd[18866]: secondary DNS address 221.5.203.98
Sun Oct 30 09:06:38 2022 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Sun Oct 30 09:06:38 2022 daemon.notice netifd: Interface 'wan_6' is enabled
Sun Oct 30 09:06:38 2022 daemon.notice netifd: Interface 'wan_6' has link connectivity
Sun Oct 30 09:06:38 2022 daemon.notice netifd: Interface 'wan_6' is setting up now
Sun Oct 30 09:06:41 2022 daemon.notice netifd: Interface 'wan_6' is now up
Sun Oct 30 09:06:45 2022 user.notice firewall: Reloading firewall due to ifup of Modem (wan)
Sun Oct 30 09:06:46 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of Modem (wan)
Sun Oct 30 09:06:56 2022 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Sun Oct 30 09:06:56 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan (pppoe-wan)
Sun Oct 30 09:07:07 2022 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Sun Oct 30 09:07:07 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan_6 (pppoe-wan)
Sun Oct 30 09:36:53 2022 daemon.notice miniupnpd[7718]: remove port mapping 40719 UDP because it has expired
Sun Oct 30 09:36:53 2022 daemon.notice miniupnpd[7718]: remove port mapping 40719 TCP because it has expired
Sun Oct 30 09:50:53 2022 daemon.warn miniupnpd[7718]: upnp_event_process_notify: connect(192.168.12.209:2869): Operation timed out
Sun Oct 30 09:50:58 2022 daemon.warn miniupnpd[7718]: upnp_event_process_notify: connect(192.168.12.209:2869): Operation timed out
Sun Oct 30 10:18:54 2022 daemon.notice miniupnpd[7718]: remove port mapping 40720 UDP because it has expired
Sun Oct 30 10:18:54 2022 daemon.notice miniupnpd[7718]: remove port mapping 40720 TCP because it has expired
Sun Oct 30 11:50:09 2022 cron.err crond[24723]: crond (busybox 1.35.0) started, log level 5
Sun Oct 30 11:54:58 2022 cron.err crond[26106]: crond (busybox 1.35.0) started, log level 5
Sun Oct 30 11:55:51 2022 cron.err crond[27208]: crond (busybox 1.35.0) started, log level 5
Sun Oct 30 19:14:02 2022 daemon.notice miniupnpd[7718]: remove port mapping 34487 UDP because it has expired
Mon Oct 31 01:00:00 2022 cron.err crond[27208]: USER root pid 15139 cmd echo "" > /tmp/vssr.log
Mon Oct 31 02:00:00 2022 cron.err crond[27208]: USER root pid 16553 cmd /usr/bin/lua /usr/share/vssr/subscribe.lua
Mon Oct 31 02:00:00 2022 cron.err crond[27208]: USER root pid 16554 cmd /usr/bin/lua /usr/share/vssr/update.lua
Mon Oct 31 21:06:35 2022 daemon.notice netifd: Modem (18857): udhcpc: sending renew to server 192.168.1.1
Mon Oct 31 21:06:35 2022 daemon.notice netifd: Modem (18857): udhcpc: lease of 192.168.1.2 obtained from 192.168.1.1, lease time 259200
[... 643 lines cut ...]
Thu Nov  3 12:59:50 2022 daemon.info pppd[19239]: Renamed interface ppp0 to pppoe-wan
Thu Nov  3 12:59:50 2022 daemon.info pppd[19239]: Using interface pppoe-wan
Thu Nov  3 12:59:50 2022 daemon.notice pppd[19239]: Connect: pppoe-wan <--> wan
Thu Nov  3 12:59:53 2022 daemon.info pppd[19239]: syncppp not active
Thu Nov  3 12:59:53 2022 daemon.info pppd[19239]: CHAP authentication succeeded: Authentication success,Welcome!
Thu Nov  3 12:59:53 2022 daemon.notice pppd[19239]: CHAP authentication succeeded
Thu Nov  3 12:59:53 2022 daemon.notice pppd[19239]: peer from calling number 60:08:10:09:C8:30 authorized
Thu Nov  3 12:59:53 2022 daemon.notice pppd[19239]: local  LL address fe80::dc9b:8f12:04c9:40a0
Thu Nov  3 12:59:53 2022 daemon.notice pppd[19239]: remote LL address fe80::6208:10ff:fe09:c830
Thu Nov  3 12:59:53 2022 daemon.notice netifd: Network device 'pppoe-wan' link is up
Thu Nov  3 12:59:54 2022 daemon.notice netifd: Interface 'wan' is now up
Thu Nov  3 12:59:54 2022 daemon.notice pppd[19239]: local  IP address 123.145.XXX.XXX
Thu Nov  3 12:59:54 2022 daemon.notice pppd[19239]: remote IP address 123.145.96.1
Thu Nov  3 12:59:54 2022 daemon.notice pppd[19239]: primary   DNS address 221.7.92.98
Thu Nov  3 12:59:54 2022 daemon.notice pppd[19239]: secondary DNS address 221.5.203.98
Thu Nov  3 12:59:54 2022 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Thu Nov  3 12:59:54 2022 daemon.notice netifd: Interface 'wan_6' is enabled
Thu Nov  3 12:59:54 2022 daemon.notice netifd: Interface 'wan_6' has link connectivity
Thu Nov  3 12:59:54 2022 daemon.notice netifd: Interface 'wan_6' is setting up now
Thu Nov  3 12:59:56 2022 daemon.notice netifd: Interface 'wan_6' is now up
Thu Nov  3 13:00:04 2022 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Thu Nov  3 13:00:04 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan (pppoe-wan)
Thu Nov  3 13:00:04 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Thu Nov  3 13:00:15 2022 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Thu Nov  3 13:00:15 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan_6 (pppoe-wan)

ramondelee, 2022-11-05:
补充下版本号，OpenWrt R22.10.20 / LuCI Master (git-22.289.58375-2a380a3)

ZqinKing, 2022-11-15:
redmi ax6000 现在能编译了吗？

bf30075, 2023-03-25:
我也这问题

coolsnowwolf, 2026-09-05:
日志显示 ISP 重连后 PPPoE 完成 CHAP 认证并成功获得 IPv4、DNS 与 IPv6 接口，符合运营商定期强制重连场景；没有 PPP 协议失败、内核或 netifd 错误，当前无 LEDE 代码修复。

closed 2026-09-05 by coolsnowwolf