coolsnowwolf/lede #10379: 编译 luci-app-mosdns 时发生错误

xchawai, 2022-11-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译 luci-app-mosdns 时发生错误

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：
Collected errors:
 * check_data_file_clashes: Package luci-app-mosdns wants to install file /home/xchawai/lede/build_dir/target-x86_64_musl/root-x86/etc/init.d/mosdns
        But that file is already provided by package  * mosdns
 * opkg_install_cmd: Cannot install package luci-app-mosdns.
 * check_data_file_clashes: Package luci-app-mosdns wants to install file /home/xchawai/lede/build_dir/target-x86_64_musl/root-x86/etc/init.d/mosdns
        But that file is already provided by package  * mosdns
 * opkg_install_cmd: Cannot install package luci-i18n-mosdns-zh-cn.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/xchawai/lede'
make[1]: *** [package/Makefile:111: /home/xchawai/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/xchawai/lede'
make: *** [/home/xchawai/lede/include/toplevel.mk:230: world] Error 2

hcym, 2022-11-06:
你看看git，除非昨天晚上改了什么
最近一直正常的，用官方编译就改一个压缩选择，也是一直正常

xchawai, 2022-11-06:
> 你看看git，除非昨天晚上改了什么 最近一直正常的，用官方编译就改一个压缩选择，也是一直正常

抱歉，我加的第三方源里也有mosdns，好像冲突了，这个该怎么解决呢？
https://github.com/kenzok8/small-package

Plutonium141, 2022-11-06:
编译时不选mosdns  之后下载mosdns的ipk手动安装

xchawai, 2022-11-06:
> 编译时不选mosdns 之后下载mosdns的ipk手动安装

？？？好吧也算一种方法

coldrook, 2022-11-06:
##fixed mosdns
rm -rf ./feeds/packages/net/mosdns
cp -r -f ./feeds/第三方源的文件 ./feeds/packages/net/mosdns

xchawai, 2022-11-06:
> ##fixed mosdns rm -rf ./feeds/packages/net/mosdns cp -r -f ./feeds/第三方源的文件 ./feeds/packages/net/mosdns

谢谢！！

closed 2022-11-06 by xchawai