coolsnowwolf/lede #10382: 更新最新源码后编译出错

mitsukileung, 2022-11-07:
ipq807x

Create index file './feeds/packages.index' 
Create index file './feeds/luci.index' 
Create index file './feeds/routing.index' 
Create index file './feeds/telephony.index' 
Create index file './feeds/oui.index' 
Create index file './feeds/helloworld.index' 
WARNING: Makefile 'package/feeds/packages/python-paho-mqtt/Makefile' has a dependency on 'python3-uuid', which does not exist
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed oui.
Installing all packages from feed helloworld.
这是缺少依赖？

Applying ./patches/001-procd_ujail_static_binary_fix.patch using plaintext: 
patching file jail/elf.c
Hunk #1 FAILED at 240.
1 out of 1 hunk FAILED -- saving rejects to file jail/elf.c.rej
Patch failed!  Please fix ./patches/001-procd_ujail_static_binary_fix.patch!
make[3]: *** [Makefile:181: /home/mitsuki/lede/build_dir/target-aarch64_cortex-a53_musl/procd-default/procd-2021-11-04-0ee8e734/.prepared_7809c874b685fcee9579ba572f1de4ed_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/mitsuki/lede/package/system/procd'
time: package/system/procd/default/compile#0.36#0.19#0.49
    ERROR: package/system/procd failed to build (build variant: default).
make[2]: *** [package/Makefile:116: package/system/procd/compile] Error 1
make[2]: Leaving directory '/home/mitsuki/lede'
make[1]: *** [package/Makefile:110: /home/mitsuki/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/mitsuki/lede'
make: *** [/home/mitsuki/lede/include/toplevel.mk:230：world] 错误 2

zonyitoo, 2022-11-07:
我提交的patch，我来定位一下

CJLUGaoXinyu, 2022-11-07:
同样的问题

bi4tms, 2022-11-07:
编译x86内核的固件出现同样报错

mitsukileung, 2022-11-07:
> 我提交的patch，我来定位一下

谢谢大佬，可以了

bi4tms, 2022-11-07:
> > 我提交的patch，我来定位一下
> 
> 谢谢大佬，可以了

请问如何更新

bi4tms, 2022-11-07:
> 我提交的patch，我来定位一下

大佬，请问如何更新

dl250, 2022-11-07:
没好吧，还是老样子

mitsukileung, 2022-11-07:
> 没好吧，还是老样子

同步这个 https://github.com/zonyitoo/lede/commit/cb367c32ed51d14815befaeabbff42f2117896a0 patch就可以了

binge8, 2022-11-07:
还是老样子，actions全新环境编译

> > 没好吧，还是老样子
> 
> 同步这个 [zonyitoo@cb367c3](https://github.com/zonyitoo/lede/commit/cb367c32ed51d14815befaeabbff42f2117896a0) patch就可以了

Oumourin, 2022-11-07:
还没好+1

closed 2022-11-07 by aiamadeus

qiuzhongqiang, 2022-11-07:
同样的问题呢。请大佬修复

zonyitoo, 2022-11-08:
理论上昨天已经修复了，编译结果都出来了

yifang5775, 2022-11-08:
还是会提示
WARNING: Makefile 'package/feeds/packages/python-paho-mqtt/Makefile' has a dependency on 'python3-uuid', which does not exist

PR #10384 (2022-11-07): procd: ujail upgrade patch for latest version
ref #10381
fix #10382

Q：你知道这是`pull request`吗？(使用 "x" 选择)
* [x] 我知道
commit 39ed1b72 (2022-11-07): procd: ujail upgrade patch for latest version (#10384)

Fix: #10382