coolsnowwolf/lede #10386: softher vpn连不上

lvmingye523, 2022-11-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：新的版本连不上，用2个月前的版本就能连上，设置操作完全一致。l2tp的端口查看config也是被打开的。很奇怪

### (2) 路由器型号和固件版本

A：r5s 。内核版本5.15。

### (3) 详细日志

A：日志没有关于softher报错问题。一切看起来正常

jdnt, 2022-11-08:
是否用了passwall2或者xray分流[分流总结点]，前段时间我这么配置后softether的l2tp死活连不上，只能用softether自带的openvpn，直到我用回了passwall……

richaradlee, 2022-11-08:
同样的问题，昨天编译的最新固件安装后，Sofether就死活连不上了。插件就个passwall在用。。。

lvmingye523, 2022-11-08:
有大佬关注一下吗，难道全能连上吗？

lvmingye523, 2022-11-09:
> 查了下，Sofether版本变更了，现在是Sofether5 ，原来的是4 ，但版本变了，设置也变了？？有哪位大神设置成功了？

早就是5了。2个月前我编的版本就是5了，一样的。管理工具。我看防火墙端口规则也写进去了。很奇怪

richaradlee, 2022-11-09:
各位别发愁了，请参考这个：https://github.com/coolsnowwolf/lede/issues/4546 里面有答案。
我是卸载了5.03 安装了4.38-9760 一切又恢复了。。。真不容易

lvmingye523, 2022-11-09:
> 各位别发愁了，请参考这个：#4546 里面有答案。 我是卸载了5.03 安装了4.38-9760 一切又恢复了。。。真不容易

这里面的是让关闭flow，但是应该不是这个问题。可能真的是版本号的问题吧。

Mousehk, 2022-11-11:
同样的问题 softetherVPN  配置好连不上  之前版本一直用着好好的

lvmingye523, 2022-11-12:
哪个大佬出来修一下。。。right论坛也有很多人有类似的反应

hellcatjack, 2022-11-13:
编译了11月10日版本，发现softether ddns无法上报ipv6地址，全球ipv6地址一栏一直显示无法找到对象。
测试openvpn，结果andorid的openvpn客户端无法连接，随后通过softether管理器重新生成openvpn配置文件后连接成功。
看样子这版本变化有点大啊。。。

hellcatjack, 2022-11-13:
https://downloads.openwrt.org/releases/packages-19.07/x86_64/packages/softethervpn5-bridge_5.01.9674-1_x86_64.ipk
https://downloads.openwrt.org/releases/packages-19.07/x86_64/packages/softethervpn5-client_5.01.9674-1_x86_64.ipk
https://downloads.openwrt.org/releases/packages-19.07/x86_64/packages/softethervpn5-libs_5.01.9674-1_x86_64.ipk
https://downloads.openwrt.org/releases/packages-19.07/x86_64/packages/softethervpn5-server_5.01.9674-1_x86_64.ipk

有问题的兄弟先临时降级一下吧。

hellcatjack, 2022-11-14:
https://downloads.openwrt.org/releases/packages-19.07/aarch64_generic/packages/softethervpn5-bridge_5.01.9674-1_aarch64_generic.ipk
https://downloads.openwrt.org/releases/packages-19.07/aarch64_generic/packages/softethervpn5-client_5.01.9674-1_aarch64_generic.ipk
https://downloads.openwrt.org/releases/packages-19.07/aarch64_generic/packages/softethervpn5-libs_5.01.9674-1_aarch64_generic.ipk
https://downloads.openwrt.org/releases/packages-19.07/aarch64_generic/packages/softethervpn5-server_5.01.9674-1_aarch64_generic.ipk
不好意思，没看到楼主是r5s，用这个吧。

2703519, 2022-11-14:
先退回5.01.9674吧，等大佬修

lvmingye523, 2022-11-14:
> https://downloads.openwrt.org/releases/packages-19.07/aarch64_generic/packages/softethervpn5-bridge_5.01.9674-1_aarch64_generic.ipk [...]

果然是版本的问题，5.01就好使，5.02都连不上

goodwinY, 2022-11-18:
卸载了后在软件包里再安装一遍就行

lvmingye523, 2022-11-19:
> 卸载了后在软件包里再安装一遍就行

是先在软件包里点移除吗，再点安装

goodwinY, 2022-11-19:
> > 卸载了后在软件包里再安装一遍就行
> 
> 是先在软件包里点移除吗，再点安装

建议使用5.01.9674-1版本

dlj390, 2022-11-20:
请问如何操作呢？卸载可以如何安装保留配置呢？

shenyz, 2022-11-20:
> 请问如何操作呢？卸载可以如何安装保留配置呢？

导出配置，然后导入，一般它自己会保留配置的，直接删除后安装底版本，再进去看下配置

goodwinY, 2022-11-21:
> 请问如何操作呢？卸载可以如何安装保留配置呢？

配置目录在  /usr/libexec/softethervpn

richaradlee, 2022-11-21:
L大已经修好了，版本更新了，而且最新编译的固件softethervpn已经正常了。

Mousehk, 2022-11-23:
刚编译5.15.79    softethervpn版本5.02_5180_1还是连不上啊？

richaradlee, 2022-11-24:
你编译的软件包怎么是5.02.5180-1啊？我这里编译完的是5.02.5180-3了。

richaradlee, 2022-11-24:
[image]

Mousehk, 2022-11-24:
你固件版本多少 什么架构类型的

Mousehk, 2022-11-24:
我也是啊 我全删了又重新编译了一次 同样是5.15.79  softethervpn版本还是5.02_5180_1 ？？？
这什么情况有大佬知道吗？

goodwinY, 2022-11-24:
> 我也是啊 我全删了又重新编译了一次 同样是5.15.79 softethervpn版本还是5.02_5180_1 ？？？ 这什么情况有大佬知道吗？

我测试过了编译安装实体机，不得行，不要挣扎了，还是先不要编译softher vpn，下载softethervpn5-5.01.9674版本，安装使用算了。

shenyz, 2022-11-25:
> > 我也是啊 我全删了又重新编译了一次 同样是5.15.79 softethervpn版本还是5.02_5180_1 ？？？ 这什么情况有大佬知道吗？
> 
> 我测试过了编译安装实体机，不得行，不要挣扎了，还是先不要编译softher vpn，下载softethervpn5-5.01.9674版本，安装使用算了。
是的  我看了老版和新版的区别，新版的500 4500端口监听地址全是IPV6，没有IPV4

zzy19868, 2022-11-30:
> 你编译的软件包怎么是5.02.5180-1啊？我这里编译完的是5.02.5180-3了。

最新的我编译出来也是-1，奇怪。

hellcatjack, 2022-12-04:
目前softethervpn5已经revert到5.01.9674，大家可以正常build了。

WYC-2020, 2022-12-06:
https://github.com/coolsnowwolf/lede/issues/10520
看这个帖子 里面我又说怎么解决

coolsnowwolf, 2026-09-05:
SoftEther 新旧版本连接差异没有错误日志或完整配置，无法确认是 LEDE 包问题，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf