coolsnowwolf/lede #10389: X86 编译失败

selfcan, 2022-11-08:
time: package/utils/bzip2/compile#0.40#0.13#0.52
make[3]: Entering directory '/home/deepin/文档/lede/package/system/ca-certificates'
mkdir -p /home/deepin/文档/lede/dl
SHELL= flock /home/deepin/文档/lede/tmp/.ca-certificates_20210119.tar.xz.flock -c '  	/home/deepin/文档/lede/scripts/download.pl "/home/deepin/文档/lede/dl" "ca-certificates_20210119.tar.xz" "daa3afae563711c30a0586ddae4336e8e3974c2b627faaca404c4e0141b64665" "" "@DEBIAN/pool/main/c/ca-certificates"    '
mkdir -p /home/deepin/文档/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/deepin/文档/lede/tmp/.root-copy.flock -c 'cp -fpR /home/deepin/文档/lede/build_dir/target-x86_64_musl/ca-certificates-20210119/.pkgdir/ca-bundle/. /home/deepin/文档/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/deepin/文档/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.ca-bundle_installed
echo "ca-bundle" >> /home/deepin/文档/lede/staging_dir/target-x86_64_musl/pkginfo/ca-certificates.default.install
mkdir -p /home/deepin/文档/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/deepin/文档/lede/tmp/.root-copy.flock -c 'cp -fpR /home/deepin/文档/lede/build_dir/target-x86_64_musl/ca-certificates-20210119/.pkgdir/ca-certificates/. /home/deepin/文档/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/deepin/文档/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.ca-certificates_installed
echo "ca-certificates" >> /home/deepin/文档/lede/staging_dir/target-x86_64_musl/pkginfo/ca-certificates.default.install
make[3]: Leaving directory '/home/deepin/文档/lede/package/system/ca-certificates'
time: package/system/ca-certificates/compile#0.33#0.20#0.75
make[3]: Entering directory '/home/deepin/文档/lede/feeds/packages/lang/python/python3'
mkdir -p /home/deepin/文档/lede/dl
SHELL= flock /home/deepin/文档/lede/tmp/.Python-3.9.15.tar.xz.flock -c '  	/home/deepin/文档/lede/scripts/download.pl "/home/deepin/文档/lede/dl" "Python-3.9.15.tar.xz" "12daff6809528d9f6154216950423c9e30f0e47336cb57c6aa0b4387dd5eb4b2" "" "https://www.python.org/ftp/python/3.9.15"    '
rm -f /home/deepin/文档/lede/build_dir/target-x86_64_musl/Python-3.9.15/.built
touch /home/deepin/文档/lede/build_dir/target-x86_64_musl/Python-3.9.15/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/deepin/文档/lede/build_dir/target-x86_64_musl/Python-3.9.15=Python-3.9.15 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -fno-inline -I/home/deepin/文档/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/deepin/文档/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/deepin/文档/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/deepin/文档/lede/build_dir/target-x86_64_musl/Python-3.9.15=Python-3.9.15 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -fno-inline -I/home/deepin/文档/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/deepin/文档/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/deepin/文档/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " LDFLAGS="-L/home/deepin/文档/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/deepin/文档/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro -L/home/deepin/文档/lede/build_dir/target-x86_64_musl/Python-3.9.15 -L/home/deepin/文档/lede/staging_dir/target-x86_64_musl/usr/lib -L/home/deepin/文档/lede/staging_dir/target-x86_64_musl/lib" make  -C /home/deepin/文档/lede/build_dir/target-x86_64_musl/Python-3.9.15/. AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/deepin/文档/lede/build_dir/target-x86_64_musl/Python-3.9.15=Python-3.9.15 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size READELF="x86_64-openwrt-linux-musl-readelf" CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" LD="x86_64-openwrt-linux-musl-gcc" ;
make[4]: Entering directory '/home/deepin/文档/lede/build_dir/target-x86_64_musl/Python-3.9.15'
Running code to generate profile data (this can take a while):
# First, we need to create a clean build with profile generation
# enabled.
make profile-gen-stamp
make[5]: Entering directory '/home/deepin/文档/lede/build_dir/target-x86_64_musl/Python-3.9.15'
Building with support for profile generation:
make build_all_generate_profile
make[6]: Entering directory '/home/deepin/文档/lede/build_dir/target-x86_64_musl/Python-3.9.15'
make build_all CFLAGS_NODIST=" -fprofile-generate" LDFLAGS_NODIST=" -fprofile-generate" LIBS="-ldl  -lpthread -lm"
make[7]: Entering directory '/home/deepin/文档/lede/build_dir/target-x86_64_musl/Python-3.9.15'
 CC='x86_64-openwrt-linux-musl-gcc' LDSHARED='x86_64-openwrt-linux-musl-gcc -shared -L/home/deepin/文档/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/deepin/文档/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro -L/home/deepin/文档/lede/build_dir/target-x86_64_musl/Python-3.9.15 -L/home/deepin/文档/lede/staging_dir/target-x86_64_musl/usr/lib -L/home/deepin/文档/lede/staging_dir/target-x86_64_musl/lib -L/home/deepin/文档/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/deepin/文档/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro -L/home/deepin/文档/lede/build_dir/target-x86_64_musl/Python-3.9.15 -L/home/deepin/文档/lede/staging_dir/target-x86_64_musl/usr/lib -L/home/deepin/文档/lede/staging_dir/target-x86_64_musl/lib -flto -fuse-linker-plugin -ffat-lto-objects -flto-partition=none -g -fprofile-generate' OPT='-DNDEBUG -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/deepin/文档/lede/build_dir/target-x86_64_musl/Python-3.9.15=Python-3.9.15 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro' 	_TCLTK_INCLUDES='' _TCLTK_LIBS='' 	_PYTHON_PROJECT_BASE=/home/deepin/文档/lede/build_dir/target-x86_64_musl/Python-3.9.15 _PYTHON_HOST_PLATFORM=linux-x86_64 PYTHONPATH=/home/deepin/文档/lede/build_dir/target-x86_64_musl/Python-3.9.15/build/lib.linux-x86_64-3.9:./Lib _PYTHON_SYSCONFIGDATA_NAME=_sysconfigdata python3.9 ./setup.py  build
running build
running build_ext
Header file /home/deepin/\�\�\�\�\�\�/lede/staging_dir/target-x86_64_musl/usr/include/ffi.h does not exist
INFO: Can't locate Tcl/Tk libs and/or headers
building '_hashlib' extension
x86_64-openwrt-linux-musl-gcc -fPIC -Wno-unused-result -Wsign-compare -DNDEBUG -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/deepin/文档/lede/build_dir/target-x86_64_musl/Python-3.9.15=Python-3.9.15 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/deepin/文档/lede/build_dir/target-x86_64_musl/Python-3.9.15=Python-3.9.15 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -fno-inline -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/deepin/文档/lede/build_dir/target-x86_64_musl/Python-3.9.15=Python-3.9.15 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -fno-inline -I/home/deepin/文档/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/deepin/文档/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/deepin/文档/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include -flto -fuse-linker-plugin -ffat-lto-objects -flto-partition=none -g -std=c99 -Wextra -Wno [... cut ...]
make[3]: Leaving directory '/home/deepin/文档/lede/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/compile#1.06#0.41#2.37
    ERROR: package/feeds/packages/python3 failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/python3/compile] Error 1
make[2]: Leaving directory '/home/deepin/文档/lede'
make[1]: *** [package/Makefile:110: /home/deepin/文档/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/deepin/文档/lede'
make: *** [/home/deepin/文档/lede/include/toplevel.mk:230：world] 错误 2

1054009064, 2022-11-08:
用纯英文目录试试。

ecrasy, 2022-11-08:
新手建议使用别人已经成功的config文件练手
直接拷贝别人的config文件到lede源码目录
覆盖.config文件
make defconfig && make menuconfig
在次基础上修改定制你需要的模块
然后再make

liuhejun2020, 2022-11-08:
这个是编译X86目标文件Legacy类型的固件遇到的,X86 64的和N1的没有出现这个错误,怀疑是源码里对Legcay支持的不够,或者需要什么库在feed.conf.default里需要加上

HollyMadCow, 2022-11-09:
先别放在中文路径下试试

liuhejun2020, 2022-11-10:
我的编译环境都是英文目录啊，还用过云编译也是出错，只要目标文件是Legacy 就不行，其他的可以。

closed 2022-11-11 by aiamadeus