coolsnowwolf/lede #10394: RTL8111 网卡IRQ高，CPU负载大，系统卡顿

wd2es7, 2022-11-09:
具体问题
1. 固件为[openwrt-x64-R22.10.1-squashfs-combined-efi.img.gz]
2. 运行2-10小时后出现带宽降低，CPU负载高的情况
3. 其中网卡的IRQ很高
4. 刷回 [openwrt-R20.4.8-x64-combined-squashfs.1.rar] 后一切正常；本人观察到该固件使用的是RTL8169驱动

$ cat /proc/interrupts
            CPU0       CPU1       CPU2       CPU3       
   4:         30          0          0          0   IO-APIC    4-edge      ttyS0
   8:          0          0          0          0   IO-APIC    8-edge      rtc0
   9:          0          0          0          0   IO-APIC    9-fasteoi   acpi
  47:          0         52          0          0   IO-APIC   47-fasteoi   mmc0
 119:          0      15135          0          0   PCI-MSI 311296-edge      ahci[0000:00:13.0]
 120:          0          0          0          0   PCI-MSI 327680-edge      xhci_hcd
 121:          0          0    1370033          0   PCI-MSI 32768-edge      i915
 122:          0          0          0   58130886   PCI-MSI 524288-edge      eth0
 123:          0          0          0          0   PCI-MSI 1048576-edge      eth1
 124:          0          0          0          0   PCI-MSI 1572864-edge      eth2
 125:          0          0   54094149          0   PCI-MSI 2097152-edge      eth3
 126:          0          0          0         67   PCI-MSI 442368-edge      snd_hda_intel:card0
 NMI:          0          0          0          0   Non-maskable interrupts
 LOC:   17309092   17607914   19417091   18238876   Local timer interrupts
 SPU:          0          0          0          0   Spurious interrupts
 PMI:          0          0          0          0   Performance monitoring interrupts
 IWI:          0          0         19          4   IRQ work interrupts
 RTR:          0          0          0          0   APIC ICR read retries
 RES:     120608     126855     138213     158233   Rescheduling interrupts
 CAL:     709464     691885     622841     682402   Function call interrupts
 TLB:       8022       9418      10108      13199   TLB shootdowns
 TRM:          0          0          0          0   Thermal event interrupts
 THR:          0          0          0          0   Threshold APIC interrupts
 DFR:          0          0          0          0   Deferred Error APIC interrupts
 MCE:          0          0          0          0   Machine check exceptions
 MCP:        144        144        144        144   Machine check polls
 ERR:          0
 MIS:          0
 PIN:          0          0          0          0   Posted-interrupt notification event
 NPI:          0          0          0          0   Nested posted-interrupt event
 PIW:          0          0          0          0   Posted-interrupt wakeup event

$ dmesg | grep eth0
[   24.616526] eth0: 0xffffc90000031000, 00:e0:4c:69:01:81, IRQ 122
[   29.053091] r8168: eth0: link up
[   29.056905] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   44.679166] eth0: renamed from veth136c6ad

$ ethtool -i eth0 
driver: r8168
version: 8.050.03-NAPI
firmware-version: 
expansion-rom-version: 
bus-info: 0000:01:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: no

### (2) 路由器型号和固件版本

A：路由器为N3710的软路由，4口RTL8111的千兆接口

个人猜测可能是“启用 RTL 鸡血驱动（ RTL81xx 和 RTL8125系列）”导致的问题

求解，谢谢各位

hcym, 2022-11-09:
我的还是去年五月的最好用，呵呵

minlang112, 2022-11-09:
没遇到此类情况，建议说明“网卡IRQ高，CPU负载大”的时候在跑什么应用

wd2es7, 2022-11-09:
> 没遇到此类情况，建议说明“网卡IRQ高，CPU负载大”的时候在跑什么应用

openclash
DDNS
Docker - Alist

这三个全部关掉依然可以复现问题

closed 2022-11-13 by coolsnowwolf