coolsnowwolf/lede #10400: 编译X86 Legacy 出错日志，编译其他的X86 64不出错

liuhejun2020, 2022-11-11:
编译路由器型号：N2600，必须用 Legacy 刷机
出错部分日志
SHELL= flock /home/lhj/lede/tmp/.ddnsto-binary-3.0.2.tar.gz.flock -c '          /home/lhj/lede/scripts/download.pl "/home/lhj/lede/dl" "ddnsto-binary-3.0.2.tar.gz" "76e6dc6a51176062a625f7bc9931d8709f81146603f5a7da853f1dc9ab2a05a3" "" "http://fw.koolcenter.com/binary/ddnsto/"    '
touch /home/lhj/lede/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.prepared_8a7184727e1bb6e204f9ac4880a021bf_6664517399ebbbc92a37c5bb081b5c53_check
. /home/lhj/lede/include/shell.sh; gzip -dc /home/lhj/lede/dl/ddnsto-binary-3.0.2.tar.gz | tar -C /home/lhj/lede/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/lhj/lede/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2
touch /home/lhj/lede/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.prepared_8a7184727e1bb6e204f9ac4880a021bf_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/lhj/lede/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.configured_*
rm -f /home/lhj/lede/staging_dir/target-i386_pentium-mmx_musl/stamp/.ddnsto_installed
touch /home/lhj/lede/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/lhj/lede/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.built
touch /home/lhj/lede/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.built_check
touch /home/lhj/lede/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.built
rm -rf /home/lhj/lede/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.pkgdir/ddnsto.installed /home/lhj/lede/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.pkgdir/ddnsto
mkdir -p /home/lhj/lede/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.pkgdir/ddnsto
install -d -m0755 /home/lhj/lede/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.pkgdir/ddnsto/usr/sbin /home/lhj/lede/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.pkgdir/ddnsto/etc/config /home/lhj/lede/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.pkgdir/ddnsto/etc/init.d /home/lhj/lede/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.pkgdir/ddnsto/etc/uci-defaults
install -m0755 /home/lhj/lede/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/ddnsto.i386 /home/lhj/lede/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.pkgdir/ddnsto/usr/sbin/ddnstod
install: cannot stat '/home/lhj/lede/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/ddnsto.i386': No such file or directory
make[3]: *** [Makefile:70: /home/lhj/lede/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.pkgdir/ddnsto.installed] Error 1
make[3]: Leaving directory '/home/lhj/lede/feeds/nas/network/services/ddnsto'
time: package/feeds/nas/ddnsto/compile#0.40#0.18#0.61
    ERROR: package/feeds/nas/ddnsto failed to build.
make[2]: *** [package/Makefile:116: package/feeds/nas/ddnsto/compile] Error 1
make[2]: Leaving directory '/home/lhj/lede'
make[1]: *** [package/Makefile:110: /home/lhj/lede/staging_dir/target-i386_pentium-mmx_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lhj/lede'
make: *** [/home/lhj/lede/include/toplevel.mk:230：world] 错误 2
lhj@debian:~/lede$

EnnawYang, 2022-11-11:
Legacy默认配置编译一遍过没出错，另外我没有搜到ddnsto这个包，是你自己另外加的吗  
N2600不是支持64位的吗，被屏蔽了？

liuhejun2020, 2022-11-11:
这个ddnsto是后加的，但是在编译其他目标文件时都正常通过，我现在去掉这个选项再编译一下，我记得上次去掉这个选项后其他地方也出错了，等我编译一下上个日志

EnnawYang, 2022-11-11:
[image]
并没有提供32(i386)位的文件

coolsnowwolf, 2022-11-13:
ddnsto.i386': No such file or directory

closed 2022-11-21 by aiamadeus