coolsnowwolf/lede #10405: 最新编译的lede固件没有菜单没有ttyd终端，请大佬帮忙修复，自己不会搞。。

759071793, 2022-11-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：今天编译的lede固件ttyd终端菜单没有显示，

### (2) 路由器型号和固件版本

A：
X86
OpenWrt R22.11.11 / LuCI Master (git-22.314.43289-84299a5)

### (3) 详细日志

A：Fri Nov 11 11:13:55 2022 daemon.err ttyd[16225]: Error loading shared library libwebsockets.so.17: No such file or directory (needed by /usr/bin/ttyd)

[image]

lovehackintosh, 2022-11-11:
make menuconfig 勾选`LuCI ---> Applications ---> luci-app-ttyd`

759071793, 2022-11-11:
> make menuconfig 勾选`LuCI ---> Applications ---> luci-app-ttyd`

你好这个我编译的时候勾选了的，我的config文件一直没动过，我还单独在软件包里面卸载重新在线安装的，也不行。

759071793, 2022-11-11:
libwebsockets.so这个文件要怎么安装呢。。

hgh188, 2022-11-15:
一样的问题 反复编译了

breakings, 2022-11-15:
`make package/feeds/packages/ttyd/{clean,compile} -j8 V=s `

hgh188, 2022-11-15:
> `make package/feeds/packages/ttyd/{clean,compile} -j8 V=s `
[image]

flyhong666, 2022-11-15:
是的，菜单没显示了，咋回事？

759071793, 2022-11-15:
> make package/feeds/packages/ttyd/{clean,compile} -j8 V=s

大佬牛逼，有效！

759071793, 2022-11-15:
> `make package/feeds/packages/ttyd/{clean,compile} -j8 V=s `
> [image]

make package/feeds/packages/ttyd/clean
make package/feeds/packages/ttyd/compile -j8 V=s