coolsnowwolf/lede #10408: target/linux/generic/pending-5.4/499-mtd-spi-nor-add-xmc-xm25qh128c.patch need fix

EnnawYang, 2022-11-11:
Target: ramips / mt7621
Device: d-team_newifi-d2 |  redmi-ac2100

logs:
```
Applying /home/xw/openwrt/openwrt/target/linux/generic/pending-5.4/497-mtd-mtdconcat-add-dt-driver-for-concat-devices.patch using plaintext: 
patching file drivers/mtd/Kconfig
patching file drivers/mtd/Makefile
patching file drivers/mtd/composite/Kconfig
patching file drivers/mtd/composite/Makefile
patching file drivers/mtd/composite/virt_concat.c

Applying /home/xw/openwrt/openwrt/target/linux/generic/pending-5.4/498-mtd-mtdconcat-select-readwrite-function.patch using plaintext: 
patching file drivers/mtd/mtdconcat.c

Applying /home/xw/openwrt/openwrt/target/linux/generic/pending-5.4/499-mtd-spi-nor-add-xmc-xm25qh128c.patch using plaintext: 
patching file drivers/mtd/spi-nor/spi-nor.c
Hunk #1 FAILED at 2565.
1 out of 1 hunk FAILED -- saving rejects to file drivers/mtd/spi-nor/spi-nor.c.rej
Patch failed!  Please fix /home/xw/openwrt/openwrt/target/linux/generic/pending-5.4/499-mtd-spi-nor-add-xmc-xm25qh128c.patch!
make[3]: *** [Makefile:122: /home/xw/openwrt/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/linux-5.4.223/.prepared] Error 1
```

closed 2022-11-11 by aiamadeus

dwj0, 2022-11-11:
499补丁应用出错了，和485补丁重复了，试试把499补丁删除

EnnawYang, 2022-11-11:
> 499补丁应用出错了，和485补丁重复了，试试把499补丁删除

已经修复了

commit fced2da8 (2022-11-11): ramips: fixes typo error in kang

Fixes: 9a383de4afa6ae520147763262e521da97696224
Closed: #10408