coolsnowwolf/lede #10415: 更新源码后编译出错

mitsukileung, 2022-11-11:
X86

make[3]: Entering directory '/home/mitsuki/lede/feeds/packages/sound/forked-daapd'
mkdir -p /home/mitsuki/lede/dl
SHELL= flock /home/mitsuki/lede/tmp/.forked-daapd-27.4.tar.xz.flock -c '  	/home/mitsuki/lede/scripts/download.pl "/home/mitsuki/lede/dl" "forked-daapd-27.4.tar.xz" "00f71c687df268a3c4db77cecf37972e76832a99e34f09750d07a92934a0bfa8" "" "https://github.com/ejurgensen/forked-daapd/releases/download/27.4/"    '
rm -rf /home/mitsuki/lede/build_dir/target-x86_64_musl/forked-daapd-27.4/.pkgdir/forked-daapd.installed /home/mitsuki/lede/build_dir/target-x86_64_musl/forked-daapd-27.4/.pkgdir/forked-daapd
mkdir -p /home/mitsuki/lede/build_dir/target-x86_64_musl/forked-daapd-27.4/.pkgdir/forked-daapd
install -d -m0755 /home/mitsuki/lede/build_dir/target-x86_64_musl/forked-daapd-27.4/.pkgdir/forked-daapd/usr/sbin
install -m0755 /home/mitsuki/lede/build_dir/target-x86_64_musl/forked-daapd-27.4/ipkg-install/usr/sbin/forked-daapd /home/mitsuki/lede/build_dir/target-x86_64_musl/forked-daapd-27.4/.pkgdir/forked-daapd/usr/sbin/
install -d -m0755 /home/mitsuki/lede/build_dir/target-x86_64_musl/forked-daapd-27.4/.pkgdir/forked-daapd/etc
install -m0600 ./files/forked-daapd.conf /home/mitsuki/lede/build_dir/target-x86_64_musl/forked-daapd-27.4/.pkgdir/forked-daapd/etc/forked-daapd.conf
install: cannot stat './files/forked-daapd.conf': No such file or directory
make[3]: *** [Makefile:86: /home/mitsuki/lede/build_dir/target-x86_64_musl/forked-daapd-27.4/.pkgdir/forked-daapd.installed] Error 1
make[3]: Leaving directory '/home/mitsuki/lede/feeds/packages/sound/forked-daapd'
time: package/feeds/packages/forked-daapd/compile#0.39#0.18#0.52
    ERROR: package/feeds/packages/forked-daapd failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/forked-daapd/compile] Error 1
make[2]: Leaving directory '/home/mitsuki/lede'
make[1]: *** [package/Makefile:110: /home/mitsuki/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/mitsuki/lede'
make: *** [/home/mitsuki/lede/include/toplevel.mk:230：world] 错误 2

Cptmacmillan2022007, 2022-11-12:
我也是这个错误，是在线编译的。

和这个有关系：
···
install -m0600 ./files/forked-daapd.conf /home/mitsuki/lede/build_dir/target-x86_64_musl/forked-daapd-27.4/.pkgdir/forked-daapd/etc/forked-daapd.conf
install: cannot stat './files/forked-daapd.conf': No such file or directory
···

liwenjie119, 2022-11-12:
同样，昨天下午编译出错：
install: cannot stat './files/forked-daapd.conf': No such file or directory

bootli, 2022-11-12:
一样

install: cannot stat './files/forked-daapd.conf': No such file or directory

Beginner-Go, 2022-11-12:
fixed.

wjz304, 2022-11-12:
+1..., try again.

wjz304, 2022-11-12:
error again.

Beginner-Go, 2022-11-12:
Fixed.

mitsukileung, 2022-11-13:
> Fixed.

configure: error: Unable to link function av_init_packet
configure: WARNING: cache variable pkg_cv_LIBAV_VERSION contains a newline
make[3]: *** [Makefile:80: /home/mitsuki/lede/build_dir/target-x86_64_musl/owntone-28.5/.configured_a17fb5ef857664f03cd0ce37cc5ea591] Error 1
make[3]: Leaving directory '/home/mitsuki/lede/feeds/packages/sound/owntone'
time: package/feeds/packages/owntone/compile#16.50#2.56#20.36
ERROR: package/feeds/packages/owntone failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/owntone/compile] Error 1
make[2]: Leaving directory '/home/mitsuki/lede'
make[1]: *** [package/Makefile:110: /home/mitsuki/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/mitsuki/lede'
make: *** [/home/mitsuki/lede/include/toplevel.mk:230：world] 错误 2

还是会出错

wjz304, 2022-11-13:
error again.

> configure: error: Unable to link function av_init_packet configure: WARNING: cache variable pkg_cv_LIBAV_VERSION contains a newline make[3]: *** [Makefile:80: [...]

me too

ecrasy, 2022-11-13:
源码更新了需要
feeds update && feeds install

mitsukileung, 2022-11-13:
> 源码更新了需要 feeds update && feeds install

最新源码也是一样

wjz304, 2022-11-13:
> 源码更新了需要 feeds update && feeds install

在线的，每次都是新环境，

ecrasy, 2022-11-13:
那你要自己测试找原因啊
已经提示owntone编译失败了
自己动手丰衣足食

1582130940, 2022-11-15:
Generating index for package ./luci-app-music-remote-center_1-6_all.ipk
Generating index for package ./owntone_28.5-1_x86_64.ipk
我复现不出来啊，建议make distclean重新跑一下

mitsukileung, 2022-11-15:
> Generating index for package ./luci-app-music-remote-center_1-6_all.ipk Generating index for package ./owntone_28.5-1_x86_64.ipk 我复现不出来啊，建议make distclean重新跑一下

make distclean 后正常编译

closed 2022-11-15 by mitsukileung

wjz304, 2022-11-16:
```
HEAD is now at 486bebd Merge pull request #252 from rogers0/PR/fix_gcc9
Submodule 'libcork' (https://github.com/shadowsocks/libcork.git) registered for path 'libcork'
Cloning into '/home/ing/OpenWrt_Build/openwrt/tmp/dl/simple-obfs-0.0.5/libcork'...
fatal: error reading section header 'shallow-info'
fatal: clone of 'https://github.com/shadowsocks/libcork.git' into submodule path '/home/ing/OpenWrt_Build/openwrt/tmp/dl/simple-obfs-0.0.5/libcork' failed
Failed to clone 'libcork'. Retry scheduled
Cloning into '/home/ing/OpenWrt_Build/openwrt/tmp/dl/simple-obfs-0.0.5/libcork'...
fatal: error reading section header 'shallow-info'
fatal: clone of 'https://github.com/shadowsocks/libcork.git' into submodule path '/home/ing/OpenWrt_Build/openwrt/tmp/dl/simple-obfs-0.0.5/libcork' failed
Failed to clone 'libcork' a second time, aborting
make[3]: *** [Makefile:71: /home/ing/OpenWrt_Build/openwrt/dl/simple-obfs-0.0.5.tar.xz] Error 1
```
最终关闭 CONFIG_PACKAGE_luci-app-easymesh 后编译通过，是否为根本原因不详，仅作记录，勿关注。